morrownr/88x2bu #1: CONFIG_POWER_SAVING = y

nickpitlosh, 2020-09-22:
changing this to n makes the device actually connect and work properly on my ubuntu 20.04 system.

morrownr, 2020-09-22:
Would you be so kind as to provide some details? What did you experience prior to changing the power saving setting? What adapter are you using? What AP/router are you using and is power on or off in the AP/router?

I use this driver on multiple systems accessing 3 different Ap/routers and I am not seeing this problem. I need information in order to investigate. Thanks.

nickpitlosh, 2020-09-22:
the device would repeatedly enter and leave powersaving mode. when i get a chance i'll pull the journalctl log.

nickpitlosh, 2020-09-23:
```
sudo dmesg
[sudo] password for nick: 
[    0.000000] microcode: microcode updated early to revision 0x21, date = 2019-02-13
[    0.000000] Linux version 5.8.0-18-generic (buildd@lgw01-amd64-057) (gcc (Ubuntu 10.2.0-5ubuntu2) 10.2.0, GNU ld (GNU Binutils for Ubuntu) 2.35) #19-Ubuntu SMP Wed Aug 26 15:26:32 UTC 2020 (Ubuntu 5.8.0-18.19-generic 5.8.4)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.8.0-18-generic root=UUID=5f8b429a-4db0-49f0-97e5-db0d39df21c3 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000040003fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040004000-0x0000000040004fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040005000-0x00000000cbcf4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cbcf5000-0x00000000cbd37fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cbd38000-0x00000000cc428fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cc429000-0x00000000cc63efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cc63f000-0x00000000dc315fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc316000-0x00000000dc615fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dc616000-0x00000000dc740fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc741000-0x00000000dca5efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dca5f000-0x00000000dcffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dcfff000-0x00000000dcffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dd000000-0x00000000df9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041f5fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: ZOTAC ZBOX-ID18/ZBOX-ID18, BIOS B211P010 01/24/2014
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1496.529 MHz processor
[    0.003634] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003637] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.003650] last_pfn = 0x41f600 max_arch_pfn = 0x400000000
[    0.003657] MTRR default type: uncachable
[    0.003658] MTRR fixed ranges enabled:
[    0.003661]   00000-9FFFF write-back
[    0.003662]   A0000-BFFFF uncachable
[    0.003663]   C0000-DBFFF write-protect
[    0.003664]   DC000-E7FFF uncachable
[    0.003666]   E8000-FFFFF write-protect
[    0.003667] MTRR variable ranges enabled:
[    0.003669]   0 base 000000000 mask C00000000 write-back
[    0.003671]   1 base 400000000 mask FE0000000 write-back
[    0.003672]   2 base 0E0000000 mask FE0000000 uncachable
[    0.003673]   3 base 0DE000000 mask FFE000000 uncachable
[    0.003675]   4 base 0DD800000 mask FFF800000 uncachable
[    0.003676]   5 base 41F800000 mask FFF800000 uncachable
[    0.003677]   6 base 41F600000 mask FFFE00000 uncachable
[    0.003678]   7 disabled
[    0.003679]   8 disabled
[    0.003680]   9 disabled
[    0.004074] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.004252] total RAM covered: 16334M
[    0.004742]  gran_size: 64K 	chunk_size: 64K 	num_reg: 10  	lose cover RAM: 246M
[... 1041 lines cut ...]

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.8.0-18-generic -j3..............................................................................................
cleaning build area...

DKMS: build completed.

88x2bu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.8.0-18-generic/updates/dkms/

depmod.........

DKMS: install completed.
dkms-install.sh was successful.
``
```

nickpitlosh, 2020-09-23:
```
nick@zotac:~$ sudo dmesg
[sudo] password for nick: 
[    0.000000] microcode: microcode updated early to revision 0x21, date = 2019-02-13
[    0.000000] Linux version 5.8.0-18-generic (buildd@lgw01-amd64-057) (gcc (Ubuntu 10.2.0-5ubuntu2) 10.2.0, GNU ld (GNU Binutils for Ubuntu) 2.35) #19-Ubuntu SMP Wed Aug 26 15:26:32 UTC 2020 (Ubuntu 5.8.0-18.19-generic 5.8.4)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.8.0-18-generic root=UUID=5f8b429a-4db0-49f0-97e5-db0d39df21c3 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000040003fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040004000-0x0000000040004fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040005000-0x00000000cbcf4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cbcf5000-0x00000000cbd37fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cbd38000-0x00000000cc428fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cc429000-0x00000000cc63efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cc63f000-0x00000000dc315fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc316000-0x00000000dc615fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dc616000-0x00000000dc740fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc741000-0x00000000dca5efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dca5f000-0x00000000dcffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dcfff000-0x00000000dcffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dd000000-0x00000000df9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041f5fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: ZOTAC ZBOX-ID18/ZBOX-ID18, BIOS B211P010 01/24/2014
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1496.488 MHz processor
[    0.003636] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003638] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.003652] last_pfn = 0x41f600 max_arch_pfn = 0x400000000
[    0.003659] MTRR default type: uncachable
[    0.003660] MTRR fixed ranges enabled:
[    0.003662]   00000-9FFFF write-back
[    0.003663]   A0000-BFFFF uncachable
[    0.003664]   C0000-DBFFF write-protect
[    0.003666]   DC000-E7FFF uncachable
[    0.003667]   E8000-FFFFF write-protect
[    0.003668] MTRR variable ranges enabled:
[    0.003670]   0 base 000000000 mask C00000000 write-back
[    0.003672]   1 base 400000000 mask FE0000000 write-back
[    0.003674]   2 base 0E0000000 mask FE0000000 uncachable
[    0.003675]   3 base 0DE000000 mask FFE000000 uncachable
[    0.003676]   4 base 0DD800000 mask FFF800000 uncachable
[    0.003678]   5 base 41F800000 mask FFF800000 uncachable
[    0.003679]   6 base 41F600000 mask FFFE00000 uncachable
[    0.003680]   7 disabled
[    0.003681]   8 disabled
[    0.003681]   9 disabled
[    0.004079] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.004257] total RAM covered: 16334M
[    0.004748]  gran_size: 64K 	chunk_size: 64K 	num_reg: 10  	lose cover RAM: 246M
[... 3136 lines cut ...]
[   12.342517] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342523] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342529] RTW: 0x100: DA 0B 12 B8  C1 67 02 1C  BF CE 35 B8  52 09 03 52  
[   12.342536] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[   12.342542] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[   12.342548] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342554] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF  
[   12.342578] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342584] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342590] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342596] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342602] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[... cut ...]
[   12.342644] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342650] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342656] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342662] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342668] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342674] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   12.342680] RTW: 0x260: FF FF FF FF  FF FF

nickpitlosh, 2020-09-23:
nick@zotac:~/88x2bu$ vim Makefile 
nick@zotac:~/88x2bu$ sudo ./dkms-remove.sh 

-------- Uninstall Beginning --------
Module:  88x2bu
Version: 5.8.7.2
Kernel:  5.8.0-18-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

88x2bu.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.8.0-18-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod.........

DKMS: uninstall completed.

------------------------------
Deleting module version: 5.8.7.2
completely from the DKMS tree.
------------------------------
Done.
dkms-remove.sh was successful.
nick@zotac:~/88x2bu$ sudo ./dkms-install.sh 
Copying driver to: /usr/src/88x2bu-5.8.7.2

Creating symlink /var/lib/dkms/88x2bu/5.8.7.2/source ->
                 /usr/src/88x2bu-5.8.7.2

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.8.0-18-generic -j3.................................................................
cleaning build area...

DKMS: build completed.

88x2bu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.8.0-18-generic/updates/dkms/

depmod....

DKMS: install completed.
dkms-install.sh was successful.

nickpitlosh, 2020-09-23:
```
nick@zotac:~$ ping google.com
PING google.com (172.217.5.78) 56(84) bytes of data.
64 bytes from lax17s15-in-f78.1e100.net (172.217.5.78): icmp_seq=1 ttl=118 time=24.9 ms
64 bytes from lax17s15-in-f14.1e100.net (172.217.5.78): icmp_seq=2 ttl=118 time=27.0 ms
64 bytes from lax17s15-in-f14.1e100.net (172.217.5.78): icmp_seq=3 ttl=118 time=22.2 ms
64 bytes from lax17s15-in-f14.1e100.net (172.217.5.78): icmp_seq=4 ttl=118 time=28.5 ms
64 bytes from lax17s15-in-f14.1e100.net (172.217.5.78): icmp_seq=5 ttl=118 time=21.6 ms
64 bytes from lax17s15-in-f14.1e100.net (172.217.5.78): icmp_seq=6 ttl=118 time=23.7 ms
64 bytes from lax17s15-in-f14.1e100.net (172.217.5.78): icmp_seq=7 ttl=118 time=29.7 ms
64 bytes from lax17s15-in-f14.1e100.net (172.217.5.78): icmp_seq=8 ttl=118 time=28.4 ms
^C
--- google.com ping statistics ---
8 packets transmitted, 8 received, 0% packet loss, time 7008ms
rtt min/avg/max/mdev = 21.629/25.744/29.660/2.873 ms
nick@zotac:~$ sudo dmesg
[sudo] password for nick: 
[    0.000000] microcode: microcode updated early to revision 0x21, date = 2019-02-13
[    0.000000] Linux version 5.8.0-18-generic (buildd@lgw01-amd64-057) (gcc (Ubuntu 10.2.0-5ubuntu2) 10.2.0, GNU ld (GNU Binutils for Ubuntu) 2.35) #19-Ubuntu SMP Wed Aug 26 15:26:32 UTC 2020 (Ubuntu 5.8.0-18.19-generic 5.8.4)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.8.0-18-generic root=UUID=5f8b429a-4db0-49f0-97e5-db0d39df21c3 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000040003fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040004000-0x0000000040004fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040005000-0x00000000cbcf4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cbcf5000-0x00000000cbd37fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000cbd38000-0x00000000cc428fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cc429000-0x00000000cc63efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cc63f000-0x00000000dc315fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc316000-0x00000000dc615fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dc616000-0x00000000dc740fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc741000-0x00000000dca5efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dca5f000-0x00000000dcffefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dcfff000-0x00000000dcffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dd000000-0x00000000df9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041f5fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: ZOTAC ZBOX-ID18/ZBOX-ID18, BIOS B211P010 01/24/2014
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1496.517 MHz processor
[    0.003635] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003638] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.003651] last_pfn = 0x41f600 max_arch_pfn = 0x400000000
[    0.003658] MTRR default type: uncachable
[    0.003659] MTRR fixed ranges enabled:
[    0.003661]   00000-9FFFF write-back
[    0.003662]   A0000-BFFFF uncachable
[    0.003664]   C0000-DBFFF write-protect
[    0.003665]   DC000-E7FFF uncachable
[    0.003666]   E8000-FFFFF write-protect
[... 1228 lines cut ...]
mac_hid                16384  0
nick@zotac:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp2s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 00:01:2e:4d:b8:1d brd ff:ff:ff:ff:ff:ff
3: wlx1cbfce35b852: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 1c:bf:ce:35:b8:52 brd ff:ff:ff:ff:ff:ff
[... cut ...]
    inet6 fe80::d6cd:ae13:a812:313f/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
4: wlx1cbfce354e4c: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 1c:bf:ce:35:4e:4c brd ff:ff:ff:ff:ff:ff
nick@zotac:~$ ip route list
default via 192.168.1.1 dev wlx1cbfce35b852 proto dhcp metric 600 
169.254.0.0/16 dev wlx1cbfce35b852 scope link metric 1000 
192.168.1.0/24 dev wlx1cbfce35b852 proto kernel scope link src 192.168.1.27 metric 600 
nick@zotac:~$ 

```

nickpitlosh, 2020-09-23:
the only real change here is a reboot and reinstall with the powersaving disabled.

for other reasons though, I'm blowing away this installation tonight and changing over to efi boot.

During the collection of this for you i thought that the issue could have been caused by a previous firmware from another driver i tried, but after deleting it and replicating the issue still, i think there may be another problem. 

your thoughts are appreciated.

nickpitlosh, 2020-09-23:
also sorry about not giving you more details initially, ive never asked for any support on github before and this is the "first" issue with this particular project.

the real issue here is that wifi is a black hole of proprietary intrigue and fuckery though.

morrownr, 2020-09-23:
Thanks for the information. I would appreciate knowing the specific wifi adapter and ap/router you are using as well as that might help in the research of the issue.

nickpitlosh, 2020-09-23:
its a no name brand from amazon that literally has no brand on the box.
the chip is a rtl8812bu

is that helpful?

nickpitlosh, 2020-09-23:
this is the item link https://www.amazon.com/gp/product/B07RBRM3Y6/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1

morrownr, 2020-09-23:
Thanks for the info. I have had some time to do some research this afternoon. I'm seeing messages that indicate some other Realtek drivers have seen this problem. I don't have an explanation or a patch yet but will continue searching as I have time. If you can compile with CONFIG_POWER_SAVING = n until such time as a fix is available, that would be great.

morrownr, 2020-10-19:
A new version of the driver is available (5.8.74). There are some really big changes. I'd be interested to know if you see better results from this new version.

morrownr, 2020-10-23:
Issue closed due to new version available.

closed 2020-10-23 by morrownr