morrownr/88x2bu #2: rtl8812bu Compiling error

Nordlendingen, 2020-09-24:
kali:~/Downloads/88x2bu-master$ uname -r
5.8.0-kali1-amd64

kali:~/Downloads/88x2bu-master$ uname -v
#1 SMP Debian 5.8.7-1kali1 (2020-09-14)

kali:~/Downloads/88x2bu-master$ sudo ./dkms-install.sh
Copying driver to: /usr/src/88x2bu-5.8.7.2
Error! DKMS tree already contains: 88x2bu-5.8.7.2
You cannot add the same module/version combo more than once.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.8.0-kali1-amd64 -j3...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.0-kali1-amd64 (x86_64)
Consult /var/lib/dkms/88x2bu/5.8.7.2/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.8.0-kali1-amd64 -j3...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.0-kali1-amd64 (x86_64)
Consult /var/lib/dkms/88x2bu/5.8.7.2/build/make.log for more information.
An error occurred while running dkms-install.sh.

error when compiling the driver. is there because im running kernel 5.8?

morrownr, 2020-09-24:
This driver should be good with kernel 5.8. I have been testing it with the Ubuntu 20.10 daily builds (kernel 5.8). I have Kali installed on a test system so I will take a look later this morning.

morrownr, 2020-09-24:
Test on Kali complete. Result: Successful. Details: I started the test by updating the system. Then I rebooted. I then ran "sudo ./dkms-remove.sh" to remove any driver I had installed previously. Then i rebooted again. Then I ran "sudo ./dkms-install.sh" to install the driver as it currently is in my repo. Then I rebooted again and everything worked.

In the information you provided, I see the following:

"Error! DKMS tree already contains: 88x2bu-5.8.7.2
You cannot add the same module/version combo more than once."

I suspect the failure to install was because you already had a version of the driver installed. Recommend you step through the process as I outlined above. Regards.

Nordlendingen, 2020-09-24:
I tried now with a fresh installation with kali. and i still have trouble to get it to install properly. 

[make.log](https://github.com/morrownr/88x2bu/files/5276888/make.log)

closed 2020-09-24 by Nordlendingen

morrownr, 2020-09-24:
Something must be missing from the build environment. With Ubuntu, Mint and others based on Ubuntu if you install dkms, everything needed for the build environment is installed. Kali is different. Did you install the kernel header files?

nickpitlosh, 2020-09-25:
bc is not installed according to the make log.