morrownr/88x2bu #3: Driver Fails to Load

PengDre, 2020-09-26:
I purchased a WiFi USB dongle which the seller claims uses the RTL8812BU chipset - https://www.amazon.com/gp/product/B08BYBL2TQ/ .

It shows up using lsusb as:  Bus 001 Device 016: ID 0bda:b812 Realtek Semiconductor Corp. 

Kernel:  Linux version 5.4.0-48-generic (buildd@lcy01-amd64-023) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)
) #52~18.04.1-Ubuntu

Since the driver is not loaded, it does not show up for iwconfig or ifconfig.

I followed the install instructions and rebooted.  When plugging the device in, numerous errors and warnings are added to the syslog, please see attached.
[syslog.txt](https://github.com/morrownr/88x2bu/files/5285886/syslog.txt)

Could be an issue with the backport-iwlwifi package ...

morrownr, 2020-09-26:
Can I get you to run the dkms-remove.sh script per the readme so as to clean out the previous installation? Reboot after the script is finished.

Once that is complete download and try again. Your log file may indicate a problem with power saving. I have changed the setting in the makefile so let's see what happens.

PengDre, 2020-09-28:
Using the Sep. 30 source code, I still get errors/warnings.  Log file attached.  I did not reboot - the whole idea of a modular kernel is that you don't need to do that :) - though I did forcibly unload the prior 88x2bu module using 'rmmod' prior to rebuilding.
[syslog.txt](https://github.com/morrownr/88x2bu/files/5291267/syslog.txt)

morrownr, 2020-09-28:
I was working on the code last night and reverted the change I made so as to test it and some other things. Right now there is a line in Makefile like this:

CONFIG_POWER_SAVING = y

Use a text editor and change the "y" to an "n" before compiling and let's see what happens.

If that does not work, have you considered turning the internal wifi off, if that is possible? It should work with both functioning as I have multiple devices on some systems but we need to rule some things out.

FYI: I have been researching the issue but I have not found a solution yet. I'll bet that we find a solution.

FYI: I understand the theory behind modules, however, I am working to make the installation of this driver something that non-techies can do. Techies are welcome to do as they please.

PengDre, 2020-09-28:
OK, made that change to the Makefile.  I am attaching the syslog output with both the other wireless enabled (syslog_on.txt) and disabled (using 'ifconfig wlo1 down') (syslog_off.txt), making sure to unload the 88x2bu module between tests.  It fails again in both cases and there doesn't seem any significant difference in the log files - the diff (after removing the kernel timestamps) of the two is also attached (syslog_diff.txt). 

Maybe to help things along, I am a quite good C coder and fairly familiar with Linux systems, though I don't have experience coding the kernel. 

[syslog_diff.txt](https://github.com/morrownr/88x2bu/files/5295019/syslog_diff.txt)
[syslog_off.txt](https://github.com/morrownr/88x2bu/files/5295020/syslog_off.txt)
[syslog_on.txt](https://github.com/morrownr/88x2bu/files/5295022/syslog_on.txt)

morrownr, 2020-10-08:
PengDre, sorry for the delay. I've been a little busy.

I've looked things over again. Is your system in Secure Mode? If so, try turning Secure Mode off and see what happens.

PengDre, 2020-10-08:
Hi, 

I don't mean to be uncooperative but that's a big pain, I would have to reboot a number of times (as I can only change that option on my computer in Windows).  The kernel loads the module, wouldn't that fail if the module is not properly signed?  Wouldn't there be a syslog entry about that?  I don't mind helping test but this seems like a waste of time.  Maybe adding more debugging statements to the code, or turning more on?

morrownr, 2020-10-23:
Issue closed due to new version available.

closed 2020-10-23 by morrownr