morrownr/88x2bu #4: Bless you morrowrn !!

N9VV, 2020-09-28:
I don't know how to send the author a note so I am posting here.
I have an AC1200 and it is working like GANGBUSTERS thanks to your git for 88x2bu
862mb/s (5Ghz) -- :+1:

morrownr, 2020-09-28:
Leaving a note here is just fine. Knowing what works is important also. If you would be so kind as to answer a few questions, it would add to our knowledge:

What distro and version of Linux are you using?

What computer are you using the adapter with?

Is this adapter the only wifi in the system?

Is there anything that could be expanded or improved in the Readme file?

Thanks,

Nick

Escentaphraktos, 2020-11-10:
Hello morrownr! I also just want to thank you for the driver. your driver is the only one that currently works for me. Although I do want to ask if there was a way to change some advanced settings of the wifi adapter? like how one can change the settings on the advanced tab of the wifi adapter on windows through devmgmt where we can change settings like enabling vht, disabling 802.11b, channel width, power savings etc. I want to get as much performance as I can from the wifi adapter, but only if possible :)

Generic No Brand AC1200mbps Wifi Adapter/ RTL 8812BU
Pop!_OS 20.04 LTS x86_64 
Acer Aspire A315-42G V1.10 2019 Model
Kernel 5.8.0-7625-generic
I disabled the internal wifi module which is the Atheros QCA9377, it's just terrible.
And as for the readme, well just the capability of changing advanced settings that I mentioned above cos I want the lowest ping possible on csgo xd
Thanks once again man.

morrownr, 2020-11-10:
Hello Escentaphraktos,

"where we can change settings like enabling vht, disabling 802.11b, channel width"

Those are settings that you need to change in your wifi AP/wifi router.

"power savings etc."

This setting can be changed in the driver but it is also a setting that can be changed in many wifi routers. To change the setting in the driver:

Remove the driver using the uninstallation instructions then reboot.

Before reinstalling, edit the following file:

Makefile

Only use a Linux/Unix compatible editor.

As I write this you are looking for line 59:

CONFIG_POWER_SAVING = y

Change the "y" to an "n" and save, then close.

Following installation instructions.

morrownr, 2020-11-10:
One last note on Power Saving: While Power Saving has been problematic in the past at times on Linux on some systems, this driver handles Power Saving very well and based on my benchmarking, it doesn't seem to slow things down.

morrownr, 2020-11-10:
Concerning your remark about this being the only driver that works for you... this doesn't surprise me as this driver is based on code that has been dramatically improved by Realtek. Look at the benchmarks in the README. This driver is FAST, feature rich and stable!

closed 2020-12-03 by morrownr