morrownr/88x2bu #5: Wifi Packet Injection

KeerthanGopi, 2020-10-04:
Does it support wifi packet injection?

morrownr, 2020-10-04:
Unknown. I haven't tested that capability but am willing to do so if you provide me with an outline of how to test the capability. A link to an article that shows how to set up and test the capability would work.

KeerthanGopi, 2020-10-05:
https://null-byte.wonderhowto.com/how-to/check-if-your-wireless-network-adapter-supports-monitor-mode-packet-injection-0191221/

morrownr, 2020-10-07:
I had time this afternoon to read the article you posted. That is a somewhat dated article but I gave it a try. Here is what I found: The aircrack-ng (airmon-ng) package in the Ubuntu 20.04 repos is very dated and caused nothing but problems. That is to be expected with the networking subsystem in Linux over the last few years. I went to the Aircrack-ng site and it appears a lot of work has been done but it seems downloading and compiling the source is the only option and the required dev setup would require me to put in a lot of work to modify things here. If you can find a current version available as a .deb, let me know where it is and I will try to set aside some time to continue testing.

KeerthanGopi, 2020-10-08:
Which version did you install?
I found .deb file for aircrack-ng 1.6 here:
https://packages.ubuntu.com/focal/aircrack-ng

KeerthanGopi, 2020-10-08:
This is version 1.6:
https://packages.ubuntu.com/focal/aircrack-ng
Is that what you installed?

morrownr, 2020-10-08:
I just checked on my Mint 20 test box and Aircrack-ng is indeed v1.6. My bad. In previous testing, the old version was what was available and when I ran into the same problems, I assumed it was the old version. My bad. Let me take another look at I have time. Have you tested?

KeerthanGopi, 2020-10-10:
I am currently in a situation where i cannot test and i am relying on your test for my use.

morrownr, 2020-10-11:
I understand. I'll take a closer look once I have taken care of some priority issues.

KeerthanGopi, 2020-10-23:
Have you tested?

morrownr, 2020-11-06:
I had time today to address the issue of packet injection. I need testers to tell me if it is working.

morrownr, 2020-11-07:
I was able to test Packet Injection today by using the new instructions in the README. Result:

"Injection is working!"

I have no experience pen testing so I could really use some experienced pen testers to see what they think.

KeerthanGopi, 2020-11-09:
Thanks @morrownr.

KeerthanGopi, 2020-11-09:
Can you show the output below " Injection is Working"?
Does it show a value more than 0?

morrownr, 2020-11-09:
To provide your answer I used the Packet Injection documentation in the README and during the process I discovered that the documentation was misleading. I have now reworked the documentation. Here is the output of my test:

15:38:31  ... $ sudo aireplay-ng --test wlan0
15:38:31  Trying broadcast probe requests...
15:38:31  Injection is working!
15:38:32  Found 1 AP

15:38:32  Trying directed probe requests...
15:38:32  8C:59:73:FE:8B:F5 - channel: 36 - 'myAPname'
15:38:32  Ping (min/avg/max): 0.826ms/4.058ms/6.667ms Power: -35.77
15:38:32  30/30: 100%

KeerthanGopi, 2020-11-12:
Thanks a lot @morrownr.

closed 2020-11-12 by KeerthanGopi