morrownr/88x2bu #6: Doesn't connect to network after reboot

arkur-code, 2020-10-09:
Hello. I have ubuntu 20.04 and after reboot my TP-link t4u not connecting to the network with authentication error. I have to delete the network settings (forget the network) and reconnect with the password.

morrownr, 2020-10-09:
Does it stay connected after you forget the network and reconnect?

arkur-code, 2020-10-09:
Yes

morrownr, 2020-10-09:
Keep me posted. In the near future, I am going to do some extensive testing to see if I can shake out any issues.

morrownr, 2020-10-13:
I was able to test this issue this evening. I tested all 4 drivers I maintain on this site. It seems to me that it is normal to have to "Forget Connection" if you remove one wifi adapter and connect another one that is going use the same router/ap. This may be needed if you remove a driver and install a new version as well. It appears this is normal operations but I will keep an eye open for similar problems that aren't normal.

arkur-code, 2020-10-13:
i'm sorry. I do not speak English well. I meant restarting the computer, not removing the adapter. Every time I reboot, I need to have to "Forget Connection"

morrownr, 2020-10-13:
Your English is very good. Habla espanol? Do you mind if I ask what your native language is?

Back to topic: Let me suggest you post this issue in the Networking and Wireless forum at: https://ubuntuforums.org/ There are several folks there that are really good at troubleshooting issues such as this. It is not normal to have to "Forget Connection" after each reboot so there is an issue but it is likely an issue with the networking subsystem or its configuration. If it is an issue with the driver, I would like to know so I can attempt to fix it.

arkur-code, 2020-10-15:
My native language is Russian.I asked  on the Russian forum of ubuntu. They said that there are solutions for 8192eu described https://github.com/Mange/rtl8192eu-linux-driver. I tried to use this commands, but it didn't works

arkur-code, 2020-10-15:
IСпасибо за ссылку на форум, я попробую там задать вопрос.

closed 2020-11-07 by morrownr