morrownr/88x2bu #7: Make file Err (-Wno-date-time)

m4lory, 2020-10-10:
Hi, trying to compile with included dkms-install script make “time and date” error while running make.

Sorry if is trivial issue to fix for savvy user, but might be good to know if unaware.

What I did was remove IF statement that check for version and make `EXTRA_CFLAGS += -Wno-date-time` always on.  This was near line 23 of included Make file.

I have to remove the existing folder /use/src/88x2bu-<ver> and run the script once more after apply change to Make file.

The driver has successful compiled on my old desktop with Kali.  Monitor mode work now.  Hope this help.

Unknow Hiro USB adapter, also.

morrownr, 2020-10-11:
I have one more driver that I need to finish the support for kernel 5.8 and then I am going to spend some time looking into reports like this. If you have additional information as times passed, add it to this report. I'm glad you got it going.

morrownr, 2020-10-13:
I had time to test this evening. I am not seeing this issue on Mint or Ubuntu. I do have an installation of Kali and will test when able. This may be specific to Kali.

morrownr, 2020-10-23:
Issue closed due to new version available.

closed 2020-10-23 by morrownr