morrownr/88x2bu #8: Kali Linux Raspberry Pi 4 - Can't bring interface up

nerotix, 2020-10-20:
I've been trying to get this driver up and running on my raspberry pi 4 for the last couple of hours but no luck so far. The raspberry pi is running Kali 2020.1 with the latest upgrades etc. Installing the driver seems to work just fine and no errors occur.

However, I can't seem the interface up, I've tried "ifconfig wlan0 up" and "ip link set wlan0 up", both times I get the error "SIOCSIFFLAGS: Operation not permitted" (yes I'm running it as root so sudo isn't needed).

Kernel version = 4.19.118-Re4son-v8l+

I've been checking my logs but can't find anything related to the issue.

Hope someone can assist me in this matter.

morrownr, 2020-10-21:
I have a Pi 3 and Pi 4 but have not found a need to test this driver. When I have time, I might work on this but your best bet might be in the Kali forums.

PieGuy314, 2020-11-05:
> I've been trying to get this driver up and running on my raspberry pi 4 for the last couple of hours but no luck so far. The raspberry pi is running Kali 2020.1 with the latest upgrades etc. [...]

Might I ask how you've compiled the driver for the Raspberry Pi 4? Have quickly looked at the Makefile and can't see any Pi related options.

morrownr, 2020-11-06:
I had time to add the appropriate lines in the Makefile today. I will try to document it as soon as I have time.

PieGuy314, 2020-11-06:
> I had time to add the appropriate lines in the Makefile today. I will try to document it as soon as I have time.

Looks straightforward. I was previously using another repo for the rtl8812bu and had to add 'EXTRA_CFLAGS += -fno-stack-protector' to get in to compile on arm64. 

Thanks so much for the speedy reply.

PieGuy314, 2020-11-06:
-fno-stack-protector needed.

''' bash
ERROR: "__stack_chk_guard" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!
'''

morrownr, 2020-11-07:
PieGuy314,

Thanks for the help with the Makefile. I have added the line you suggested.

Could I get you to document the steps needed to compile on RasPi ARM and ARM64? It would help me get the documentation in the README much faster.

I have a Pi 3B and Pi 4B so I have the hardware to test but both systems are busy right now. Hopefully, at some point in the next weeks I can do some testing with them. I also have 2 more driver to add RasPi support and Packet Injection support to, That will keep me busy for a while.

PieGuy314, 2020-11-07:
> Could I get you to document the steps needed to compile on RasPi ARM and ARM64? It would help me get the documentation in the README much faster.

Here goes...

'''sudo apt-get install -y bc build-essential dkms git raspberrypi-kernel-headers
git clone https://github.com/morrownr/88x2bu.git
cd 88x2bu
sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC=n/g' Makefile
'''
For armv6/v7 (Raspberry Pi Zero, 1, 2)...
'''sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
'''
For armv8 (Raspberry Pi 3, 4)...
''' sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
'''
Then...
''' sudo ./dkms-install.sh
'''
If everything goes well, reboot. If not consult the build logs.

morrownr, 2020-11-09:
Thanks. I'm working on adding the docs and I hope to test on my RPI 4B soon.

closed 2020-11-16 by morrownr