morrownr/88x2bu #10: TP-Link AC1300 (T4U v3) not working on Ubuntu 20.10

staticdev, 2020-10-25:
I did the installation, reboot and still not working.

Some outputs that might be useful:

```
$ sudo modprobe 88x2bu
# outputs nothing - that is a good sign since it normally gives an error

$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1c4f:0034 SiGma Micro XM102K Optical Wheel Mouse
Bus 001 Device 003: ID 2a7a:938f  CASUE USB KB
Bus 001 Device 008: ID 04e8:6864 Samsung Electronics Co., Ltd GT-I9070 (network tethering, USB debugging enabled)
Bus 001 Device 005: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ sudo iw dev
# outputs nothing
```

Thanks.

morrownr, 2020-10-25:
This should be fixed now. Please use the ./dkms-remove.sh script to remove the version of the driver you have now and then reboot. Delete the version you downloaded and download again. Follow the instructions and it should work this time.

Why? The new version of the driver that I very recently posted has many changes and additions relative to the previous version so I could not simply merge a few changes. I had to start with the new version of the driver and then work toward adding the various things I have done to improve the driver. Something I had not done yet was add the supplemental device IDs and yours was one of them. This means that the driver simply did not know if was supposed to support your device. It does now.

Thanks for the report. This is how we make things better. Let us know how it goes.

staticdev, 2020-10-25:
@morrownr I had to reinstall Ubuntu, and installed this new version (5.8.7.4). Things are much better now.. but still with a problem.

With the new version, my TP-Link AC1300 is now recognized, I can see and authenticate the networks, but when I enter the browser or anything, I can't use the connection.

New outputs:

```sh
$ sudo modprobe 88x2bu
# outputs nothing - that is a good sign since it normally gives an error

$ lsusb
Bus 002 Device 002: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1c4f:0034 SiGma Micro XM102K Optical Wheel Mouse
Bus 001 Device 003: ID 2a7a:938f  CASUE USB KB
Bus 001 Device 007: ID 04e8:6864 Samsung Electronics Co., Ltd GT-I9070 (network tethering, USB debugging enabled)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ sudo iw dev
phy#0
	Interface wlxd037458a60d1
		ifindex 3
		wdev 0x1
		addr d0:37:45:8a:60:d1
		ssid ZiggoFBB1763
		type managed
		txpower 18.00 dBm
```

staticdev, 2020-10-25:
@morrownr UPDATE: after one more restart, the internet started working.

PR #11 (2020-10-25): Removed beta from 20.10 and sort tested distributions
Closes #10
closed 2020-10-25 by morrownr