morrownr/88x2bu #12: TP-Link Archer T3U v1 not working in Debian

fschlaef, 2020-10-25:
Hello,

I am unable to make this driver work in Debian.

System configuration :

Debian 11 (bullseye), fully updated, kernel 5.9.0.1
Driver version : 5.8.7.4 (I saw https://github.com/morrownr/88x2bu/issues/10 and tried to delete everything and reinstall but no luck)

The wireless card is recognized by `iw dev` 

```
phy#9
        Interface wlxd03745b1a9ba
                ifindex 12
                wdev 0x900000001
                addr d0:37:45:b1:a9:ba
                type managed
                txpower -100.00 dBm
```
Also by `iwconfig`

```
wlxd03745b1a9ba  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

`lsusb` also shows it 

```
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 2: Dev 11, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 5000M
```

But the device will not connect despite correct configuration used in `/etc/network/interfaces`

When I use `wpa_supplicant` I get this :

```
Successfully initialized wpa_supplicant
nl80211: kernel reports: Authentication algorithm number required
```

Using `wpa_cli`

````
Selected interface 'wlxd03745b1a9ba'

Interactive mode

> scan
OK
<3>CTRL-EVENT-SCAN-FAILED ret=-1
```

Note that I succeeded once with scanning but at the last step (`enable_network`) it threw a firmware error and errored out. Sadly I did not copy the message.

Let me know if you need more information !

morrownr, 2020-10-25:
@fschlaef Let me throw out some ideas:

- I saw where you said "tried to delete everything...". That has me concerned. dkms has big advantages but deleting stuff to get rid of a driver is not one of those advantages. Let's try this:

Clean (delete) everything from your ~/src folder. Download v5.8.7.4 again and follow the installation instructions again but don't run "sudo ./dkms-install.sh", do run "sudo ./dkms-remove.sh". Once that completes, do a cold reboot. Once that is complete and you start the system again, run this "sudo dkms status" and let's make sure dkms does not show any installed driver. If there is no driver shown, go ahead and run "sudo ./dkms-install.sh" and reboot.

i would also like to see the device ID. Can you run "lsusb" and post the results?

fschlaef, 2020-10-25:
@morrownr I should have explained what I did better but I did in fact run ./dkms-remove.sh and rebooted, and also rebooted after every attempt to add / remove a driver, be it though dkms or manually.

I tried version 5.8.7.2 and it works great in USB 2.0 mode but instantly crashes in USB 3.0 mode too. I am starting to believe my motherboard is in fact the problem, it is an old GA-P55 from 2009 with a Renesas/NEC uPD720200 and USB 3 was terrible back then, even in Windows.

Here is the complete syslog of the driver crash, containing login, attempting to transfer a file over Samba, crash, and reboot : [88x2bu_crash.txt](https://github.com/morrownr/88x2bu/files/5435915/88x2bu_crash.txt)

I reinstalled 5.8.7.4 and it works great in USB 2.0 mode.

`lsusb` shows this :

```
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 011 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 010 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 2357:012d TP-Link Archer T3U [Realtek RTL8812BU]
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

morrownr, 2020-10-25:
@fschlaef I am aware that there have been compatibility issues with early versions of USB 3. In fact, I had to think long and hard about setting things up turn on USB 3 by default. It is possible to compile the driver to automatically try to change from USB 2 to USB 3 mode but I thought leaving it at its default, which is to change nothing, might be a good idea so I went with the external file. Maybe I should default the external to 0, change nothing, and change the docs to let folks know that they can change one setting in that file to test USB 3. What do you think is best?

fschlaef, 2020-10-26:
@morrownr I think my case is very much an edge case, I don't know how many people use your driver in conjunction with decade-old hardware like I am doing.
Maybe you could leave USB 3 on by default but search for known buggy USB 3 controller like mine and then revert to USB 2. But then again, this chipset was included in tons of motherboards/PCIe cards in the early 2010s since it was the first USB 3 controller, so maybe it is just my specific model bugging out. I should have added that despite having 2 USB 3.0 ports on that motherboard, only one of them is recognized as such by Linux, for reasons unknown.

I say don't bother too much, and maybe just add a warning in documentation regarding old USB 3 controllers.

morrownr, 2020-10-26:
After thinking about this, I decided to continue installing the file but I changed the default to no change. I changed the README to make it clear how to turn on USB 3 support. We'll see how this works.

I have hardware in use that is more than a decade old. One of my two dedicated test systems has no USB 3 support but it does have a dual core 64 bit Intel processor running at 3.0 GHz and 4 gb of ram. It runs well with today's software. Thanks for the report and suggestions.

fschlaef, 2020-10-26:
Thanks for your help ! I am closing this issue since there is in fact no problem with Debian after all 😃

closed 2020-10-26 by fschlaef