morrownr/88x2bu #9: Driver not working

fo4w46r873, 2020-10-25:
Hi, i've problem with driver, i followed the instruction and after reboot nothing changed, so i check that the driver was loaded with lsmod, but it wasn't there, so i loaded manually with modprobe but also nothing

wifi stick: TP-Link Archer T3U Plus with 8822bu chipset
distro: xubuntu-20.04

[image]

morrownr, 2020-10-25:
The first two things that come to mind are:

1. It could be an issue with Secure Mode. If Secure Mode is turned on and dkms was unable to sign the driver during the installation process, this is exactly what you would see.

2. If this is a laptop, it could be that the button or key combination that toggles wifi on and off is set to off. This would also cause exactly what you are seeing.

One of the test systems I use is set to Secure Mode and dkms does a good job of signing the driver. One time, to test, I messed the keys up and what you are seeing is exactly what I saw. In my humble opinion, Secure Mode is probably unnecessary for the individual and home office so my systems all have it turned off with the exception of one test system.

Let us know if this helps.

fo4w46r873, 2020-10-25:
Secure boot is disabled
[image]
and this is not a laptop

morrownr, 2020-10-25:
Humble apologies. The newly updated driver should work for you now. Disregard my previous post. Post #10 caused me to go over the device IDs section in the appropriate file and I saw where I still needed to pull information from the previous version... here is the reply to #10:

This should be fixed now. Please use the ./dkms-remove.sh script to remove the version of the driver you have now and then reboot. Delete the version you downloaded and download again. Follow the instructions and it should work this time.

Why? The new version of the driver that I very recently posted has many changes and additions relative to the previous version so I could not simply merge a few changes. I had to start with the new version of the driver and then work toward adding the various things I have done to improve the driver. Something I had not done yet was add the supplemental device IDs and yours was one of them. This means that the driver simply did not know it was supposed to support your device. It does now.

Thanks for the report. This is how we make things better. Let us know how it goes.

fo4w46r873, 2020-10-27:
Thanks, now it's work

closed 2020-10-27 by fo4w46r873