morrownr/88x2bu #13: Very noisy logging

nicokaiser, 2020-10-27:
The module works fine, installed via DKMS on Ubuntu Server 20.10 on arm64, thanks for the great job!

However when connected, the module writes very much status information as "warning" messages to the kernel syslog:

```
[ 3766.842132] RTW: rtw_set_ps_mode(wlxd03745178c68) Leave 802.11 power save - WIFI-LPS_CTRL_PHYDM
[ 3766.842154] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): HW port id=0
[ 3766.842167] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
[ 3766.848216] RTW: rtw_set_ps_mode(wlxd03745178c68) Enter 802.11 power save - WIFI-LPS_CTRL_PHYDM
[ 3766.848231] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): HW port id=0
[ 3766.848243] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
[ 3768.858168] RTW: Turbo EDCA =0x5ea42b
[ 3768.858293] RTW: rtw_set_ps_mode(wlxd03745178c68) Leave 802.11 power save - WIFI-LPS_CTRL_PHYDM
[ 3768.858309] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): HW port id=0
[ 3768.858322] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
[ 3768.864341] RTW: rtw_set_ps_mode(wlxd03745178c68) Enter 802.11 power save - WIFI-LPS_CTRL_PHYDM
[ 3768.864358] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): HW port id=0
[ 3768.864371] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
[ 3770.874182] RTW: Turbo EDCA =0xa630
[ 3770.874307] RTW: rtw_set_ps_mode(wlxd03745178c68) Leave 802.11 power save - WIFI-LPS_CTRL_PHYDM
[ 3770.874322] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): HW port id=0
[ 3770.874336] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
[ 3770.880167] RTW: rtw_set_ps_mode(wlxd03745178c68) Enter 802.11 power save - WIFI-LPS_CTRL_PHYDM
[ 3770.880181] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): HW port id=0
[ 3770.880192] RTW: rtl8822b_set_FwPwrMode_cmd(wlxd03745178c68): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
```

Most of the messages are power save mode changes (which are ok, but do not need to be stored as warning in the syslog) every 2 seconds and "Turbo EDCA" messages every few seconds.

Is there a way to disable this verbose logging, or is this hardcoded in the Realtek driver?

morrownr, 2020-10-27:
I've been pondering this very issue. Logging plenty of information to check function is a good thing but I'm not seeing any problems here so let me take a look later today to see if a different log level is appropriate.

You do have control: Open the file called Makefile in a text editor and change the follow line:

From: CONFIG_RTW_DEBUG = y

To: CONFIG_RTW_DEBUG = n

Then remove the driver, reboot and run the installation again.

morrownr, 2020-10-28:
I have added the ability to adjust log level in the 88x2bu.conf file. I have not documented it in the README yet but I did document it in 88x2bu.conf. I set the default to 0. Users can turn on logging as required. These changes have been pushed so feel free to remove and delete the old version and test this version.

nicokaiser, 2020-10-28:
That's awesome! Thank you very much, changing the loglevel to a low value completely solves the problem!

closed 2020-10-28 by nicokaiser