morrownr/88x2bu #15: Typo in Makefile (CONFIG_PLATFORM_ARM_RPI)

PieGuy314, 2020-11-09:
[https://github.com/morrownr/88x2bu/blob/e56ade8f7d9cf439cc1dd04f2a9278b682b9ba4d/Makefile#L1339](https://github.com/morrownr/88x2bu/blob/e56ade8f7d9cf439cc1dd04f2a9278b682b9ba4d/Makefile#L1339)

I think the ARCH definition for CONFIG_PLATFORM_ARM_RPI should be:
``` bash
ARCH ?= arm
```
The definition for CONFIG_PLATFORM_ARM64_RPI is correct.

PieGuy314, 2020-11-09:
Just as a side, noticed you've added some notes regarding compiling for Raspberry Pi in README.md. Probably worth making the point that CONFIG_PLATFORM_ARM64_RPI if for 64-bit distros.

morrownr, 2020-11-09:
PieGuy314,

I really appreciate the help from you and choogenboom to get this RasPi support ironed out. I have no idea how that "arm64" made its way into a place where it should have read "arm". I have made the correction. Thanks.

Now for some serious discussion about something that is unclear to me. You seem to be under the impression that ARM is for RasPi OS 32 bit and ARM64 is for RasPi OS 64 bit. I am under the impression that ARM is for RasPi boards that have 32 bit processors and ARM64 is for RasPi boards that have 64 bit processors (regardless of whether the OS is 32 or 64 bit.) This is something that can be sorted out with testing and I think I am going to try to move the load off of my RasPi 4B so I can use it to test in an effort to get to the bottom of this. Let me know if you have any information that helps get to the bottom of this issue so we can make it clear in the README.

PieGuy314, 2020-11-09:
https://www.raspberry-pi-geek.com/Archive/2017/23/Operating-the-Raspberry-Pi-3-in-64-bit-mode

morrownr, 2020-11-10:
Okay, so with 32 bit OS, use ARM and with 64 bit OS, use ARM64. That makes sense...what was I thinking?

PieGuy314, 2020-11-10:
Think of it less like thinking and more like learning. Learning stuff makes you think. The dev community need people like you who will make 'things' more accessible for those who lack the capability or inclination. I group myself in the latter. I just want stuff to work. But I'm still savvy enough to realise that I don't have to pay a premium for closed source software/hardware to allow me to enjoy the things I do.

morrownr, 2020-11-11:
What helps me is people like you that are willing to test and offer suggestions for the README. The work on these 3 drivers is very time consuming.

FYI: During my work on these drivers, I have found that there is really only one seller on Amazon that does Linux well and that is Brostrend. To properly support Linux when it comes to out of kernel drivers, the drivers have to constantly follow kernel development and be updated as needed. Most sellers on Amazon simply stick a driver on a cd or on their website and leave it...and it dies from bit rot... won't even compile on the next LTS version of the kernel.

closed 2020-11-16 by morrownr