morrownr/88x2bu #16: INFO: Raspberry Pi Driver Compatibility

PieGuy314, 2020-11-09:
Can report successfully building and running the current driver on the following Raspberry Pi/OS versions:

Raspberry Pi 4B (armv8) - DietPi 64-bit, PiOS 32-bit

Raspberry Pi 3B (armv8) - DietPi 32-bit, DietPi 64-bit

Raspberry Pi 2B (armv7) - PiOS 32-bit

Raspberry Pi A+ (armv6) - PiOS 32-bit
- Swapfile increased to 512MB for compilation.
- Compilation time circa 4 hrs!

Also have a Pi Zero - but will get to that when time permits and update this post accordingly.

Can also state that AP mode, packet injection and USB3.0 mode work as expected on a Pi 4B running 32-bit PiOS.

morrownr, 2020-11-09:
PieGuy314,

I really appreciate the report. It is good to see things working well on the various versions of the Raspberry Pi. I am going to start using my Pi 4B in a way that I can test this and my other drivers. As soon as I can, I will start moving the Pi support and packet injection support to my other drivers: github.com/morrow

So, 4 hours on the A+. Cool. Glad it didn't error out.

Thanks, Nick

PieGuy314, 2020-11-10:
> So, 4 hours on the A+. Cool. Glad it didn't error out.

It did the first time (didn't crash but ground to a halt). Was watching the memory usage the second (load avg > 8 with no swap space left). Third time lucky.

Managed to compile the driver on the Pi Zero (circa 45 mins). Insufficient power to drive the Pi and adapter without a powered USB hub. Kinda makes it too clunky for any practicable use. 

Have now finally fixed all of the issues and niggles I was having with the Pi 3B I run shairport-sync on to stream tunes via my hi-fi.

Have a couple of FriendlyElec ARM SBC's that could benefit from some 88x2bu love. Guess that's my next project. 

Thanks for the good work you've done in getting this driver up to date. 

Regards, -- Ade

PieGuy314, 2020-11-10:
BTW, Tested Hardware:

https://smile.amazon.co.uk/gp/product/B08523KPP9/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

https://smile.amazon.co.uk/gp/product/B084WBTTN6/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

The first one is an absolute bargain! Less than 1/3 of the price of the latter. No difference in performance.

morrownr, 2020-11-11:
The TP-Link adapter is already listed as supported but I will add the other to the README if that is okay with you. I am nearly ready to start testing on my RasPi 4B but I have a question: Did you use an external powered USB 3 extender or did you plug directly into the Pi? I'm concerned about having enough power available.

morrownr, 2020-11-12:
I now have an 8812bu based usb wifi adapter running on my RasPi 4B. It was good to step through the installation instructions myself. I gave it a try after only installing dkms as dkms requires the headers to be installed but I found out that "bc" is not installed and is required. This adventure is causing me to rethink how the installation is spelled out in the README. I need the process to be consistent, easy and dependable.

This device is working really well on the Pi. I'm not seeing any power issues so I think the 8812bu is not using a lot of power which is good.

PieGuy314, 2020-11-12:
I thought I'd listed the packages the build process depends on in another thread. The only board that wasn't able to drive the adaptor without a powered hub was the Pi Zero - but that could have been down to the funky enclosure it's in.

morrownr, 2020-11-12:
You did list the packages for the RasPi:

sudo apt-get install -y raspberrypi-kernel-headers bc build-essential

I am modifying to see what happens. I am not happy with the current installation instructions in the README. It needs to be simplified. I'm thinking a script for each option is the only way to make this work:

1. x86-64 - Ubuntu/Mint
2. ARM-32 - Raspberry Pi OS
3. ARM-64 - Raspberry Pi OS
4. x86-32 - Raspberry Pi  OS
5. ARM-64 - Ubuntu Desktop and Server for Raspberry Pi
6. ARM-32 - Ubuntu Server for Raspberry Pi

Do you see the problem?

We also need to add a section before installation that lists options for connecting a system to the internet without their 88x2bu adapter being available. Ethernet is an option but it is hard to move desktop systems. Tethering is an option but we need a good simple section that provides instructions for many options to connect to the internet in order to install this driver. Would you mind taking the lead on preparing a document that outlines the instructions for the various options?

PieGuy314, 2020-11-13:
From a personal perspective, I'm not keen on the thought of a multitude of build scripts. You've listed a few options but ignored a plethora of other systems/architectures. The 'dkms-install.sh' and 'dkms-remove.sh' save time and need little maintenance. I personally feel that editing the Makefile should be left to the user. The key for me is the documentation. It might be worth documenting how to disable the Pi's built in wifi, but in all honesty, that's already documented elsewhere. The additions you've made to 'README.md' make sense. I'd leave them as they stand and wait for questions that people raise. Just my 2c.

-- Ade

morrownr, 2020-11-13:
PieGuy314,

I think you are making sense so I am going to work on other things like making sure this driver works well on the next LTS version of the kernel (5.10). I do think a short note to point out how to disable the onboard wifi for the 3B and later would be a good idea. I also think a short primer on how to get a system connected to the internet via alternate means could be useful so if you run onto any good articles, please pass the info on.

FYI: I tested the new Ubuntu 20.10 ARM 64 Desktop distro yesterday. It lasted 30 minutes on an sd card and it was time to image something else. In my opinion, it is truly bad. The bad starts with the Gnome interface. I don't like that interface. I have tried to like it but I don't. My main OS here is Mint Cinnamon. Gnome is also slow. You can put lipstick on a pig but it is still a pig and Gnome is a pig. I'm not going to support that distro.

Cheers.

PieGuy314, 2020-11-13:
Only two GUI's I use - MacOS and a hacked Chromebook running Debian/Gnome (ie. Chromium, Nautilus, Terminal). Pretty much everything else is CLI via ssh. Installing OS's is usually via editing the boot info and/or a serial-USB gadget.

Spent too long today trying to get the driver to compile on a FriendlyElec ZeroPi (Allwinner H3 Cortex-A7 1.2 GHz | 4 cores | ARMv7). $9.99 4-core ARMv7l that way outperforms the Pi Zero. The included Makefile is inclined towards using cross-compilation for a lot of ARM based chipsets and I'm still trying to figure out why it stubbornly refuses to compile on a standalone system. But that's a project for the weekend.

BTW, to disable onboard wifi on the Raspberry Pi, download and compile the new driver with it enabled. Edit _/boot/config.txt_ and append:

``` bash
dtoverlay=disable-wifi
```
Poweroff. Plug in your 88x2bu adaptor. Reboot.
.

morrownr, 2020-11-14:
What kinds of problems are you seeing compiling on that FriendlyElec board?

morrownr, 2020-11-14:
iperf3 results using this driver on a Pi 4B:

[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.01   sec  47.9 MBytes   399 Mbits/sec    0   2.13 MBytes       
[  5]   1.01-2.00   sec  68.8 MBytes   581 Mbits/sec    0   2.38 MBytes       
[  5]   2.00-3.00   sec  66.2 MBytes   556 Mbits/sec    0   2.57 MBytes       
[  5]   3.00-4.00   sec  66.2 MBytes   556 Mbits/sec    0   2.72 MBytes       
[  5]   4.00-5.00   sec  62.5 MBytes   524 Mbits/sec    0   3.08 MBytes       
[  5]   5.00-6.00   sec  70.0 MBytes   587 Mbits/sec    0   3.08 MBytes       
[  5]   6.00-7.00   sec  66.2 MBytes   556 Mbits/sec    0   3.08 MBytes       
[  5]   7.00-8.00   sec  67.5 MBytes   566 Mbits/sec    0   3.08 MBytes       
[  5]   8.00-9.02   sec  65.0 MBytes   536 Mbits/sec    0   3.08 MBytes       
[  5]   9.02-10.00  sec  66.2 MBytes   566 Mbits/sec    0   3.08 MBytes 

That is an average of 542 Mbits/sec. Smokin'. I noticed a pretty good bump in speed with the current default version of this driver. All indications are that this is a pretty good driver. I'm also impressed with the EDUP adapter that I use. That Pi 4B can really push the bits out of that USB 3 port.      

FYI: I rewrote the installation part of the README and have been testing it. It is now 100% from the terminal (CLI) and all installation instructions are in one section.

PieGuy314, 2020-11-14:
> What kinds of problems are you seeing compiling on that FriendlyElec board?

Nothing that's not to be expected in hindsight. The OS's on a lot of these boards are usually cross compiled on x86 based systems. Subsequently, the Makefile reflects a similar build environment geared around cross-compilation. This is quite sensible.

I'd built the OS I was running via cross-compilation. The sensible approach is to rebuild it with the the driver included.

The less sensible approach may be to select an alternative ARMv7l build platform. I've an inkling that might work with minimal adjustment.

GoldenTK, 2020-11-27:
I'm getting following error on my Raspberry Pi 4B: 
```
pi@raspberrypi:~/src/88x2bu $ sudo ./install-driver.sh 
dkms appears to be installed.
Copying driver source files to: /usr/src/rtl88x2bu-5.8.7.4
Copying 88x2bu.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl88x2bu/5.8.7.4/source ->
                 /usr/src/rtl88x2bu-5.8.7.4

DKMS: add completed.
dkms add was successful.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.4.79-v8+ -j3...................................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.79-v8+ (aarch64)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.4/build/make.log for more information.
An error occurred while running: dkms build
```

Do you know what could go wrong?

morrownr, 2020-11-27:
Where is PieGuy when we need him? <grin>

GoldenTK, are you running the 64 bit Raspberry Pi OS?

GoldenTK, 2020-11-28:
Yes, I’m using latest Raspberry Pi OS Lite 64bit.

PieGuy314, 2020-11-28:
> Where is PieGuy when we need him?
> 
> GoldenTK, are you running the 64 bit Raspberry Pi OS?

I'd hazard a guess it is PiOS-64-bit given the kernel version. There's a reason it's still in Beta. I opted for DietPi a while ago for 64-bit support. Way less bleeding edge. And supports 88x2bu!

These appear to be kernel/linker related issues. I doubt the driver will need modification.

morrownr, 2020-11-28:
GoldenTK, I was testing today with my RasPi 4B with the RPi OS 32 bit with the 5.10 rc5 kernel. Worked like a charm. I have not tested with the 64 bit version of RPI OS yet. I'm going to go with what PieGuy said in that the log did not look like a driver code issue but rather a dev environment problem as it looked like all of the object files compiled fine and then the process fell apart when it was time to put everything together. I've compiled a few drivers in my time and I can't say that I've seen this before. If you have an extra sd card, you might flash some alternatives to see what happens. My best bet is that RPI OS 64 bit is in testing because it is not ready.

The section of the Makefile that supports ARM64 is standard except for one line that was recommended by another visitor:

EXTRA_CFLAGS += -fno-stack-protector

You could search for and delete that line to see what happens. You should probably run $ sudo ./remove-driver.sh to clean things up before trying another compile. Keep us updated.

GoldenTK, 2020-11-28:
Yeah, that was the problem with Raspberry Pi OS, I installed DietPi and driver is working as expected, we can add Rocketek WL1200AT to support list. Thanks for help @morrownr and @PieGuy314.

PieGuy314, 2020-11-28:
> The section of the Makefile that supports ARM64 is standard except for one line that was recommended by another visitor:
>
> EXTRA_CFLAGS += -fno-stack-protector
>
> You could search for and delete that [...]

Not recommended. The compiler with fail with an 'undefined reference to __stack_chk_fail' error. The solution is to define __stack_chk_fail(void)' as an additional function somewhere in the codebase or, alternatively, specify 'EXTRA_CFLAGS += -fno-stack-protector' in the makefile. The outcome is similar.

GoldenTK, 2020-11-28:
Btw, do you know how to run dual band connection? I can see only one wlan0 interface.

morrownr, 2020-11-29:
https://wiki.gentoo.org/wiki/Hostapd

Hopefully that helps.

closed 2020-12-14 by morrownr

ClaudioCimarelli, 2021-04-13:
> Yeah, that was the problem with Raspberry Pi OS, I installed DietPi and driver is working as expected, we can add Rocketek WL1200AT to supported devices list. Thanks for help @morrownr and @PieGuy314.

@GoldenTK I am having this issue using DietPI v6.31.2, kernel DietPi 5.4.51-v8+, ARM64, raspberry pi 3B. Could you tell me how did you make it work and on which version (kernel, os, rpi model)? Many thanks

KSofen, 2024-12-29:
The config options in the Makefile are for I386.  Do I have to edit the Makefile to compile the driver for a Raspberry Pi 4?  Does compiling this on the RP4 just handle this issue?

`CONFIG_PLATFORM_I386_PC = y
CONFIG_PLATFORM_RTL8198D = n
CONFIG_PLATFORM_ANDROID_X86 = n
CONFIG_PLATFORM_ANDROID_INTEL_X86 = n
CONFIG_PLATFORM_NV_TK1 = n
CONFIG_PLATFORM_NV_TK1_UBUNTU = n
CONFIG_PLATFORM_ARM_SUNxI = n
CONFIG_PLATFORM_RTK1319 = n
CONFIG_PLATFORM_RTK16XXB = n
CONFIG_PLATFORM_AML_S905 = n
CONFIG_PLATFORM_AML_S905_L3A = n
CONFIG_PLATFORM_HUANGLONG = n
CONFIG_PLATFORM_ARM_ROCKCHIP = n`

Any of this need editing or changing or adding something to work?

morrownr, 2024-12-29:
Hi @KSofen 

>  Do I have to edit the Makefile to compile the driver for a Raspberry Pi 4?

No

> Does compiling this on the RP4 just handle this issue?

Yes, I put autodetection routines in the Makefile and install-driver.sh to make things easy... hopefully easy. Realtek really makes their USB support difficult.

FYI: I work at another repo to help get good standards compliant WiFi 5 drivers for Realtek in the kernel:

 https://github.com/lwfinger/rtw88

Work at that site goes upstream to the Linux Mainline kernel when ready. The rtl88x2bu driver, rtw_8822bu, has been steadily improved over the last year but you won't see the improvements until your distro gets to kernel 6.12 or later. What you can do is install from the link directly above. We can use testers, especially AP mode testers.

KSofen, 2024-12-29:
Everything seemed to compile properly but wlan1 did not appear when I ran iwconfig after a reboot.  I'd really like to use it as an AP since it's faster than the RP4's built-in radio.  I have an RP4 setup as an AdGuard DNS, Wireguard server, and wireless bridge combo. I'd really like a driver that would work with OpenWrt.

morrownr, 2024-12-30:
>  I'd really like a driver that would work with OpenWrt.

The rtw88 drivers were added to OpenWRT with 24.10 rc4.

KSofen, 2024-12-30:
I did notice that in rc4 - but they don't work for the rtl8852bu chip -  there's something different about that chip or the Fenvi AX1800 USB wireless dongle.  The old driver from you and the new one you pointed me to in this thread compiled and installed just fine - but as far as Debian Linux (and rc4 OpenWrt) are concerned, nothing it recognizes is connected.

It's not a good dongle and I've spend a good deal of time screwing with it.  I really have to give up and move on.  Thanks so much for all your efforts anyway.  You're right in that Realtek does not make it easy.

morrownr, 2024-12-31:
>  but they don't work for the rtl8852bu chip

That is a WiFi 6 chipset. 

> It's not a good dongle and I've spend a good deal of time screwing with it.

Here is a message that I have pinned at issues in USB-WiFi:

https://github.com/morrownr/USB-WiFi/issues/449

If you are looking for a good adapter:

https://github.com/morrownr/USB-WiFi

Recommend you read menu items 1 and 2. You really don't have to buy adapters based on Realtek chips. The Plug and Play List has many really good adapters that work with in-kernel drivers.