morrownr/88x2bu #17: 80MHz in monitor mode

uchiit, 2020-11-13:
Hi!

Successfully deployed on RaspberryPi 4 under Ubuntu Bionic Beaver.

But.

Unable to dump traffic in 80MHz channels. HT40+ and HT40- works fine but 80MHz can show me only Management frames. And I know that there is data, because there is transmission that I can see with my Intel ax200 chipset and in Radiotap info there is 80MHz.

TP-Link Archer T4U V3. RTL8812BU.

uchiit, 2020-11-13:
*
Driver version 5.8.7.2
Can't see this adapter with 5.8.7.4 for some reason.

morrownr, 2020-11-13:
uchiit,

Can I get you to dig out the device ID for me?

$ lsusb

Let's get 5.8.7.4 working first then I'll see what the 80 MHz problem is. I'm not maintaining 5.8.7.2 anymore. 5.8.7.4 has a much more modern base code.

uchiit, 2020-11-16:
Hi morrownr,
Here it is:
~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2357:0115
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Device 003

morrownr, 2020-11-16:
uchiit,

Thank you very much for providing the information. Your device ID 2357:0115 indicated that you have a TP-Link Archer T4U V3 which is based on the rtl88x2bu chipset. That confirms that the driver you downloaded here is the right one, however, I found an error in 5.8.7.4. The driver contained a line that also showed this driver supports the rtl8812au chipset which it does not. I have corrected the problem. Please remove any version you currently have installed and install a newly downloaded 5.8.7.4 version and install it according to instructions. Please note that there are big changes in the instructions that have been posted in the last few days.

I appreciate you helping me find this bug. Please report back results.

Nick

morrownr, 2020-11-16:
uchiit, if 5.8.7.4 works for you after the fix, please test the 80MHz channel problem again. It would help if you would provide me a step by step checklist of exactly how you discovered the 80MHz problem so that I can try to duplicate it here.

Nick

uchiit, 2020-11-17:
Hi morrownr,

Driver can see my NIC now, thank you!

But problem with 80MHz still exists.

Step-by-step guide:
1. Install driver

2. Install libpcap:
sudo git clone https://github.com/the-tcpdump-group/libpcap
sudo git install gcc bison flex libnl-genl-3-dev pkg-config
cd libpcap
sudo ./configure --enable-remote
sudo make
sudo make install

3. Run rpcap:
sudo rpcap -n & (need to be in background)

4. Configure interface monitor mode and monitored channel:
_sudo iw dev wlan1 set type monitor_
_sudo ifconfig wlan1 up_
_sudo iw dev wlan1 set channel 36 HT40+_ (this config will show you all management frames and broadcast data, due to it goes on legacy speeds and also data frames from old clients (802.11a/n 20/40MHz))
_sudo iw dev wlan1 set channel 36 80MHz_ (this config will show you a little bit more then nothing, even not all mgmt frames (all my clients works on this width for sure, verified by other NICs in monitor mode))

5. Connect to this device in Windows through Wireshark using RPCAP
a) Capture>>Options...>>Manage Interfaces...>>Remote Interfaces>>plus (+)
b) Host: IP of my Ubuntu on Raspberry
    Port: 2002
c) Ok
d) Check only wlan1 checkbox
e) Ok
f) Choose **rpcap://ubuntu-IP/wlan1** interface with Link-layer header equal to "802.11 plus radiotap header"
g) Start

Now you can change channels and bands in Ubuntu CLI with help of:
sudo iw dev wlan1 set channel {channel} {bandwidth}
and wireshark will show you in real time what happens and what can be heard by wNIC

morrownr, 2020-11-17:
Hi uchiit,

Glad to hear the driver now works with your adapter. This checklist is exactly what I need. Thank you. I'll try to give you an update every other day or so.

Nick

morrownr, 2020-11-20:
I'm a little busy this week but will work on this as I have time.

morrownr, 2020-11-21:
In the following file:

/etc/modprobe.d/88x2bu.conf

... add this module parameter:

rtw_ht_enable = 2

reboot and see if that helps.

Nick

uchiit, 2020-11-22:
Maybe I need to find VHT? Due to in 802.11 standard HT stands for 802.11n (Wi-Fi 4), VHT stands for 802.11ac (Wi-Fi 5) and HE is about 802.11ax (Wi-Fi 6). And 80MHz is about 802.11ac (VHT or Wi-Fi 5). Also can be seen in this table http://mcsindex.com

morrownr, 2020-11-23:
From the driver directory, run:

$ grep -r "module_param(" *

That will show you all of the module parameters. They are all in one file which you can open up and look at. The documentation is far from being adequate. Most of my time has to go to other things right now so if you can figure out what direction to go with this issue, that is great. At some I will have more to to work it but not for a while.

Nick

P.S. If you happen to decide to edit a driver src file, remember to make sure you use a unix/linux compatible editor.

uchiit, 2020-11-24:
Hi Nick,

I see one string there, that maybe can fix this issue, but I didn't understand, where I need to make changes to edit this string.
_os_dep/linux/os_intfs.c:module_param(rtw_vht_rx_mcs_map, uint, 0644);_
Maybe it will not help, but who knows...

Anton

morrownr, 2020-11-24:
You do not need to edit the driver code to change settings for module parameters. The ./install-driver.sh script installs this file:

88x2bu.conf

and you can edit it with this command:

$ sudo nano /etc/modprobe.d/88x2bu.conf

I added this file to make it easier for users to change options. There are many options for this driver that can be changed with module parameters. The 3 items in the default file can be changed or deleted and many more can be added.

I went into the file that supports the module parameters ( os_dep/linux/os_intfs.c ) to see the section you are talking about. Here is the section that supports options for VHT:

#ifdef CONFIG_80211AC_VHT
int rtw_vht_enable = 1; /* 0:disable, 1:enable, 2:force auto enable */
module_param(rtw_vht_enable, int, 0644);

int rtw_ampdu_factor = 7;

uint rtw_vht_rx_mcs_map = 0xaaaa;
module_param(rtw_vht_rx_mcs_map, uint, 0644);
MODULE_PARM_DESC(rtw_vht_rx_mcs_map, "VHT RX MCS map");
#endif /* CONFIG_80211AC_VHT */
-----
To change the value of rtw_vht_rx_mcs_map you would simply need to add something like:

rtw_vht_rx_mcs_map=0xabab

to the appropriate line in 88x2bu.conf. I have no idea at this point what the value should be but maybe you can research it.

At this point we do not know if the problem is a bug, configuration problem or the capability is not there. I also work on 4 other wifi drivers so we will have something to use to compare as I have time to dig deeper into this issue. I suspect that this will take time but can be fixed.

Nick

uchiit, 2020-11-25:
Hi Nick!

As I can see from this dialog https://github.com/cilynx/rtl88x2bu/issues/24 there can be trouble with 80MHz monitoring and there is no such problem in this driver: https://github.com/borting/rtw88-usb
But this driver didn't work for me and I can't test it... ((
Maybe you can find differences in configuration between your version and this one?

Anton

morrownr, 2020-11-27:
Hi Anton,

Maybe I can find the difference. Given the information and checklist that you provided and the multiple drivers and adapters that I have available, I'll bet we can track this issue down. It may take time but I think it can be done.

morrownr, 2020-11-30:
Anton,

I took a look at rtw-88-usb. That driver is so different in so many ways it would be very difficult to narrow down differences without it taking far more time than I have available. I do have a plan. I maintain several different drivers here: github.com/morrownr

The most modern driver is the one for 8812au. What I will do is test my other drivers with your checklist and if one passes, I can do a code comparison to see if I can figure out what the problem.

FYI: I do have hardware for all of the drivers on this site so I can test. FWIW: I plan to bring a 5th driver online soon. It is for the rtl8821au. I brought the rtl8814au online yesterday. 

Nick

uchiit, 2020-12-07:
Hi Nick,

That's great. You can write if you need any help with packet description/understanding of 802.11 frames and processes.

Anton

morrownr, 2020-12-07:
Anton,

Check out Issue #1 at my 8814au repo: github.com/morrow/8814au

The poster and I are both seeing operations in both bands work fine with bgn but no ac. This is with AP mode using hostapd.

Is this related? It could be. We have both tried rtw_vht_enable=2 without success.

Right now I am maintaining 5 drivers. Here is how I rank them in 3 categories:

Most modern:
8812au
8821cu
88x2bu
8821au
8814au

Cleanest log:
8812au
8821cu
8821au
8814au
88x2bu

Fastest (iperf3):
8814au (to be expected)
88x2bu
8812au
8821cu
8821au

I am going to reconfigure some things here in the lab to start testing your issue and the AP issue with the 8812au to see if I see better results with what is the most modern and trouble free driver here.

It is highly likely that I will be asking regarding packet description/understanding of 802.11 frames and processes.

closed 2021-06-21 by morrownr