morrownr/88x2bu #20: Enable the 867Mb/s mode

bernhara, 2020-11-30:
To have 867Mb/s enable, the driver flag rtw_vht_enable_ should be set to 2.

See: https://github.com/brektrou/rtl8821CU/issues/5#issuecomment-568207608

morrownr, 2020-11-30:
Bernhara, thanks for the info. I had been looking at that setting for other reasons. I'm going to change this in the code as I can't find a reason not to change it at this point.

Nick

PieGuy314, 2020-11-30:
Defaulting to an operating mode with a reduced range, esp. for an access point, makes little sense IMHO. Enabling it via modprobe isn't difficult. Having to work backwards to get a functioning driver, however, is.

morrownr, 2020-12-01:
PieGuy, I would appreciate it if you would elaborate on this issue. I have been keeping my eye on rtw_vht_enable for a while but but the lack of documentation has not helped. Given adequate documentation, I will look at adding the parameter to my options file and will document it in the README...and turn the default back to 1.

The reason I am automatically installing an options file is two fold: 1) I use these drivers and I don't like to have to dig around for documentation when I need to make a change and 2) I'd like for the documentation and configurability to be such that inexperienced users can succeed before giving up and going back to Windows or MacOS.

I'm also considering adding a parameter to the options file regarding power saving. I have gathered some information but have not done any testing yet. Do you have any advice to pass on?

morrownr, 2020-12-01:
Okay, I had to do some testing... PieGuy understated things when he said "reduced range." I would say dramatically reduced range when operating in managed mode. I reverted the change but I sure would like additional information so I can properly document this and add an option to the 88x2bu.conf file.

bernhara, 2020-12-01:
Having an explicit commented option in 88x2bu.conf is what I had really in mind.
This is currently the way I activated 876Mb/s for my tests.

morrownr, 2020-12-01:
bernhara,

I understand and appreciate the information. I had seen no information about the consequences of making 2 the default until PieGuy jumped in. I'll add it to the README and 88x2bu.conf file but I need to have a good understanding of what is going on. If my understanding is correct, this is needed to enable "full" speed when operating in AP mode. When I say full speed, I think I am saying that 80 MHz channel width is available but I'd like to make that clear. Here is the setting from the source code:

int rtw_vht_enable = 1; /* 0:disable, 1:enable, 2:force auto enable */

The comment sort of documents the setting but it leaves more questions than answers. I would like to document this better than what you see and if you or anyone else here can help in that regard I would appreciate it.

morrownr, 2020-12-01:
Need help. Here is what I have so far to document this option:

-----
______ ______  options: ( rtw_vht_enable )
```
  0 = Disable
  1 = Enable  (default)
  2 = Force auto enable
```
  Note: Unless you know what you are doing, don't mess with this.

-----
I need a good name... see the blanks and a good explanation for exactly what this does in the note would go a long way.

Thoughts?

PieGuy314, 2020-12-01:
Keeping the options, disabled by default, but well documented in the conf file is, I feel, the right way to go. Far easier to document and provide the user with a system that can be tuned to suit specific requirements.

The same goes for things like power saving. Automagically using 'iwconfig' (which is called by 'wireless-tools') when it encounters a 'wireless-power' statement in '/etc/network/interfaces' is, I feel, the most appropriate way to manage power saving. It's device specific, not driver specific. By all means include the various driver specific options in the conf file. But I'd implore you not to enable them by default. (Spent several weeks a long time ago trying to make an el-cheapo wifi nub work in AP mode reliably - all boiled down to power saving being enabled but poorly documented.)

morrownr, 2020-12-02:
PieGuy,

I hear you loud and clear. I've been in those frustrating situations before myself. Good documentation solves many problems and that is what I am attempting to do with the drivers here. However, we can't just "disable" all options by default because the default for many options is not "disabled" and disabling them would cause problems. rtw_vht_enable is a good example. The default is enabled and it is the "high speed" option that yields undesirable results in some situations.

Back to my question: What should I call this option? "Enable VHT options:" ?

I have a folder where I store all of the documentation regarding Realtek USB WiFi devices I have been able to find. I have never found a "name" for most of these options or module parameters if we want to get technical. I guess my only option is a lot of testing to see how to actually describe the function.

Then you have options like "rtw_power_mgnt=0" and "rtw_ips_mode=1". I would like to add an option for folks to turn power savings off. The default is on as per the Makefile and it works very well in most situations but not all. The problem is that it is not clear which options to expose in 88x2bu.conf and how to properly document them. This may be another case where large amounts of testing and reading code may be required. I appreciate programmers but I appreciate the ones that document their code more.

Before I go, just to be clear: I agree with you that great care should be used to change defaults.

Nick

morrownr, 2020-12-02:
All,

Please tell me if you think this is adequate and accurate documentation for adding this option to 88x2bu.conf:

-----
Enable VHT options: ( rtw_vht_enable )
```
  0 = Disable
  1 = Enable  (default)
  2 = Force auto enable

```
  Note: Unless you know what you are doing, don't change the default for this option. A non-default setting can degrade performance in some operational modes. Setting this option to 2 will allow 80 MHz channel width for AP mode.
-----

Thoughts?

bernhara, 2020-12-02:
A simple test:
```
insmod ./88x2bu.ko  rtw_switch_usb_mode=0 rtw_drv_log_level=4 rtw_vht_enable=0 rtw_ant_num=2
```

Then 
`hostapd -dd /etc/hostapd/hostapd-pirel.conf`
gives
```
...
hw vht capab: 0x0, conf vht capab: 0x1020
Driver does not support configured VHT capability [VHT_CAP_SHORT_GI_80]
...
```
With rtw_vht_enable=1 or rtw_vht_enable=2
```
insmod ./88x2bu.ko  rtw_switch_usb_mode=0 rtw_drv_log_level=4 rtw_vht_enable=1 rtw_ant_num=2
```
hostapd starts well.

bernhara, 2020-12-02:
Sorry, rtw_vht_enable=1 or rtw_vht_enable=2 are not equivalent.

rtw_vht_enable=1 :
```
insmod ./88x2bu.ko  rtw_drv_log_level=4 rtw_vht_enable=1 rtw_ant_num=2
iwconfig
...
wlx1cbfceb41964  IEEE 802.11bgn  ESSID:"PIREL"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:5.18 GHz  Access Point: 1C:BF:CE:B4:19:64
          Bit Rate:300 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

While with rtw_vht_enable=2:

```
insmod ./88x2bu.ko  rtw_drv_log_level=4 rtw_vht_enable=2 rtw_ant_num=2
iwconfig
...
wlx1cbfceb41964  IEEE 802.11bgn  ESSID:"PIREL"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:5.18 GHz  Access Point: 1C:BF:CE:B4:19:64
          Bit Rate:867 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

```

morrownr, 2020-12-02:
bernhara,

You example directly above showing the difference in bit rate is in AP mode, correct?

morrownr, 2020-12-05:
bernhara,

Would you please take a look at the new version of 88x2bu.conf. And download and test to make sure I didn't make typos or other mess up adding rtw_vht_enable? I would appreciate it.

Nick

morrownr, 2020-12-08:
bernhara,

Would you mind looking at the hostapd issue as posted for this driver?

https://github.com/morrownr/8814au

morrownr, 2020-12-14:
Support added to 88x2bu.conf and README.

closed 2020-12-14 by morrownr