morrownr/88x2bu #21: how does it compare to others drivers

mirsella, 2020-12-02:
hello, thanks for your work.
do you know how it compare to others 88x2bu drivers like [Rincat's](https://github.com/RinCat/RTL88x2BU-Linux-Driver) (the one I'm using) and [cilynx's](https://github.com/cilynx/rtl88x2bu)

I will try it when getting back home, but if you know some details that make this repo "better" than the other, I would like to know so I can change to this driver, thanks

morrownr, 2020-12-02:
Hello mirsella,

I am familiar with Rincat's repo. Rincat does a good job. The primary difference is that Rincat's driver is based on the Realtek v5.8.7.2 code base whereas this driver is based on the Realtek v5.8.7.4 code. My testing has shown a fairly significant speed increase with the v5.8.7.4 code. I use iperf3 for testing. You may not notice the speed increase depending on how you use your device but it is there nevertheless.

I am not familiar with cilynx's repo but a quick look shows that it is based on a relatively old code base.

My main goals here are

- maintain easy-to-understand documentation that is presented in a professional manner
- easy installation for those with less experience
- add features that make the experience better
- upgrade to the most modern version of the driver available in a timely manner

One thing that is different here than most driver repos is that I am maintaining multiple drivers. I have 4 available now and will be adding another soon. You can see all here:

https://github.com/morrownr

If you decide to install this driver, please take the time to write down notes so as to provide feedback. 

Nick

mirsella, 2020-12-02:
thanks, work for me so i guess I'm better with a newer version

closed 2020-12-02 by mirsella