morrownr/88x2bu #23: Add support for multiple kernel versions on the same machine

Rubo3, 2020-12-07:
Hi! Thank you for your work! On my machine I run Arch Linux with two kernel packages, the latest linux and linux-lts. I would like to use this driver for my Archer T3U on both kernels, but I noticed that the install script created a directory  named after the driver name and its version, so when I compiled it agains the linux-lts headers it worked flawlessly and the driver was added to the dkms tree successfully, but when I did the same agains the linux headers, dkms found there already was the driver directory, so it stopped. Please, can you add support for multiple kernels? Do you think it's feasible by just adding to the dkms directory name the kernel version?

morrownr, 2020-12-07:
Interesting. Let me take a look and think about this.

morrownr, 2020-12-14:
I took a look and think this can be done but right now I am busy with kernel 5.10 support and testing. I maintain 5 drivers on this site which is a lot of work.

Could I get you to do an extensive internet search on this topic? I suspect this has come up before. We have to make sure any solution is good with dkms as dkms is heavily involved with the installation, maintenance and removal of this driver.

Also, would you mind adding a section on Arch Linux to the README? I'm talking about the section where you install the requirements which can vary from distro to distro. That could help fellow Arch users.

Regards

Rubo3, 2020-12-14:
Sure! I'll do it when I have some time. Speaking of Arch, do you think it would be useful to add a PKGBUILD for this module (and any other of yours) on the [AUR](https://aur.archlinux.org/)?

morrownr, 2020-12-20:
@Rubo3 

I took the time to reread the dkms docs while going through the current install-driver.sh file. I did this after reverting most of your pull request when testing showed really had things happening. 

dkms add has no -k switch as far as I can tell. The usage of -k you added to build and install is simply going to duplicate default action. All of this causes me to wonder if I have a good understanding of what is going on.

The distros I use for testing all have multiple kernels installed and I choose the one to boot during the boot process, If i want to install this driver, I simply install it. This all happens on various Debian based distros. I am not familiar with Arch.

I think we need to backup and discover what the real problem is. If I install Manjaro on one of my test boxes, is that going to work for testing purposes? If so, write me a checklist that I can use to duplicate the problem.

Lastly, you mentioned something about a PKGBUILD. Please educate me what that is.

morrownr, 2020-12-21:
I like to learn new things. To get a handle on this issue I decided to install Arch. Well, that was an experience and not a good one. I blow OS's put with clean installs a lot here in order to test on clean installations. I don't have time to deal with the Arch installation. So, I decided to try Manjaro. Much much better. I followed the instructions here for installation of this driver and it didn't work. I tracked the problem down to the installation not finding the head files. pacman says the header files are installed and when I look for them, they are not there. So where does Manjaro/Arch put the header files? For this issue to be resolved, I have to understand what is going on. Help would be appreciated.

Rubo3, 2020-12-21:
Hi, sorry for the late reply! I'd be really glad to help you.
It's strange that `dkms` has no `-k` option, which docs did you read? Based on what I found on its [GitHub page](https://github.com/dell/dkms) and by running `man dkms` it does. Specifically, this is a section extracted from its `man` page:
```
OPTIONS
[...]
       -k <kernel-version>/<arch>
              The  kernel  and  arch  to perform the action upon.  You can specify multiple kernel version/arch pairs on the
              command line by repeating the -k argument with a different kernel version and arch.  However, not all  actions
              support  multiple  kernel  versions  (it will error out in this case).  The arch part can be omitted, and DKMS
              will assume you want it to be the arch of the currently running system.
```

I have also tested the final changes before issuing the merge request. First, I removed the driver from the dkms tree, then I ran the install script on Linux 5.4, dkms added it to its tree, built it against that kernel headers, and installed it. Then, I run it on Linux 5.9, dkms found it was already in its tree, built it against that kernel headers, and installed it. I tried to remove it in both kernels and repeated the experiment again in different order. I dind't have, nor I have as of now, any problem: I'm actually connected to my home network thanks to your driver! I would go as far to say that it's even more stable than the official Windows driver, I don't really have any issue on Linux, and I can't say the same for Windows.

Now I'm wondering if one of the possible causes is the update of the installation script? Maybe if you have the driver added to the dkms tree without `-k`, then you pull the latest changes from this repo and then run the installation script again, you may have a problem because now dkms is running with the `-k` option?

Speaking of Arch, I'm sorry to hear you did not have a good experience :( If it helps, you are not alone. I can't speak for Manjaro, since now I only have Arch on my machines, anyway I must say I did not experience your problem, I installed `linux linux-firmware linux-headers linux-lts linux-lts-headers dkms git` via `pacman`. These are the directories from the results of the queries to its database to list all the files installed by `linux-headers` and 'linux-lts-headers` respectively, with `pacman -Ql linux-headers` and `sudo pacman -Ql linux-lts-headers`: [(1)](https://privatebin.net/?4e515e1a5fcb84b8#7CrNdefbDRmWrL1DMDxtsGfp7aaeot9jt5Zs99Qh9LaL) and [(2)](https://privatebin.net/?f6e9693f682edaf5#7jJapErmeNG5Jrz8oCMKy4oCcAwwWkX8ghBbjt4gcnxk). Please note they will expire tomorrow night UTC+2.

Please, can you provide me the kernel version you are running on Manjaro? I have used it in the past, and it didn't always run the latest kernel by default, so when you install the `linux-lts-headers` package, you are installing the headers for the current latest version of the Linux kernel supported by Arch, which may not be the one you are running. In which case, you have to download the specific `linux**-headers`, replacing `**` with the kernel version: for example, `linux-58 headers` for Linux 5.8.

I hope we fix the problem soon!

morrownr, 2020-12-21:
I didn't say that dkms doesn't have the -k switch. I said dkms add does not have the -k switch. Here is a link to the docs I use:

[](http://manpages.ubuntu.com/manpages/groovy/en/man8/dkms.8.html)

You had added the -k switch to add. After merging your code, I starting testing and the install was blowing up on Debian based distros so I had to revert. We've got to find a solution that will work on both Debian and Arch based distros.

Regarding the stability of the driver: Yes, this is a very good driver. In fact, all 5 of the drivers on this site are very good. The only issues that have been reported on any of the drivers are specific issues for what I will call high end functions but not many of those. We are having a hard time getting any of these drivers to open in 11ac in AP Mode. They work fine in AP Mode to include channel width of 80 and 11n but not 11ac. Most people have internet access that is far slower than what can be done currently but if they should do 11ac, it is time to figure out what the problem is... and it may not be the drivers, it may be hostapd.

Back on topic: Regarding my Manjaro installation, the version of the kernel as shown by "uname -r" is "5.9.11-3-MANJARO"

The error is shown as 'your kernel headers cannot be found at "/usr/lib/modules/5.9.11-3-MANJARO/build"'

I tried installing the lts headers and did a lot of reading but nothing worked. The headers are not where they should be and I don't know why. Maybe it is my Debian mind that is confusing things but I can have a dozen kernels installed on a Debian based distro and select which one to boot during boot. If I ask for headers to be installed, they are installed. Usually I don't have to ask for headers to be installed as they are automatically installed for anything like dkms that needs them.

I think the priority right now is for me to get the correct headers installed in Manjaro and for me to be a little more familiar with Arch based distros. If you can help me do this, that would be great.

morrownr, 2020-12-25:
Rubo3,

Take a look at install-driver.sh, remove-driver.sh and dkms.conf. I had to carefully reread the dkms docs, rework the files and test a lot on debian based distros. This rework works just a little different but should be acceptable on debian distros. It includes the ideas you had to make it work on Arch. I would appreciate if you would test on Arch and report back.

I finally managed to install the headers in the right place on Manjaro. The reworked driver is installing fine on kernel 5.9 on Manjaro. Now if I can figure out how to switch to the 5.4 kernel.

Rubo3, 2020-12-25:
Merry Christmas! (and sorry for the late reply, again.)

I did some testing and it seems to work well, but I had to edit `install-driver.sh` in order to let it continue when `dkms add ...` exits with status code 3 (for the driver is already in dkms tree), because otherwise I wouldn't be able to install it for different kernels. Furthermore, I didn't notice last time that `remove-driver.sh` removes the source directory and the configuration file in addition to the driver, so I think you should decide if you want it to remove all the drivers (i.e. for all kernels) with the `--all` flag, and I think it would be ok, or the driver in the current kernel only, with the `-k ${KRNL_VERSION}` flag, in which case I don't think it would be wise to remove the source directory and the configuration file. Maybe some more logic could be added to it, for example it functionality could be altered by passing a `--all` or `-k` flag to the script.

Another thing I noticed, when I tried to uninstall the driver for `linux` (I currently use `linux-lts`, not `linux`, as my main kernel), this was the script output:
```bash
-------- Uninstall Beginning --------
Module:  rtl88x2bu
Version: 5.8.7.4
Kernel:  5.9.14-arch1-1 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

88x2bu.ko.xz:
 - Uninstallation
   - Deleting from: /usr/lib/modules/5.9.14-arch1-1/
rmdir: failed to remove '': No such file or directory
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod.....

DKMS: uninstall completed.
```
despite that the driver was still loaded even after a reboot, so I installed it again and whilst it worked (and nothing changed for me, as the driver continued to work flawlessly), it still bothers me that it wasn't effectively removed. The older version of the script didn't behave like that.

I'm glad to hear the driver is working fine on 5.9. To switch kernels, my advise is to use GRUB, since it compiles a list of the currently installed kernels when its configuration file is generated, and at boot it lets you choose which one to load. At least, this is the method I use. Let me know if you encounter any trouble, if you decide to go this way.

morrownr, 2020-12-25:
Merry Christmas. 

Thanks for the additional information. At this point, I find myself more or less back where we started. I don't have a good handle on what the problem is. Let me throw out some things:

It is unclear why you are trying to install the driver into a kernel when it is already installed. That is what the error code 3 from dkms add is telling you. Let me point something out... direct from the dkms docs:

AUTOINSTALL=
              If    this     directive     is     set     to     yes     then     the     service
              /etc/rc.d/init.d/dkms_autoinstaller  will  automatically try to install this module
              on any kernel you boot into.   See  the  section  on  dkms_autoinstaller  for  more
              information.

I have AUTOINSTALL set to yes in dkms.conf. The Debian based systems that I use and test with generally have multiple kernels installed. If I use GRUB to boot into a non-default kernel that does not have this driver installed, dkms jumps into action and installs it automatically. Is that not happening in Arch/Manjaro? While I do have Manjaro installed on a test system, I am still trying to figure out how to switch kernels. Kernels 5.4 and 5.9 are installed but when I call up grub, it only offers 5.9 twice and no 5.4. Debian based distros would offer me every kernel that is installed.

What I need to know is how to get this Manjaro install to boot to kernel 5.4 so I can see for myself what is going on.

FYI: I certainly don't know everything about every Linux distro but I have been using Linux for a long time. My operating system history goes like this:

CP/M
PC-DOS
DR-DOS
OS/2
Linux (since around kernel v1, mid-90's)

Have you tried booting into the different kernel to see if the driver is then automatically installed? Or, if it is not installed on the new kernel, have you tried to install it into the new kernel while booted into the new kernel?

Cheers.

morrownr, 2020-12-25:
Follow up: After searching, I found the information I needed to switch to a different kernel on Manjaro. It turns out that kernel 5.4 was not installed so I installed it. During the installation dkms kicked in and installed all prerequirements including the correct headers and it automatically compiled each usb wifi driver that I had installed for the other kernel... 5 in total. They were usable as soon as I booted into kernel 5.4.

So, tell me what you are seeing that tells you there is a problem.

I do see a problem and would like your help. The installation steps do not currently fully account for Arch/Manjaro. Arch uses pacman, not apt so some steps need to be rewritten. Can I get you to work that issue?

Cheers,

Nick

morrownr, 2021-01-03:
As far as I can tell this issue is resolved.

closed 2021-01-03 by morrownr