morrownr/88x2bu #24: Word of Warning: Current 'raspberry-kernel-headers' (5.4.79) not 64-bit friendly

PieGuy314, 2020-12-07:
Have just upgraded my RPI3B. New kernel. New kernel source. Compilation of the 88x2bu driver failed to the headers 5.4.79 not been 'fixed' for 64-bit compilation. Consequently, no 88x2bu 64-bit driver.

This has happened before and will probably happen again. I'd caution anyone who's running 64-bit PiOS or DietPi *NOT* to upgrade the kernel until the headers have been fixed.

I'm reverting to 32-bit as a consequence of my dependancy on the 88x2bu driver.

morrownr, 2020-12-07:
PieGuy, do you want to close #16?

Reports from users of other drivers on this site make it clear that the 64 bit version of RasPiOS is broken right now. On a separate subject, I moved to 32 bit 2020-12-02 for testing and there may be some issues with it as well. The drivers are compiling okay and seem to work okay but my 4B is experiencing a screen freeze after variable amounts of time. I'm going to erase reinstall and run from eth0 for a day or two to see if I can rule some things out. Things are not good in RasPiOS land right now.

PieGuy314, 2020-12-07:
> PieGuy, do you want to close #16?

Sure, you've captured everything in the documentation you've compiled. I used it today!

> Reports from users of other drivers on this site make it clear that the 64 bit version of RasPiOS is broken right now. On a separate subject, I moved to 32 bit 2020-12-02 for testing and there may [...]

Have done a fresh install of 32-bit DietPi-OS. I'm running headless so can't comment regarding screen freezes. Did however, stream 320kps audio via shairport-sync to my hifi for over 6 hours without any glitching.

I'm confident the driver code is okay but will run some diagnostics to check throughput, etc. These things happen from time-to-time. 64-bit is in beta for a reason. At the end of the day, "If it ain't broke, don't fix it!" is probably my takeaway lesson for today.

morrownr, 2020-12-07:
One thing that has been catching my eye for a while is some log entries. I pushed a little script called save-log.sh to the repo today. You might find it handy to check the log entries as it cuts out everything but the RTW entries and cleans some things up so I can more easily use Meld to compare logs between different drivers or difference driver versions. Of the 5 drivers I maintain, this driver has the "dirtiest" log. It seems to work well but it is just not on par with the quality of the 8812au driver. Granted, the 8812au driver is a slightly more recent driver. Let me know if you see any log entries that are of concern.

Nick

PieGuy314, 2020-12-07:
Logs are clean here. And have been since I've been using this repo. Are you using the default logging level?

PieGuy314, 2021-01-03:
Hi Nick,

Believe you can close this now. Rebuilt the driver using the latest kernel-headers (1.20201201-1) just now - working again.

morrownr, 2021-01-03:
Hi PieGuy, that is good to hear.

closed 2021-01-03 by morrownr