morrownr/88x2bu #26: Probe Response in hostapd WPS mode does not contain metadata

crgorect, 2020-12-17:
Hi, I am using this driver to create an AP with WPS mode. I have populated the metadata fields and hostapd seems to believe that the beacon is being created properly. When I do an iw scan on a different pi I am picking up other devices wps data, but not that of this given AP. Ideally I would be able to get the Model Number, Serial Number, Device Type, and Device Name properly populated. On a different note that I haven't tried to investigating any insights into station count and utilization outside of WPS would be great as well. Example scan is here [iw-scan.txt](https://github.com/morrownr/88x2bu/files/5712433/iw-scan.txt).

* hostapd logs

> WPS: Use configured UUID - hexdump(len=16): 36 13 59 00 1d d2 11 b2 86 21 57 53 53 51 51 55
WPS: Converting display to virtual_display for WPS 2.0 compliance
WPS: Build Beacon IEs
WPS:  * Version (hardcoded 0x10)
WPS:  * Wi-Fi Protected Setup State (2)
WPS:  * AP Setup Locked
WPS:  * Version2 (0x20)
WPS: Build Probe Response IEs
WPS:  * Version (hardcoded 0x10)
WPS:  * Wi-Fi Protected Setup State (2)
WPS:  * AP Setup Locked
WPS:  * Response Type (3)
WPS:  * UUID-E
WPS:  * Manufacturer
WPS:  * Model Name
WPS:  * Model Number
WPS:  * Serial Number
WPS:  * Primary Device Type
WPS:  * Device Name
WPS:  * Config Methods (200c)
WPS:  * Version2 (0x20)
random: Got 20/20 random bytes
Get randomness: len=32 entropy=0
GMK - hexdump(len=32): [REMOVED]
Get randomness: len=32 entropy=0
Key Counter - hexdump(len=32): [REMOVED]
WPA: Delay group state machine start until Beacon frames have been configured
WPS: Building WPS IE for (Re)Association Response
WPS:  * Version (hardcoded 0x10)
WPS:  * Response Type (3)
WPS:  * Version2 (0x20)

* example hostapd.conf

> interface=wlan3
driver=nl80211
ssid=testap
channel=7
hw_mode=g
wme_enabled=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=windtalker
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
ctrl_interface=/var/run/hostapd
#WPS configuration (AP configured, do not allow external WPS Registrars)
wpa_psk_file=/etc/hostapd/hostapd.psk
eap_server=1
wps_state=2
ap_setup_locked=1
wps_pin_requests=/var/run/hostapd.pin-req
#fields being dropped
uuid=357ac280-1dd2-11b2-8601-5753435f5555
device_name=Node
manufacturer=TestCo
model_name=MN
model_number=001
serial_number=00001
device_type=0-DEADBEEF-1
os_version=00117000
config_methods=label display
#upnp_iface=wlan3
#friendly_name=WPS Access Point
#vendor_elements=dd0411223301
#assocresp_elements=dd0411223301

morrownr, 2020-12-19:
@crgorect 

I am currently learning hostapd. I am nowhere near the point where I can help you explore this issue. I'll leave this issue open as this site does see a lot of traffic so maybe someone can help.

morrownr, 2021-04-04:
It has been a while since this issue was posted. Is this still an issue? I have learned a lot while testing and using hostapd over the last few months. While I don't have an answer for this specific issue, what I have learned is that this driver is not the best for AP mode. These are the driver/adapter combos I recommend:

mt7612u/ driver is in-kernel
mt7610u/ driver is in-kernel
rtl8812au/ I have a driver up on this site. It is the best Realtek Linux I have ever seen.
rtl8811au/  I have a driver up on this site.

I would avoid these chipsets for AP mode applications:

rtl8812bu
rtl8811cu
rtl8814au

crgorect, 2021-04-05:
I apologize for not updating. We actually went about finishing up development and got proper addressing working and at that point after testing we did in fact get the metadata to populate correctly and can make modifications. Sorry for not posting an update right away! Thanks for your help!

closed 2021-04-05 by crgorect