morrownr/88x2bu #29: Using 2.4 Ghz while 5Ghz is available

Chaosdave34, 2020-12-28:
I have a 5 GHz usb wifi dongle and router, but it uses the 2.4 GHz channel. I am using Ubuntu 20.10 arm64 on Raspberry Pi 4.

morrownr, 2020-12-28:
Chaosdave34,

You can make sure that you use 5 GHz by doing a couple of things:

1. The first thing to do is make sure that your wifi router is using different names (SSID) for the 2.4g and 5g. Many routers come from the factory with the names of the two bands set to the same name which means the driver picks and it will pick what it thinks if the best signal. The best signal doesn't always mean the best results. Often 2.4g is picked because 2.4g is often the stronger signal. Go into your wifi router setup and ensure the names (SSIDs) are different..., such as: 2.4g = "myWiFi-2g" and "myWiFi-5g".

2. In Ubuntu, there is a button called "Forget Connect." Use that button to forget all connections except for the one 5g connection that you want to use. If the system is not setup to use 2.4g, then it won't use it

Good luck.

Chaosdave34, 2020-12-28:
Now, when I try to connect to 5G Wifi i just get an Wrong Password error. I am sure it is right because I have copied it from router setting.

morrownr, 2020-12-28:
Did you reboot the router? Do you reboot the computer? Did you forget all connections in Ubuntu?

Chaosdave34, 2020-12-29:
I rebooted the router and PC 2x and it Stil doesn't work. I can connect to 2,4GHz Wifi but when i connect to 5GHz Wifi it says wrong password or nothing happens.

PS: BecauseI am using Ubuntu 20.10 and not Raspberry Pi OS i haven' ran /raspi64.sh. Should I remove the driver and reinstall it wirh using /raspi64.sh.

morrownr, 2020-12-29:
Yes, running raspi64.sh should be done if running Ubuntu on a Raspberry Pi 4B. I guess I need to find a way to make that clear in the documentation. If you have an idea how to best make that clear, let me know.

Recommend that you run the remove-driver.sh script and delete the driver folder and then redownload and try again . I don't know if this combination has been tested so don't be discouraged if a problem comes up. I will see about testing that combo later today if I have time. Let me know how it goes.

Chaosdave34, 2020-12-29:
I would change "For 32 bit Raspberry Pi OS:" t0 "For Raspberry Pi using 32-bit OS".

But my problem still remains. 2Ghz Wi-Fi works, but when I try to connect to 5 GHz Wi-Fi I get "auth failure" error. Also, when I have one Wi-Fi for both and try to change with "sudo iwconfig wlan1 freq 5G" it changes nothing.

morrownr, 2020-12-29:
I have Ubuntu loading onto an sd card right now. I have a busy day ahead of me but will try to test this evening. Like I said, this is a combination that is new so it may take some time to work out the problems. I've had reports from RasPi OS 64 bit beta users and RasPi 64 is broken and it is the OS, not this driver. I had to give up messing with it and will wait until it is released to see where it stands. Is Ubuntu 64 in better shape? We will see. I appreciate you reporting this and hope you can stay in touch until we figure out what the problem is.

Chaosdave34, 2020-12-29:
Now it works: I have just set log level to 6 and rebooted the system, and now I can use 5 GHz Wi-Fi. Thanks for trying to help me.

morrownr, 2020-12-30:
I just finished testing Ubuntu 20.10 Desktop (RasPi version). Things went well. I had to look twice at the menu as Ubuntu makes two menu entries:

WiFi
USB WiFi

and I nearly overlooked that. My primary work systems run Linux Mint with Cinnamon as I find that interface FAR better than Gnome.

Another issue while testing: Just because you can do something, doesn't mean you should. I overclocked the RasPi4b and Ubuntu performed terribly. It is unusable. There are two big problems with Ubuntu these days:

1. Gnome is a DOG. There is a lot to be said for some of the utilities that the project has provided but Gnome is a DOG slow !??!!?1?!. I like a nice snappy interface and Gnome doesn't even provide that on stronger hardware than a Pi.
2.  I have tried to like the layout and function of the Gnome interface but it just does not work for me. It does not help my work flow, it impedes it. 

Sorry about the rant. I made some changes to the README. If you have time, take a look and tell me if makes more sense now.

Nick

Chaosdave34, 2020-12-30:
Thanks for helping me. Now the 5 GHz Wi-Fi is working. The Reason why I am using Ubuntu is, that their packets get more updates than debian. I am using the KDE plasma desktop, because it works better. 
The Readme makes more sence now, but it also worked without the raspi64.sh script (now I am using it with the script of course).
I think overclocking the Pi4 up to 2000Mhz boosts the performance a bit and it is also possible to play Minecraft on Ubuntu 20.10 (1.8.9: average: 25fps,max 60fps) and other apps, like Firefox work pretty well. 
But its right, the Gnome desktop is not my favorite, especially when you have a bigger screen. I have no idea about Linux Mint screen, but I can recommend you KDE plasma. 

David

morrownr, 2020-12-30:
There is a lot to like about KDE. I tested it last about a year ago. It was buggy buggy buggy. Hopefully some progress has been made on the front. The problem with KDE as it relates to running on a Raspberry Pi is the same as Gnome. It is far too heavy for the platform. The RasPis are fun but don't have a lot of power. About the best desktop gui interface one can use is the one used on the Raspberry Pi OS. Canonical should probably look at supporting the RasPi 4b and later with something like xfce.

Glad you got things lined out. If you find time to make suggestions regarding things here, let me know.

You can close this issue out anytime you are ready.

Nick

Chaosdave34, 2020-12-31:
You should try KDE on Raspberry pi again. They have changed the Problem with the many bugs and I am usingit like half an year now and I can't complain now. Please read the following article: https://www.forbes.com/sites/jasonevangelho/2019/10/23/bold-prediction-kde-will-steal-the-lightweight-linux-desktop-crown-in-2020/?sh=325e528226d2.

Thanks again for the help.
I will probably read through the Readme and will tell you any good suggestions and then I will close it.

David

closed 2021-01-03 by Chaosdave34