morrownr/88x2bu #30: FTBS: ‘BIT_APP_FCS_8821C’ undeclared

Ygrex, 2020-12-28:
Building from the latest commit 0c094b2ef8dc90b30b9ac62a6f7eab966d33fe96 results in
```
rtw_recv.c:3928:34: error: ‘BIT_APP_FCS_8821C’ undeclared (first use in this function); did you mean ‘BIT_APP_FCS_8822B’?
```
steps to reproduce:
```
$ sudo ./install-driver.sh 
Installing rtl88x2bu-5.8.7.4
Copying source files to: /usr/src/rtl88x2bu-5.8.7.4
Copying 88x2bu.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl88x2bu/5.8.7.4/source ->
                 /usr/src/rtl88x2bu-5.8.7.4

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' -j4 KVER=5.9.10 KSRC=/lib/modules/5.9.10/build................................(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.8.7.4 not found
Error! Bad return status for module build on kernel: 5.9.10 (armv7l)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.4/build/make.log for more information.
An error occurred while running: dkms build : 10
```
and logs:

```
$ cat /var/lib/dkms/rtl88x2bu/5.8.7.4/build/make.log
DKMS make.log for rtl88x2bu-5.8.7.4 for kernel 5.9.10 (armv7l)
Tue 29 Dec 2020 01:10:41 AM MSK
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.9.10/build M=/var/lib/dkms/rtl88x2bu/5.8.7.4/build  modules
make[1]: Entering directory '/mnt/media/backup/linux-5.9.10'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_sta_mgt.o
/var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_recv.c: In function ‘fill_radiotap_hdr’:
/var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_recv.c:3928:34: error: ‘BIT_APP_FCS_8821C’ undeclared (first use in this function); did you mean ‘BIT_APP_FCS_8822B’?
 3928 |  if (rtw_hal_rcr_check(padapter, BIT_APP_FCS_8821C))
      |                                  ^~~~~~~~~~~~~~~~~
      |                                  BIT_APP_FCS_8822B
/var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_recv.c:3928:34: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:283: /var/lib/dkms/rtl88x2bu/5.8.7.4/build/core/rtw_recv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1784: /var/lib/dkms/rtl88x2bu/5.8.7.4/build] Error 2
make[1]: Leaving directory '/mnt/media/backup/linux-5.9.10'
make: *** [Makefile:2419: modules] Error 2
```

morrownr, 2020-12-29:
Okay, I violated rule #1 which states that I should never commit a change when I don't have time to test. I have reverted the problem and will readdress it later. It shows to be compiling good here. Please uninstall and delete the driver you downloaded, then redownload and try again. Sorry about that.

Ygrex, 2020-12-29:
thank you, cb9e0c5e599241199dc02130e73263cb0e7880de works for me

closed 2020-12-29 by Ygrex