morrownr/88x2bu #31: Cudy WU1300S (rtl8822bu) Problems

TheWranger, 2020-12-30:
Hey, I've tried using this driver with my Customer WU1300S and haven't been able to get it work in a proper manner
- Can't connect to 5GHz (despite them appearing), only 2.4GHz connections
- Connection speed on 2.4GHz connection is incredibly slow and inconsistent
- Disconnecting the adapted and putting it in a different USB port might stop it from being detected at all
I'm on Linux Mint 20 with kernel 5.8.0-33 generic. The ID in lsusb is just 0bda:b812

I tried also using the driver listed on Cudy's website but I am having the same issues with that driver too, so I'm not certain this is an issue with this driver or if it's something related to my machine causing this.

TheWranger, 2020-12-30:
Tried disabling power saving using sudo ./edit-options.sh but after a reboot it has not had any effect on the poor performance

morrownr, 2020-12-30:
Near the end of the README.md (the main screen) are two sections that might be of interest. One is about recommended router settings and the other is info regard USB. Given those a look and also make sure you delete any old networks connections:

- lower right screen, left click on the wifi signal, selection network connects, select each network connection under WiFi, click on the minus for each until all are gone, try connecting again.

See if any of that helps.

TheWranger, 2020-12-30:
I can use both 5GHz and 2.4GHz connections fine and at full speed when using other devices so I think my router settings are okay. My issue with swapping USB ports seems to be because of putting in my front ports instead of the rear, though.

I cleared all my current connections the

TheWranger, 2020-12-30:
`wlxb44bd6248cfa  IEEE 802.11bgn  ESSID:"TP-LINK_F8CC"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.412 GHz  Access Point: C4:71:54:35:F8:CB   
          Bit Rate:144.4 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=99/100  Signal level=-35 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
`
is what `iwconfig` returns. Sorry about closing the thread, I'm having to do alot of this from my phone and I accidentally clicked it.

morrownr, 2020-12-30:
Set the log level to 4 and reboot. Use the system for about 5 minutes and then run save-log.sh. The log will be called rtw.log. Post the contents here. Let's see if we can see anything in the log.

Post the contents of 88x2bu.conf...not the one in the driver directory but the one that is operating on the driver.

iwconfig is telling us that you don't have an 802.11ac connection. That is interesting.

TheWranger, 2020-12-31:
Here is config
```
# /etc/modprobe.d/88x2bu.conf
#
# Purpose: Allow easy access to specific driver options
#
# Edit the following line to change options:
options 88x2bu rtw_drv_log_level=4 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_switch_usb_mode=0 
#
# -----
#
# Log level options: ( rtw_drv_log_level )
#
# 0 = NONE (default)
# 1 = ALWAYS
# 2 = ERROR
# 3 = WARNING
# 4 = INFO
# 5 = DEBUG
# 6 = MAX
#
# Note: You can save a log of RTW log entries by running the following in a terminal:
#
# $ sudo ./save-log.sh
#
# -----
#
# LED control options: ( rtw_led_ctrl )
#
# 0 = Always off
# 1 = Normal blink (default)
# 2 = Always on
#
# -----
#
# VHT enable options: ( rtw_vht_enable )
#
#  0 = Disable
#  1 = Enable (default)
#  2 = Force auto enable (use caution)
#
# Notes:
# - Unless you know what you are doing, don't change the default for rtw_vht_enable.
# - A non-default setting can degrade performance greatly in some operational modes.
# - For AP mode, such as when you are using Hostapd, setting this option to 2 will
#   allow 80 MHz channel width.
#
# -----
#
# Power saving options: ( rtw_power_mgnt )
#
# 0 = Disable power saving
# 1 = Power saving on, minPS (default)
# 2 = Power saving on, maxPS
#
# Note: 0 may be useful in unattended server setups or if dropouts are experienced.
#
# -----
#
# USB mode options: ( rtw_switch_usb_mode )
#
# 0 = No switch (default)
# 1 = Switch from usb 2.0 to usb 3.0
# 2 = Switch from usb 3.0 to usb 2.0
#
# Note: When changing USB options, a cold boot is recommended.
#
# -----
```
Here is log at level 4 whilst being connected to 2.4ghz connection
[... 804 lines cut ...]
 RTW: recv_decache: tid=0 seq=2734 frag=0
 RTW: Turbo EDCA =0xa42b
 RTW: recv_decache: tid=0 seq=2751 frag=0
 RTW: Turbo EDCA =0x5ea42b
 RTW: recv_decache: tid=0 seq=2791 frag=0
 RTW: Turbo EDCA =0xa42b
 RTW: Turbo EDCA =0x5ea42b
 RTW: Turbo EDCA =0xa42b
 RTW: Turbo EDCA =0x5ea42b
 RTW: recv_decache: tid=0 seq=2836 frag=0
 RTW: recv_decache: tid=0 seq=2857 frag=0
 RTW: Turbo EDCA =0xa42b
 RTW: Turbo EDCA =0x5ea42b
 RTW: Turbo EDCA =0xa42b
 RTW: Turbo EDCA =0x5ea42b
 RTW: recv_decache: tid=0 seq=2879 frag=0
 RTW: Turbo EDCA =0xa42b
 RTW: Turbo EDCA =0x5ea42b
 RTW: recv_decache: tid=0 seq=2901 frag=0
 RTW: Turbo EDCA =0xa42b
 RTW: Turbo EDCA =0x5ea42b
 RTW: Turbo EDCA =0xa42b
 RTW: Turbo EDCA =0x5ea42b
 RTW: recv_decache: tid=0 seq=2957 frag=0
```

morrownr, 2020-12-31:
Good job providing information. Here are some thoughts:

1. Change action line in the options file to read:

options 88x2bu rtw_drv_log_level=4 rtw_led_ctrl=1

Yes, delete the last 3 items and then do a cold boot. See what happens.

2. The stable kernel for Linux Mint 20 is 5.4, not 5.8. Operating systems are incredibly complex things. You are using software and utilities that are tested on kernel 5.4 but a kernel that is about a year newer. It is fun to play but this is inviting issues such as this. Recommend you move back to kernel 5.4 and see what happens.

For what it is worth: I have tested this driver with kernel 5.4 and with kernel 5.8. With kernel 5.8, the testing was done on Ubuntu 20.10, not Mint 20. Ubuntu 20.10 was tested extensively with kernel 5.8 and that is the stable kernel that it uses. Mint 20.1 is due out soon and it will still use kernel 5.4. I'd recommending sticking with the kernel that the Mint developers use.

Question: This adapter is USB 3 capable. How are you connecting to your computer? Directly or with a cable? If you are using a cable, is the cable USB 3 capable? Is the slot you are hooking it into a USB 3 port?

TheWranger, 2020-12-31:
I set my configuration options to what you listed and haven't experienced any change. Reverting to kernel 5.4.0-58 still doesn't anything, either. I'm using a USB 3.0 port at the rear of my computer for the wifi adapter

TheWranger, 2020-12-31:
Here is a new log with the aforementioned changes, after a coldboot and running for 5 mins:
https://pastebin.com/8xzfi1Fc
I also feel I should mention that the connection is so slow that I commonly cannot even use google's speedtest - I just get an error. I type `ping google.com` and I get an error about an unknown hostname, but then after I use USB tethering from my phone to connect to the router I have no problem pinging google (in the log I did not use USB tethering).

morrownr, 2020-12-31:
There is no doubt that there is a problem. You have selected the right driver based on the device ID. The driver is installing. I am currently checking your log vs mine as I am using the same driver on this system I am typing on right now. I am also using Linux Mint 20. 

I am now asking myself, what would I do if I faced the same situation. The answer is that I would immediately dive into the router... remember the part where I recommended that you look at the section in the README that talks about router settings? I remember you saying something like "my other things work with the router therefore it can't be the router."

Okay, I get that what you said seems to make sense... the problem is that it isn't true. I've been using Linux for about 25 years now. Please humor me and use my list of recommended routers settings so we can eliminate that as a possible problem. I'm here if you need to ask questions.

TheWranger, 2020-12-31:
I cycled through all of your recommended router settings, and by doing so I was able to get my 5GHz connection working and at full-speed too. Apologies for not doing this sooner. I still have the same problems with my 2.4GHz connection though, even after cycling through all of the recommended settings the speed is still incredibly poor and I struggle to load websites

morrownr, 2020-12-31:
Okay, it is good to make a little progress. No need to apologize, we all get locked in once in a while.

The next thing I would do at this point is load up a "wifi analyzer" app on my smart phone to take a look at congestion on the 2.4g band. Figure out which of the channels 1, 6 and 11 is least congested and set that as a fixed channel in the router. Reboot the router.

Recommend you read this article:

https://metis.fi/en/2017/10/txpower/

I had noticed your signal was very strong...maybe too strong.

Also, recommend you add "rtw_switch_usb_mode=1" to your 88x2bu.conf and reboot. Make sure that your adapter is in a USB 3 port.

Let's see what happens.

TheWranger, 2021-01-02:
Thank you for sharing the article regarding transmit power, but even after decreasing the power transmitted it didn't help much with the connection. It's definitely something I'll keep on a lower setting though, now that I've read more about it. I also didn't experience any change after editing the configuration and changing the USB mode.

I think I'll just close this issue and stick with using the 5GHz connection. I doubt I will end up at all needing the 2.4GHz connection with this adapter and it seems almost certain that this is an issue with my setup, not an issue with the driver. If for whatever reason someone else has this problem, this issue could always just be referenced. I hope I did not take up too much of your time, thank you so much for all of your help.

closed 2021-01-02 by TheWranger