morrownr/88x2bu #33: Installed the driver, AP's showing up but can't connect

equinix5, 2021-01-05:
Hi, thanks for your time in creating this driver =)

I'm running Mint 20 Cinnamon (4.6.7) Kernel 5.4.0-58-generic.
I've got a Comfast CF-812AC (Chipset RTL8812BU)
I've followed your instructions and when i go to the network icon I can see "realtek 802.11ac NIC" and AP's but I get a connection failed error when I try to connect it keeps saying Auhentication required and it won't connect.

There is another wifi adapter installed on this computer (will move the hard drive into another computer that doesn't have onboard wifi). When I disconnect the Qualcomm Atheros wifi adapter (see screenshot) it will also disable the "USB Wi-Fi"

Any idea how to get this working ?

[image]

morrownr, 2021-01-05:
There should be no problem having both the Atheros device and the Realtek adapter in the machine at the same time. Here is what I would do:

Select "Network Connections" - it is an option under the little wifi antenna in the lower right of the screen

Under "WiFi", select and then delete (with the minus character) all WiFi connections

 Now go back to the main network selector like you have pictured above

Turn the Atheros device off

Turn the Realtek device on

Select the ap you want to connect to

Tell me what happens

Side note: You might want to take a look at the "Recommended WiFi Settings" part of the README. Router settings often make things more difficult than they should be.

morrownr, 2021-01-27:
Hit and run.

Closed.

closed 2021-01-27 by morrownr

equinix5, 2021-01-27:
> Hit and run.
> 
> Closed.

Hi..​ My apologies for not getting back to you.. I tried what you've mentioned and it didn't work... 

I was doing this to help a friend using teamviewer... Told him to buy another USB wifi adapter as he was being inpatient 🤷🏻‍♂️