morrownr/88x2bu #34: Error when installing to RaspbianOS

Jordonbc, 2021-01-19:


morrownr, 2021-01-19:
Problems with the 64 bit version of Raspberry Pi OS are not new. However, maybe now is a good time to take another look. I'll see about loading it up for testing.

Can I get you to run and post the results from the following? ( from the driver directory )

$ make clean
$ make

As you can probably tell, I've been down this road before. Not sure what the problem with the Raspberry Pi engineers is but they sure seem to be struggling with getting a 64 bit version successfully out the door. If you need your pi working now, use the 32 bit version of the OS.

Regards

Jordonbc, 2021-01-19:
Yeah, I know 64 bit is still in beta but I use my pi as a media server which can use more than 4 gigs and unfortunatly I cant use ethernet.

This is the output of those 2 commands:
```
root@piserver:~/src/88x2bu# make clean
#make -C /lib/modules/5.10.5-v8+/build M=/root/src/88x2bu clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko */*/*/*.ur-safe
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko */*/*.ur-safe
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko */*.ur-safe
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko */*.ur-safe
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions *.ur-safe

root@piserver:~/src/88x2bu# make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.10.5-v8+/build M=/root/src/88x2                                                      bu  modules
make[1]: Entering directory '/root/linux-a896a00ebffd0b016627f65ff236b660e66e782                                                      b'
  CC [M]  /root/src/88x2bu/core/rtw_cmd.o
  CC [M]  /root/src/88x2bu/core/rtw_security.o
  CC [M]  /root/src/88x2bu/core/rtw_debug.o
  CC [M]  /root/src/88x2bu/core/rtw_io.o
  CC [M]  /root/src/88x2bu/core/rtw_ioctl_query.o
  CC [M]  /root/src/88x2bu/core/rtw_ioctl_set.o
  CC [M]  /root/src/88x2bu/core/rtw_ieee80211.o
  CC [M]  /root/src/88x2bu/core/rtw_mlme.o
  CC [M]  /root/src/88x2bu/core/rtw_mlme_ext.o
  CC [M]  /root/src/88x2bu/core/rtw_mi.o
  CC [M]  /root/src/88x2bu/core/rtw_wlan_util.o
  CC [M]  /root/src/88x2bu/core/rtw_vht.o
  CC [M]  /root/src/88x2bu/core/rtw_pwrctrl.o
  CC [M]  /root/src/88x2bu/core/rtw_rf.o
  CC [M]  /root/src/88x2bu/core/rtw_chplan.o
  CC [M]  /root/src/88x2bu/core/rtw_recv.o
  CC [M]  /root/src/88x2bu/core/rtw_sta_mgt.o
  CC [M]  /root/src/88x2bu/core/rtw_ap.o
  CC [M]  /root/src/88x2bu/core/mesh/rtw_mesh.o
  CC [M]  /root/src/88x2bu/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/src/88x2bu/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/src/88x2bu/core/rtw_xmit.o
  CC [M]  /root/src/88x2bu/core/rtw_p2p.o
  CC [M]  /root/src/88x2bu/core/rtw_rson.o
  CC [M]  /root/src/88x2bu/core/rtw_tdls.o
  CC [M]  /root/src/88x2bu/core/rtw_br_ext.o
  CC [M]  /root/src/88x2bu/core/rtw_iol.o
  CC [M]  /root/src/88x2bu/core/rtw_sreset.o
  CC [M]  /root/src/88x2bu/core/rtw_btcoex_wifionly.o
  CC [M]  /root/src/88x2bu/core/rtw_btcoex.o
  CC [M]  /root/src/88x2bu/core/rtw_beamforming.o
  CC [M]  /root/src/88x2bu/core/rtw_odm.o
  CC [M]  /root/src/88x2bu/core/rtw_rm.o
  CC [M]  /root/src/88x2bu/core/rtw_rm_fsm.o
  CC [M]  /root/src/88x2bu/core/rtw_rm_util.o
  CC [M]  /root/src/88x2bu/core/efuse/rtw_efuse.o
  CC [M]  /root/src/88x2bu/os_dep/osdep_service.o
  CC [M]  /root/src/88x2bu/os_dep/linux/os_intfs.o
  CC [M]  /root/src/88x2bu/os_dep/linux/usb_intf.o
  CC [M]  /root/src/88x2bu/os_dep/linux/usb_ops_linux.o
  CC [M]  /root/src/88x2bu/os_dep/linux/ioctl_linux.o
  CC [M]  /root/src/88x2bu/os_dep/linux/xmit_linux.o
  CC [M]  /root/src/88x2bu/os_dep/linux/mlme_linux.o
  CC [M]  /root/src/88x2bu/os_dep/linux/recv_linux.o
  CC [M]  /root/src/88x2bu/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /root/src/88x2bu/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /root/src/88x2bu/os_dep/linux/wifi_regd.o
  CC [M]  /root/src/88x2bu/os_dep/linux/rtw_android.o
[... 103 lines cut ...]
  CC [M]  /root/src/88x2bu/hal/btc/halbtc8822bwifionly.o
  CC [M]  /root/src/88x2bu/hal/btc/halbtc8822b1ant.o
  CC [M]  /root/src/88x2bu/hal/btc/halbtc8822b2ant.o
  CC [M]  /root/src/88x2bu/platform/platform_ops.o
  CC [M]  /root/src/88x2bu/core/rtw_mp.o
  LD [M]  /root/src/88x2bu/88x2bu.o
  MODPOST /root/src/88x2bu/Module.symvers
ERROR: modpost: "__const_udelay" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_from_user" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "_raw_spin_unlock" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_lock" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_unlock" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "cpu_hwcap_keys" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "csum_ipv6_magic" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__warn_printk" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "arm64_const_caps_ready" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__udelay" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "preempt_schedule_notrace" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_to_user" [/root/src/88x2bu/88x2bu.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:111: /root/src/88x2bu/Module.symvers] Error 1
make[2]: *** Deleting file '/root/src/88x2bu/Module.symvers'
make[1]: *** [Makefile:1712: modules] Error 2
make[1]: Leaving directory '/root/linux-a896a00ebffd0b016627f65ff236b660e66e782b'
make: *** [Makefile:2420: modules] Error 2
```

morrownr, 2021-01-19:
A guy using the 8814au driver here reported what looks like the same issue with the Ubuntu mainline kernel v5.10.7 about 2 days ago. Let me see what I can figure out.

morrownr, 2021-01-19:
I just burned the 64 bit beta of Raspberry Pi OS to an sd card and it is updating now. I expect to be seeing what you are seeing soon.

morrownr, 2021-01-19:
Test complete. I have good news and bad news.

The good news is that the installation worked fine and the adapter is connected to the internet and I am getting good speed.

The bad news is that I was not able to duplicate your problem so figuring out what is different between my setup and yours is what we are going to have to do. Here is what I did:

- Erase a 32gb sd card
- Flash the img of the 2020-08 Raspberry Pi OS 64 bit beta (my hardware is a 4gb RasPi 4b)
- $ sudo apt update
- $ sudo apt upgrade
- $ sudo reboot
- I followed the instructions exactly - https://github.com/morrownr/88x2bu
- I rebooted, put the password in and called up cnn.com to read the news

You could run the remove-driver.sh script, then reboot and try again.

I have seen that Module.symvers problem before and it seems it is related to a headers issue. Could it be that the headers have been updated between the time you had the problem and I tested? Who knows? Clean up and try again. It is hard to figure out what the problem is when you can't duplicate it.

Jordonbc, 2021-01-20:
Just did the exact same thing on my pi 4b 8GB and it works, which means it must be a conflict with what I had installed. Im going to test with some other software that I use like open media vault.

I'm not getting the full speed of my TP-Link Archer T3U, around 120 mbps AC with power savings off and usb3 on.

Anyways thanks for getting it working :)

I will report back here if I find any conflicts.

morrownr, 2021-01-20:
Please do come back and add any information that might be of value to others. I think we should leave this issue open for others to read. I can mark it solved.

As far as speed is concerned: The 550 Mb/s speed shown in the README is using amd64 equipment. I have never been able to get close to that with Raspberry Pi hardware so I suspect the Pi hardware has some limitations. Turning USB 3 on is a good thing. I haven't seen a good reason to turn power saving off as power saving just works in this driver. You might want to optimize the settings in your wifi router to see if you can get some more speed.

Cheers and good luck.

crgorect, 2021-02-03:
Just a fun fact we've been learning on our end as we talk to various chip manufacturers. It seems turning on USB 3.0 can have a pretty negative effect with 5Ghz frequencies. It seems it can cause various noise issues with a lot of the current chip sets out there

morrownr, 2021-02-09:
I have read the same thing.  USB 3 is not mankind's greatest creation.

closed 2021-12-05 by morrownr