morrownr/88x2bu #35: "Operation not permitted" Raspberry Pi 4 Kali Linux

Starman0620, 2021-01-19:
Hello, I've been trying to get my rtl8812bu adapter working with Kali Linux on my Pi 4B and I was recommended this driver, however upon successfully building/installing it, I'm unable to do anything with the adapter. 
If I try to bring it UP, set it into monitor mode, etc it throws an "Operation not permitted" error. I've run the 64 bit Raspberry Pi script and the fix described in #8 to no avail (though #8 did fix build errors I had)

morrownr, 2021-01-19:
Hi. Can you clarify that you are running the 64 bit version of Raspberry Pi OS?

Does the adapter work in client mode?

Starman0620, 2021-01-19:
> Hi. Can you clarify that you are running the 64 bit version of Raspberry Pi OS?
> 
> Does the adapter work in client mode?

It's not Raspberry Pi OS, it's Kali. But yes, it is the 64 bit version.

No, it does not work whatsoever, the only sign of it even existing in software is when I run `ip a`

morrownr, 2021-01-19:
Okay, the ARM64 version of Kali Linux. I see that you said that twice now. Sorry for the non attention to detail.

So, we need to figure out where things are going wrong:

Did the onscreen output of install-driver.sh tell you that the driver installed successfully?

From the driver directory, run:

$ sudo dkms status

and post the results.

From the driver directory, run:

$ make clean
$ make

and post the output of $ make

That should be a good start.

Starman0620, 2021-01-19:
dkms status: rtl88x2bu, 5.8.7.4, 4.19.118-Re4son-v8l+, aarch64: installed

[buildlog.txt](https://github.com/morrownr/88x2bu/files/5838613/buildlog.txt)

morrownr, 2021-01-19:
I can see that I am going to have to try to duplicate this problem.

Can you provide me the exact link to the ARM64 version of Kali that you are using?

Thanks

Starman0620, 2021-01-19:
> I can see that I am going to have to try to duplicate this problem.
> 
> Can you provide me the exact link to the ARM64 version of Kali that you are using?
> 
> Thanks

https://images.kali.org/arm-images/kali-linux-2020.4-rpi4-nexmon-64.img.xz

morrownr, 2021-01-20:
I downloaded the image and burned it to an sd. After booting with my rpi4b, I did an update and upgrade and something went terribly wrong. Won't boot now. Will try again tomorrow if I have time.

morrownr, 2021-01-20:
I was able to burn the Kali 64 for RasPi image and install it and update it. Whew, I should get paid by the hour. 

Report: I hooked up my 8812bu adapter via usb 3 cable to my raspi4b. I followed the installation instructions on this site to include running raspi64.sh. The driver compiled with one minor compiler warning which should be no problem. The driver installed. I turned on usb 3 support per instructions. I connected to my access point with password. All indications were normal until I tried to connect to the internet. No go. I then tried the onboard wifi and it didn't work either. I did some checking and indications are that this driver is doing its job. This driver can't fix broken networking.

Conclusion: Something is really wrong but all indications are that it is not this driver... that the problem is somewhere in Kali. I will set the sd card with Kali on it aside for now and if you can bring evidence that the driver is causing problems, I'll be glad to investigate further.

Recommendation: Use Raspberry Pi OS. Install the apps you need on it.

Regards.

Starman0620, 2021-01-20:
Alright, I appreciate your help. Thanks

closed 2021-01-20 by Starman0620

Starman0620, 2021-01-21:
Oh yeah, figure I should mention this here since it might help. I believe the re4son kernel that Kali Pi comes with on default has a module for rtl8812au which could be interfering. Even with a completely fresh installation of Kali Pi (32 and 64 bit) `ip a` still detects my rtl8812bu adapter whereas with Pi OS, that is not the case. This might be something worth looking into for future users perhaps? Just a thought.

morrownr, 2021-01-21:
The 8812au driver should not be a problem. My main test systems here have all 5 of the Realtek drivers that I maintain here installed and they don't interfere with each other. Please do report if you find a solution.