morrownr/88x2bu #36: Cudy AC1300 WU1300S

Jimboker86, 2021-02-06:
Well I went with the Cudy WU100s like many recommended. I am so close to finally having my WIFI set up I can feel it. I'll keep working on it regardless. I am stuck on Part 5 of this Install if you have any more advice to offer. I have the driver on CD, USB, can download from the Internet using a Hotspot, and have it in a tar.gz file through /Home/Jimmy/Project/. For whatever reason I can't seem to access once I switch to the Terminal. No such file or directory. I figured the CD only has Windows/MAC driver. Should I do root@linux since I'm on Linux Mint and not Ubuntu?  https://newwezhanoss.oss-cn-hangzhou.aliyuncs.com/contents/sitefiles2019/10096568/files/281895..pdf?Expires=1612638716&OSSAccessKeyId=LTAIekGM1705vEQp&Signature=9heL83c+ZueYqtoPEFIop1dca2k=&response-content-disposition=inline;filename=RTL88x2BU%20linux%20Installation%20guide.pdf&response-content-type=application/pdf

morrownr, 2021-02-07:
Jimbroker86.

Sir, I need you to slow down and take a deep breathe. It would help if you would stop installing whatever you are installing until we have a talk about what you are doing...you are going to make a mess...

Now, about the only things I am clear on from your message is that you have bought yourself a Cudy USB WiFI adapter that is model number WU1300S and you are trying to install it on something that runs Linux. Beyond that, I am not following you so let me ask some questions:

What make and model of computer are you installing this adapter to?

What make and version of Linux is installed on said computer?

Info: please do not install any driver you get on a CD to Linux! Period. Linux does not work like Windows.

Info: Your WU1300S contains a Realtek 8812bu chipset. Knowing the chipset of the adapter what kernel you are running are keys to success.

What kernel are you running? Press and hold Ctrl+Alt then press T to open a terminal. Type the following at the command prompt:

uname -r

Post the results in a reply.

The chances are about 99.99% that the following driver will work for you:

https://github.com/morrownr/88x2bu

But you have to scroll down and read the Installation Information and the follow the Installation Steps.

Do not use the driver from a CD as the chance of it working are not good.

Let me conclude this message with a recommendation:

Most people that run Linux or want to run Linux are better off with USB WiFi adapters that have in-kernel support. That means the adapter is plug-n-play. You don't have to do anything beyond sticking the adapter in a USB port. You can get more information and recommended adapters at this site:

https://github.com/morrownr/USB-WiFi

morrownr, 2021-02-11:
Hit and run is not appreciated.

Jimboker86, 2021-02-13:
Dear morrownr,

Sorry wife will only let me work on converting this computer over to Linux
on the weekends.  I'm trying to stick with it but this is my 2nd WIFI
adapter I've tried to install and she is losing patience.

Make and model ASUS CM1630-06
Linux Mint 20

uname -r revealed 5.4.0-58-generic

Yes you are correct WU1300S contains a Realtek 8812bu chipset.  I tried
following the Manufacturer's guidelines got stuck then had to search other
out of date videos for answers.  Really wish I would have opted for a
plug-in-play adapter.  I will steer others that way for sure.

I would greatly appreciate any help you can offer.  I didn't think this was
such a big project when I started.  If I could just get this internet going
on Linux feel good about converting still.

Sincerely,
Jimboker86

morrownr, 2021-02-14:
JimBroker86,

I apologize for the "hit and run" comment. I've had a few folks lately that have disappeared after stopping by and requesting support. That is not cool.

I understand about the wife. I can't help you there. Good luck.

Now to business: 

Thank you for providing the computer make and model. I downloaded and read the system manual from the Asus website.  This system should run Linux Mint 20 well. It does appear that is only has USB 2 ports. That is something to keep in mind if installing a Realtek driver. Do not turn on USB 3 support with the Realtek drivers if you don't have a USB 3 port or things will go south in a hurry.

Let me reiterate in a different way. To have USB 3 support, you need 4 things:

- Adapter that supports USB 3 (good)
- Driver that supports USB 3 (good, but has to be turned on)
- Port that supports USB (not good. system doesn't have it. forget USB 3)( USB 2 is very fast anyway so who cares?)
- A cable, if used, that supports USB 3 (not used as far as I can tell)

In your case, you have an adapter that supports USB 3 but your computer does not. so do not turn USB 3 support on in the driver.

Also, keep in mind, not all adapters will work in all ports. If you think things are good to go and things aren't working, try another port.

With that said, I have no information about the current state of your Linux Mint installation. I suspect you have a little bit of a mess on your hands. Remember this: Never install a Linux driver from a CD. Never. Now, how we proceed is up to you.

You have a computer that should work wonderfully with Linux Mint. The questions appear to be:

- How do you get your frustration level down?
- How do you keep your wife happy?
- How do you get wifi working?

I can only offer what I would do given what I know about your situation. The first thing I would do is go to...

 [https://github.com/morrownr/USB-WiFi](url)

... and look at the low cost USB 2 adapters that have in-kernel support and order one. Keep in mind that even N150 adapters are still pretty fast when it comes to surfing the internet. That will do 2 things for you: It will get you on the internet without the muss and fuss of dealing with an adapter with out-of-the=kernel drivers and you will always have it handy if you install Linux to other computers in the future. I keep a couple of N150 adapters around. They are handy. My wife's system is about the same age as your computer. It has a N150 adapter based on the ar9271 chipset. It only has USB 2 ports. My wife is a heavy internet user... online radio, videos and various other things going at the same time and the internet never bogs down on that system. While it sounds cool to get the latest AC1200/AC1300/C1900 adapters, for the most part, that is overkill.

The second thing I would do is download Linux Mint Cinnamon 20.1 and update your installation medium...flash drive or DVD. At that point I would research how to do a complete wipe of the existing Linux installation while installing this new version. I wouldn't actually do the installation until I get that new N150 adapter and stick it in a port. It is handy to have internet access while installing.

 If you are determined to use the current Linux installation, I'll try to help but it could be painful.

Jimboker86, 2021-02-15:
Hey morrownr,

Thanks for your reply!  I'm reading it over trying to figure out what to
do.  You were the first person to ask me the make and model of my
computer.  This seems to be your expertise.  The ASUS tower right?  I have
a bunch of USB ports on my computer.  You're saying that there's a big
difference between USB 2.0 and USB 3.0.  USB 3.0 is a newer generation of
technology that won't work well with my older computer.

I tried to deduce this information talking to other Linux users.  The
problem is being a noob Windows user it's hard to decipher which anonymous
Linux user I can trust to follow.  I went with the Cudy AC1300 WU1300S for
that very reason.  The person who recommended it probably has a newer
computer.  I had originally purchased a Bros Trend WIFI adapter and
returned it.  This has been my biggest complaint about Linux.  There seem
to be no American made Linux compatible WIFI adapters because
Microsoft/Windows won't allow it.  From an American perspective I'm kind of
forced to settle on the Chinese made Linux compatible WIFI adapters.
Really difficult to circumvent Microsoft/China.

I have another Linksys WIFI adapter on the back of the tower that always
worked with Windows.  I'm trying not to be that person that gives up and
goes back to Windows.  If I finally clear this last WIFI hurdle I'll be
sure to post about it and let others know.  But yeah I got the two kids in
diapers so I can't always work on this project as much as I would like.
During the week I can't get to the computer to work on it.

The easiest solution is to try and return my Cudy WIFI adapter and get a
USB 2 one w/ in-kernel support.  That sounds like sage advice.  Knowing my
computer which one would you recommend?

Based on your list I would select the Panda Wireless PAU05 300 Mbps
Wireless N USB Adapter.  Part of me picks this design cause I assume I can
just plug it in and it will work.  The last couple I've gotten like this
haven't worked.  You said you've had success with the Alfa AWUSO36NHA -
Wireless B/G/N USB Adaptor and working with that ar9271 chipset?

I'm fairly certain I have the newest version of Linux Mint which probably
is the Linux Mint 20.1.  Just converted in January.  Maybe I messed it up
trying to figure out all this WIFI stuff in the terminal.  I have a
Timeshift recorded when I first installed.

Thanks again for your reply.  I really appreciate the guidance.  This
probably is my last shot as this will be the 3rd WIFI adaptor I've tried.
Let me know your recommendation and I'll let you know how it goes.  No rush
replying will probably have to wait til the weekend.

Sincerely,
Jimmy K

morrownr, 2021-02-16:
Jimmy,

Please call me Nick. I can see we are making progress. I'd say it won't be that long until you are the Linux expert that is helping noobs. Linux is really fun once your knowledge level comes up a little and it can save you a lot of money in the long run.

I have several questions to answer and some issues to address so here goes:

Yes, the manual I was reading was for the Asus tower you told me that you had. Why did I download the manual? Because I need to know what I am dealing with. Does it have components to run the version of Linux you are trying to run? It has a good dual core 64 bit AMD processor, 4 gb of ram, and a AMD video card. What's not to like about that? That is good stuff for a Linux box. One thing I was paying careful attention to was the USB ports. All of your USB ports are USB 2. That is not a bad thing. USB 2 is a good thing. USB 3 can be faster but it is not the best design that mankind has come up with. USB 2 = good. USB 3 = can be problematic. USB 2 is plenty fast enough to handle the internet service most people have and it is more than fast enough to provide a good internet experience. On down the road you might think about changing the hard drive for an ssd but that is a subject for another day.

The Cudy adapter you bought is a USB 3 capable adapter. That is okay. It is will work in a USB 2 port...well, it should anyway. But in a USB 2 port, it will not and cannot work in USB 3 mode so the USB 3 support in the Realtek driver should not be turned on or it can cause problems. That is probably enough on this topic for now.

You asked me to make a recommendation. Here goes:

If your wifi router/access point is not more than maybe 25-30 feet away from your Asus tower and no more than one wall between the two, then Panda PAU05 should work fine. If you have more distance and maybe more than one wall, you might want to go for an adapter that has a bigger antenna:

Panda PAU06 - [https://www.amazon.com/Panda-Wireless-PAU06-300Mbps-Adapter/dp/B00JDVRCI0](url)

Another option:

Panda Mid Range 150Mbps Wireless N USB Adapter - [https://www.amazon.com/gp/product/B004AC0L4Y](url)

There are plenty of options shown at [https://github.com/morrownr/USB-WiFi](url) so take a look and if you have specific questions, let me know.

You said "Part of me picks this design cause I assume I can just plug it in and it will work." when talking about the Panda PAU05. You are correct. You plug it in and it will work as the others that have in-kernel support will. You will still need to select an access point and put in a password.

Concerning your version of Linux Mint, v 20.1 was just released in January. Tell me what this says:

Administration > Systems Reports > System Information > System

Mine says:

Kernel: 5.4.0-65-generic x86_64 bits: 64 compiler: gcc v: 9.3.0 Desktop: Cinnamon 4.8.6 
           wm: muffin dm: LightDM Distro: Linux Mint 20.1 Ulyssa base: Ubuntu 20.04 focal 

You  said there is a Linksys USB adapter on the back of the tower. I'm curious what it is. If you could type:

$ lsusb

...at a terminal (Ctrl+Alt+T) and then post the results back here I can probably figure out what it is. It would be cool if we could get it going. If you don't have internet access on the Asus, Maybe you could save the info in a text file and copy it to a flash drive and go to another computer.

I think I have spoke to most of the issues. Take your time and work the issues as you have time. Taking Linux out for a test drive is kinda like buying a new car. It has the same amount of wheels and tires but the knobs and buttons are in different spots. You'll get used to it.

Nick

Jimboker86, 2021-02-22:
Hi Nick,

Thanks again for this awesome email and I really appreciate
the recommendations!  I feel like we're making progress too.  I'm feeling a
lot better about Linux after your last email.  Just glad we have a plan to
go forward.  I should be able to return the Cudy device and get my money
back.  So no loss there.  The Panda PAU06 might be the way to go since this
ASUS is in the basement.  The router is right above it but the signal has
to pass through the floor.  Maybe a floor is too thick for the PAU05?  I'll
try to do some more homework before I purchase but feeling great about
either one of those options.

When I run Administration > Systems Reports > System Information > System

Mine says:
System:    Kernel: 5.4.0-58-generic x86_64 bits: 64 compiler: gcc v: 9.3.0
Desktop: Cinnamon 4.8.5
           wm: muffin dm: LightDM Distro: Linux Mint 20.1 Ulyssa base:
Ubuntu 20.04 focal

When I ran $ lsusb
$ lsusb
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 003: ID 058f:6362 Alcor Micro Corp. Flash Card Reader/Writer
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 002: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

Don't waste too much of your time trying to figure this one out but might
as well share.  I'll mostly write this off as an anomaly.  Oddly enough
when I turned on the computer for the first time to get the System reports
and lsusb info for you I had full internet.  The internet had a strong
signal for 15 minutes.  I was able to pull up 5 websites at once.  Play a
Youtube DJ Mix and the whole 9 yards.  After 15 minutes I turned the ASUS
off to see if the Wifi would pop back up after a restart but it didn't.
Turned the computer off again and the Wifi still didn't pop back again.  At
the time both the hotspots on my phone and labtop were off.  The Cudy Wifi
Adapter was already placed back in the box as I'm looking to return and was
not connected.  Only thing I can think of is I switched out a Logitech
Receiver for a keyboard which I share with my Wife to help her carpal
tunnel.  She had just been using the Logitech Receiver on her own device.
Probably just an anomaly.

Would be cool if we could get that Linksys adapter on the back of the tower
working.  I'll try to put an order in for the PAU06 later this week.  Once
again no rush on replying.  I really appreciate the guidance.

Sincerely,
Jimmy

morrownr, 2021-02-24:
Jimmy,

So you are on Mint 20.1. Good. I'd say either one of the Panda adapters would work for you but it is your house and you probably have a better idea how well radio signals travel there than I so take your time, research it, and make the decision.

I did not see a wifi device with $ lsusb. Can I get you to do that same thing with this command?

$ lspci

I like to tinker and  knowing you have another wifi device in the system makes me want to figure it out. The fact that it came on while you were running Mint tells me that the device is supported with in-kernel drivers. Could it be that the attenna is loose. Maybe you could tighten it up? Tell me about the device. I get the feeling that it is an internal card since we did not see anything with $ lsusb. 

Maybe we could delete some interfaces:

- lower panel
-  right side
- network icon
- click it with left mouse button
- Network Connections
- under WiFi, select each item one at a time and then click on the minus (-) in the lower part of the dialog

After you delete anything and everything there, go to Network Settings and let's see what we have.

Nick

Jimboker86, 2021-02-24:
Nick,

The wireless antenna on the back of the tower is Linksys from 10 years
ago.  You're probably right that it's an internal card.  I remember having
to open up the computer and attach it from the inside.  Sorry about that
should have mentioned something sooner.  My Dad was a little more
knowledgeable and helped with that.  He doesn't know much about Linux
though.

$ lspci revealed.

  $ lspci
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] RS780 Host Bridge
00:01.0 PCI bridge: ASUSTeK Computer Inc. AMD RS780/RS880 PCI to PCI bridge
(int gfx)
00:0a.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] RS780/RS880 PCI to
PCI bridge (PCIE port 5)
00:11.0 SATA controller: Advanced Micro Devices, Inc. [AMD/ATI]
SB7x0/SB8x0/SB9x0 SATA Controller [IDE mode]
00:12.0 USB controller: Advanced Micro Devices, Inc. [AMD/ATI]
SB7x0/SB8x0/SB9x0 USB OHCI0 Controller
00:12.1 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0 USB
OHCI1 Controller
00:12.2 USB controller: Advanced Micro Devices, Inc. [AMD/ATI]
SB7x0/SB8x0/SB9x0 USB EHCI Controller
00:13.0 USB controller: Advanced Micro Devices, Inc. [AMD/ATI]
SB7x0/SB8x0/SB9x0 USB OHCI0 Controller
00:13.1 USB controller: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0 USB
OHCI1 Controller
00:13.2 USB controller: Advanced Micro Devices, Inc. [AMD/ATI]
SB7x0/SB8x0/SB9x0 USB EHCI Controller
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 SMBus
Controller (rev 3c)
00:14.1 IDE interface: Advanced Micro Devices, Inc. [AMD/ATI]
SB7x0/SB8x0/SB9x0 IDE Controller
00:14.2 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 Azalia
(Intel HDA)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD/ATI]
SB7x0/SB8x0/SB9x0 LPC host controller
00:14.4 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 PCI to PCI
Bridge
00:14.5 USB controller: Advanced Micro Devices, Inc. [AMD/ATI]
SB7x0/SB8x0/SB9x0 USB OHCI2 Controller
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 10h
Processor HyperTransport Configuration
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 10h
Processor Address Map
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 10h
Processor DRAM Controller
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 10h
Processor Miscellaneous Control
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 10h
Processor Link Control
01:05.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI]
RS780L [Radeon 3000]
01:05.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] RS780 HDMI
Audio [Radeon 3000/3100 / HD 3200/3300]
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd.
RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 03)
03:05.0 Network controller: Ralink corp. RT2561/RT61 802.11g PCI

Can you tell what the device is by looking at all that?  I tried to copy
everything from Windows over to an external harddrive before switching to
Linux Mint.  I searched the external hard drive for Linksys but couldn't
find much.  I did scroll through and find a Realtek installer 64 with a
Date modified from 2009 which could offer more info.

I also went through the Network connections like you said and cleared out
everything under WIFI hitting the minus button.  Thanks for that tip.
Before I had talked with you I think I had gone through Network settings
and made a mess of things trying to find a solution.  After I minus
buttoned or deleted everything under WIFI the Internet didn't work.  I shut
down the computer turned it back on.  Reconnected to my WIFI.  On this 1st
Restart I got strong Internet for like 7-8 minutes.  The signal was at
75%.  I felt like you almost trouble shooted the problem there.  I
restarted another time.  No internet.  Restarted another time.  No
internet.  But I made sure to minus button everything under WIFI before I
shut down.  Restarted again.  Briefly acted like it would connect then No
internet.  Now I'm on my 6th restart and I was able to pull up like 3 web
pages at 62% signal.  7th Restart.  When I check WIFI it's switched over to
Auto-connect that seems to be when I have issues.  Signal jumping around
from 70-74% but no Internet.  Minus buttoned under WIFI.  Tried again.  No
Internet.

Just using the back of tower Linksys when I was still on Windows I could
get Internet but it was a weak signal.  When I look at the connection
strength through Linux I can see my Signal is jumping from 75% at best down
to 60%.  When I used Windows it would just slowly load whatever web page.
When I use Linux the web page times out faster and says connection lost.  I
tried tightening the antenna on the back as well.

There used to be an advanced options on Network connections.  I didn't
notice it this last time.  Maybe I need to make sure I get all of those
options back to default?  Maybe there's a way for Linux just to accept a
60% signal without constantly trying to connect?  I think something with
the Auto-connect could be causing that.  Might be time to open the computer
up?

We're getting close.

Thanks again.

Sincerely,
Jimmy

morrownr, 2021-02-24:
Jimmy,

This is great. You get to teach your dad about Linux. He sounds like a guy that likes to tinker so Linux should keep him entertained.

To answer your question: Yes, I can tell you what device you have from looking at the list that $ lspci produced. See this line:

  Network controller: Ralink corp. RT2561/RT61 802.11g PCI

That tells me that said adapter is an internal PCI card adapter with a Ralink chipset known as rt2561. This chipset is supported with an in-kernel driver. This adapter only supports up to 802.11g. My recommendation is for you to take a look at the section of the documentation (main screen of this repo) that is called...

Recommended Router Settings for WiFi

Pay particular attention to the router settings for 2.4g. You might also download an app called WiFi Analyzer if you have a Linux phone (I mean Android <g>.) Use this app to determine which of the 1, 6, 11 channels is least congested and set that channel fixed. For what it is worth, Linux can handle multiple wifi devices in use so having this adapter and a new Panda adapter in the same machine is no problem. This adapter is very old and you may not get very fast performance if the signal is not that strong.

FYI: Mediatek bought Ralink some years ago so anytime you see Mediatek and Ralink mentioned, it is the same company. Mediatek does a pretty good job of supporting drivers for their products in the Linux kernel so I recommend products based on Mediatek chipsets when asked.

The adventure continues.

Nick

Jimboker86, 2021-03-07:
> Android .) Use this app to determine which of the 1, 6, 11 channels is

What's up Nick,

Thanks a ton for this Panda PAU06 recommendation!  Just got it in the mail
a couple of days ago and got a free Saturday.  100% Wifi signal.  The
set-up was super easy.  Really just had to plug it in.  The right product
makes all the difference.  Stoked I can use this computer again.

Think I'm going like Linux quite a bit.  I'll be happy to get my Music
library set back up.  I'll give Clementine a shot.  Maybe files will be
easier to transfer w/o Itunes and I can get more creative.

I'll be sure to pay it forward letting other people know about the
difference between USB 2.0 and 3.0.  If you need help promoting a web page
or anything let me know.

I'll have to give Linux for Dummies a look over as well.  Don't want to
burden the Linux community with too many questions right away.  I
appreciate your patience.  Thanks for sticking with me and giving
guidance.  You're a good soul Nick!

Take care
Jimmy

morrownr, 2021-03-08:
Jimmy,

It is good to get a report from you. There for a little while I was wondering if you believed me when I told you the Panda PAU06 would work as soon as you plug it in. Glad to hear it worked out. You are right, the right product can make all the difference in the world.

You might want to check out Audacious as far as music apps go. It is my favorite but it depends on what features you need. There are a lot of good apps to help with music. I can suggest apps to rip CDs and convert formats and you should see a big drop off in ease of transfer. There is also the new ipTV app that Linux Mint just released. It should automatically flow into your system next time you do a $ sudo apt update && sudo apt full-uodate

Take care and stop by when you can.

Nick

closed 2021-12-05 by morrownr