morrownr/88x2bu #39: when open usb3.0 hostapd failed to function

danywheeler, 2021-02-16:
In Usb2.0 mode everything working fine, when switch to usb3.0 ap is not working with following error message:

nl80211: beacon_ies - hexdump(len=6): 7f 04 00 00 00 02
nl80211: proberesp_ies - hexdump(len=6): 7f 04 00 00 00 02
nl80211: assocresp_ies - hexdump(len=6): 7f 04 00 00 00 02
nl80211: Beacon set failed: -22 (Invalid argument)
Failed to set beacon parameters

I use rpi4 8gb with ubuntu server 20.04.2.

morrownr, 2021-02-16:
I have been trying to chase down similar issues when running hostapd with this driver. Can I get you to post the contents of these two files:

/etc/hostapd/hostapd.conf

/etc/modprobe.d/88x2bu.conf

danywheeler, 2021-02-16:
sure:

options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=1

which just switch usb3.0

hostapd.conf: 

ssid=Test
wpa_passphrase=123456789

bridge=br-lan
country_code=US
interface=wlx200db04cb65d
driver=nl80211
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
macaddr_acl=0

logger_syslog=0
logger_syslog_level=4
logger_stdout=-1
logger_stdout_level=0

hw_mode=a
wmm_enabled=1

# N
ieee80211n=1
ht_capab=[MAX-AMSDU-3839][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

# AC
ieee80211ac=1
vht_capab=[MAX-AMSDU-3839][SHORT-GI-80]
vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42

nothing special about it, switch to usb 3.0 nothing worked, wpa_supplicant & hostapd all have problem,  it's working, then it's like someone unplug it and plug it, in wpa_supplicant it connects and then reconnects, in hostapd it's the same, ap shows then it disappeared.

another odd behavior is the interface index always increase, after a few minutes it's ifindex becomes 40, and phy name also changes, it's like I said, like someone is unplug and plug the wifi adapter.

morrownr, 2021-02-16:
I'm going to assume since you are using the 88x2bu driver that you have an adapter based on a 8812bu chipset. Please advice is this is not correct. Based on my assumption:

Change  -  ht_capab=[MAX-AMSDU-3839][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

To  -  ht_capab=[MAX-AMSDU-7935][HT40+][SHORT-GI-20][SHORT-GI-40]

Reason: [DSSS_CCK-40] is not supported on band 2 and you are using a channel that is band 2.
Reason: run "$ iw list" to check  MAX-AMSDU for band 1.

Change  -  vht_capab=[MAX-AMSDU-3839][SHORT-GI-80]

To  -  vht_capab=[MAX-AMSDU-11454][SHORT-GI-80]

Reason: Check MAX-AMSDU for band 2 with "$ iw list"

Comment out the following two lines and see what happens:

vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

The 8812bu is a USB 3 capable adapter. Is it plugged into a USB 3 port? Do you use a cable? If yes, is it USB 3 capable? If using a cable, try without the cable.

Let me know the status.

Nick

danywheeler, 2021-02-16:
You are right about the config, I just copy from rpi onboard wifi hostapd config, In usb2.0 it's ok, so I'm not bother to change it, just for test, then switch to usb 3.0, it's not working properly, the adapter is usb 3.0 capable and is pluged in rpi4 usb 3 port.

danywheeler, 2021-02-16:
It's a power issue, when pluged at rpi4 bottom usb 3.0 port, it has no problem, however if pluged at top usb 3.0 port, with a hub it's working even without external power pluged in the hub. that's odd, anyway that's the way to solve it with the question I posted above.

closed 2021-02-16 by danywheeler