morrownr/88x2bu #41: MAC Randomization causing compile issue

gteca, 2021-02-18:
Hi,
I am trying to use MAC randomization by defining the macro #CONFIG_RTW_CFGVENDOR_RANDOM_MAC_OUI in drv.types.h, but during compilation I get:
ERROR: modpost: "rtw_hal_set_pno_random_get_mac_addr" [full_path] undefined!
ERROR: modpost: "rtw_hal_set_hw_mac_addr" [full_path] undefined!

morrownr, 2021-02-19:
Since #42 was solved and closed, does that mean this report should also be closed?

gteca, 2021-02-19:
> Since #42 was solved and closed, does that mean this report should also be closed?

This issue is still opened, the #42 was my workaround to the problem, like reinvent the wheel

morrownr, 2021-02-28:
Go to the Makefile and change this setting

CONFIG_RTW_WIFI_HAL = n

Good luck.

morrownr, 2021-03-08:
@gteca 

Checking to see if you got the above message.

gteca, 2021-03-10:
Hi @morrownr ,
In Makefile I changed `CONFIG_RTW_WIFI_HAL = y `, because above there is `EXTRA_CFLAGS += -DCONFIG_RTW_CFGVENDOR_RANDOM_MAC_OUI`,  loaded new module, no compilation error everything finished well, but when scanning the probe requests from the adapter but the MAC is still from the manufacturer, I will keep trying and let you know if something changes.

gteca, 2021-03-14:
I will close the issue because I found I way to do it, I hope it helps others:
`rtw_random32()` generates a 32-bit integer which is fine and I can build my own random mac also, and the kernel uses the same function, to make sure, I recommend using `RTW_INFO` logger to make sure what numbers are being generated.

closed 2021-03-14 by gteca