morrownr/88x2bu #43: Change Sequence number in probe request

gteca, 2021-02-19:
Hi,
I want to change the sequence number in probe request to each 4 (ex: 0, 4, 8 ....), so far I have treid (rtw_mlme_ext.c: _issue_probereq()):
`SetSeqNum(pwlanhdr, pmlmeext->mgnt_seq)`
`pmlmeext->mgnt_seq += 4`
But no effect

morrownr, 2021-02-22:
Hello gteca,

The area that you are looking at is not on my radar at this time. I am quite busy behind the scenes in other areas. This probably does not bode well for a guy doing a college project with a timelines. I understand that.

While you may not be looking for advice, here goes anyway:

Doing serious work on the Realtek drivers is of limited benefit since Realtek does not take pull requests from the public. On the other hand, working on a driver that is in the Linux kernel or in staging allows you to contribute to Linux as the folks at "linux-wireless" would love good contributions. A driver to examine and do work on could be the in-kernel driver for the mt7612u.

You can find links to get an adapter based on the my7612u from:

[github.com/morrownr/USB=WiFi](url)

Good luck.

morrownr, 2021-02-22:
Information about getting involved with Linux Wireless:

[https://wireless.wiki.kernel.org/en/developers/documentation/submittingpatches](https://wireless.wiki.kernel.org/en/developers/documentation/submittingpatches)

gteca, 2021-02-28:
Hi Morrownr,
Thank for the advice, I will take look into the link, and write back if something changes.

closed 2021-02-28 by gteca