morrownr/88x2bu #44: Wi-Fi turns on and off

Fushko, 2021-02-23:
Device: Cudy AC1300 WU1300S (0bda:b812)
Distro: Pop OS 20.04
Kernel: 5.8.0-7642-generic

[Video evidence with `dmesg -w`](https://github.com/morrownr/88x2bu/files/6031912/wifi-issue.webm.gz)

- It's connected via USB 3 directly to the motherboard. I tried `rtw_switch_usb_mode=1`, to no avail.
- This didn't happen when I installed the driver. It started happening after a reboot.
- Unplugging it and plugging it back in fixed it. I can't reproduce now even after a reboot. Thought it was still worth noting.

P.S. Thanks for your valuable work.

morrownr, 2021-02-24:
You are welcome. My thoughts about this issue:

Some usb devices just don't work well in some usb ports. When I see a problem like this, I try a different port. If the ports on the motherboard aren't working well, try the ports on the front. This is not common but it happens. 

The next thing I would try is to check the settings in my wifi router/access point. I have a little section in the README that is about wifi settings. One thing that I have noticed lately is some people turning the power saving off with the module parameter. My options is that unless you test and for sure know it is contributing to a problem, leave that power saving on. There was a time that power saving was an issue but not anymore on modern distros with modern drivers. This site gets a lot of hits and we would be seeing it if there is a problem.

Let's leave this issue up and see if anyone else sees something similar.

Fushko, 2021-02-24:
Thanks for the suggestion.

After trying all the differtent ports, rebooting and cold booting, here's my findings:
- I can't for the life of me reproduce the issue in the OP. Maybe it was some misconfiguration on linux's part that fixed itself? Idk. 
- My mobo (Asrock B450 Pro4) has a bunch of USB 3.1 Gen 1 ports and two USB 3.1 Gen 2. On Gen 1, the thing connects pretty instantly after plugging it in. On Gen2, wifi is detected, but it fails to connect. It just tries over and over without success. The initial detection of the device is also much slower.

I'm not sure what's the culprit here, but I think it's an interesting find nonetheless.

morrownr, 2021-02-24:
Your findings are very consistent with what I think is the real issue. USB 3.1 gen 1 is basically just the USB 3 that we have had around for years. USB 3.1 gen 2 is a different animal. I'm speaking from a programmers perspective, not an end user. An end user just sees either an A style connector or a C style connector but behind the scenes there is a lot of new code to support the features of USB 3.1 gen 2 and that can lead to incompatibilities.

USB WiFi adapters tend to  lag as far as technology goes. You can already buy laptops with 802.11ax support for Linux. Just look for laptops with Intel wifi. Intel's Linux drivers tend to show up in the kernel before the products are available on the market. There are no 802.11ax USB adapters. Period. D-Link  recently announced that they would release a 802.11ax USB WiFi adapter later this year. I wish Intel would support USB WiFi adapters but they don't. We are basically stuck with Mediatek and Realtek as far as USB WiFi chipsets are concerned. Mediatek is doing a really good job working with the kernel wifi coders. In fact, there is support in-kernel for their Wave2 USB chipset (mt7663u) and no adapters have been built yet as far as I can tell. Realtek, on the other hand, does not work well with the kernel coders as far as USB is concerned. I recommend that Linux users looking to buy a USB WiFi adapter look at adapters with Mediatek chipsets. I maintain this site:

[USB WiFi Adapter Information for Linux](https://github.com/morrownr/USB-WiFi)

The difference is that if a person with programming knowledge that wants to add a feature or fix a bug, has the ability to do so with the Mediatek drivers as they are in the kernel. I can point people to how to connect with the Linux wireless folks so as to get started. On the other hand, with Realtek, there is no way to even report bugs, let alone propose fixes or added capability. I have added features and fixed a few things in this driver. Would I like to pass the fixes to Realtek? Sure. Can I do so? No, because Realtek does not offer a way to do it.

Anyway, you stirred me up... but back to topic:

All currently available Realtek based USB 3 WiFi adapters have drivers based only on USB 3.1 gen 1. These adapters depend heavily on USB support in the driver so I get why there are problems with gen 2 ports. You know that now so use your adapter in the gen 1 ports.

Something we need to keep in mind is that these AC1200 and even AC1900 USB WiFi adapters are serious overkill for the majority of users. If you take a look at internet connection speeds et al. you will find that N150 adapters still meet the needs of most users.

I probably should add some info in the README about this.

Regards,

Nick

Fushko, 2021-02-25:
> Realtek, on the other hand, does not work well with the kernel coders as far as USB is concerned

That'a shame. Though in your adapter writeup you say that "the rtl8812bu chipset may see future in-kernel driver support based on the work being done on the rtw88 in-kernel driver.", which kinda gives off a different idea.

> Something we need to keep in mind is that these AC1200 and even AC1900 USB WiFi adapters are serious overkill for the majority of users. If you take a look at internet connection speeds et al. you will find that N150 adapters still meet the needs of most users.

You are right, but this one that I'm using wasn't much more expensive than weaker models, and it works great with this driver (as long as I put it in Gen 1). I'm happy with it so far! Thanks again.

morrownr, 2021-02-26:
Quote: "the rtl8812bu chipset may see future in-kernel driver support based on the work being done on the rtw88 in-kernel driver."

I worded my statement as accurately as I could. The word "may" was included for very good reason. Realtek does not have a good history of supporting its Linux products on USB. The 8812bu chipset is not a bad chipset and this driver is not bad but for a seamless experience with Linux you need the drivers supported in-kernel. I have and support adapters that use all 5 of the Realtek drivers that I support here. I also have and support adapters based on Mediatek (Ralink) chipsets. This allows me to compare and contrast the experiences. 

If you ask me which chipset/driver combos yield the overall best experience at this time, the answer is...

mt7612u and rtl8812au

Both chipset/driver combos are outstanding. Both are excellent in managed (client) mode and outstanding in AP mode to include full 80 MHz channel width support. The driver for the rtl8812au is the best Linux driver I have ever seen Realtek kick out. It is a pleasure to maintain. On the other hand, you and I don't have to maintain the mt7612u driver because it is in the kernel and has people dedicated to maintaining it.

What is not obvious to folks that stop in to grab a driver is that I have to constantly keep an eye on the work on the kernel so as to make adjustments in these Realtek drivers. Note that all 5 Realtek drivers that I maintain here are tested up to and including Linux kernel 5.11. That didn't just happen... it takes a lot of work and Realtek is not doing it. They don't even publicly release their usb drivers. You have to find updates on a few select product sites. It is truly a pain. On the other hand, for the adapters that I support based on the Mediatek (Ralink) chipsets, all I have to do is exactly nothing.

Anyway, I thought you might like to hear some of the story behind the scenes.

Nick

Fushko, 2021-02-27:
Okay, that clears it up. 

This is weird on Realtek's part. They put out these quality drivers, but then they just abandon them and leave them to their chipset's resellers, which might or might not release them? I guess they expect the Linux community to do the maintaining work. Their Linux customer base is probably not important enough.

You know, for tinkerers like me compiling drivers is no big deal. But the vast majority of people want a seamless experience, and this behaviour should not be encouraged, so I'll make sure my next purchase will have kernel support out of the box.

morrownr, 2021-02-28:
Quote - "Their Linux customer base is probably not important enough."

I'm going to have to push back a little here in that there are far more computers running Linux in this world than any other operating system and it is not close. According to a quick check with my buddy Google, there are about 2.5 billion Android (Linux) phones in use. If you look at the Top500 list for Supercomputers, Linux, when last I checked has 100% market share. Then there are Chromebooks and tablets. Billions of IoT things. Realtek is not ignoring Linux. If they did not produce Linux drivers for wifi it would kill their ability to sell into very large markets. The usb drivers we get from Realtek are probably primarily developed and maintained to support Android with our use being a secondary consideration. You can see Android support in these drivers and it sure is easy to compile them for ARM (RasPi). 

Realtek's less than stellar support for Linux is pretty much limited to USB WiFi as they have wonderful support in-kernel for other products. Evidently they have different teams and the wired team takes a different (good) approach and supports their products the right way while the wifi team takes another approach.

You are right on when you say that tinkerers like us have no problem and even enjoy compiling a driver here and there but the vast majority of computer users really don't want to and will not tolerate doing so. To help folks that may want to use a USB WiFi adapter without compiling, I maintain the following site:

[https://github.com/morrownr/USB-WiFi](https://github.com/morrownr/USB-WiFi)

Many folks, even experienced users, do not know that there are a lot of USB WiFi adapters that work with in-kernel drivers in Linux. I really enjoy using the AC1200 MT7612u chipset based adapter I have. The driver is in the kernel. It can match adapters based on chipsets support by this driver in evenday use but... and this is a big but... it is fully compliant with Linux wireless standards. When you use utilities like "iw" or "wavemon" you will get a different look than if using one of these Realtek drivers. These Realtek drivers would need to be heavily rewritten to be able to be accepted into the kernel as they are not standards compliant in some areas.

The best thing that Linux users can do to help ourselves is to vote with our money and buy adapters that are supported in-kernel.

Enjoyed the conversation. Have a good weekend.

Nick

s-mx, 2021-05-01:
Hi! Recently I experienced the same issue and the discussion in this thread gave me a clue what problem I bumped into and changing usb port solved that. So pointing out this aspect in README could really help others who would experience that :)

morrownr, 2021-05-03:
Good idea. I did some editing. Can I get you to check the section called `Recommendations regarding USB` and let me know what you think?

Nick

s-mx, 2021-05-03:
Yeah, now Recommendations are perfect!

closed 2021-12-05 by morrownr