morrownr/88x2bu #45: USB 3.0 causes serious issue, the rpi4 onboard or usb wifi adapter 2.4Ghz band all not working.

danywheeler, 2021-02-24:
When enable usb3.0 the rpi4 onboard 2.4Ghz band ap stop working, and any other wifi adapter has the same issue, stop working on 2.4Ghz band ap, I've tested netgear A6120 and asus ac51 all these can't work on 2.4Ghz, seems have serious interference with 88x2bu, but other adapter like netgear A6120 use usb 3.0 is ok!

danywheeler, 2021-02-24:
Looks like a general issue, USB 3.0 doesn't work well with 2.4Ghz, it seems RPI4 can't use USB3.0 and 2.4Ghz at the same time. Anyone has tested this? or other adapter that works ?

danywheeler, 2021-02-24:
Using a usb wifi adapter with external antenna works on 2.4Ghz and usb 3.0 at the same time, looks like if you want 2.4Ghz and usb 3.0 working the same time, that's the way to go!

morrownr, 2021-02-24:
USB 3 is not mankind's best creation. I use a powered hub for USB 3 devices connected to the RasPi4b. USB 2 WiFi adapters work fine.

How do you like that Netgear A6120?

danywheeler, 2021-02-24:
It's ok very stable, and iperf about 300Mbps, it's using usb 2.0, not sure how to switch it, usb 3.0 maybe perform better, the overall speed is better then 8812bu with usb 3.0 on. and continues on with high throughput, by the way did you use external antenna adapter to make 2.4Ghz work? or just normal adapter plugged in the hub?

morrownr, 2021-02-24:
Regarding the Netgear A6120 - the in-kernel driver for the mt7612u chipset automatically switches to USB 3 if it detects a satisfactory situation. You can check by running...

$ lsusb -t

Mine shows - Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=mt76x2u, 5000M

The 5000M indicates it is in USB 3 mode.

I use a USB 3 powered hub for my raspi4b. It is plugged into the lower USB 3 port (if you have the orientation normal.) I actually use my raspi's with the cases upside down. A lot of things just work better that way. I can go into detail if you are interested.

For what it is worth: I have also found the mt7612u chipset to be very stable. I have an Alfa AWUS036ACM. It seems to support AC with full 80 MHz channel width in managed (client) mode but only 40 MHz channel width in AP and monitor modes. I get about 325 Mbps in AP mode and I keep the following lines off in hostapd -
```
# vht_oper_chwidth=1
# vht_oper_centr_freq_seg0_idx=42
```
The only truly stable, at high speeds in AP mode, Realtek chipsets that I have found are the rtl8812au and rtl8811au.

Nick

morrownr, 2021-03-01:
Disregard what I said about the adapter based on the 7612u chipset not supporting 80 MHz channel width. It does and does a really good job in AP mode.

Something I discovered late last week and it turns out that it has been causing a lot of problems is that the USB subsystem on the RasPi4b, and probably other PIs, does not handle backfeed from some powered USB hubs well at all. I am going to document everything I discovered and post it. This might be an issue for you? Something that likely is an issue for you is the amount of electricity required to run that A6120.

gusarg81, 2021-04-28:
Nick! Hola!

Do you know if the 7612u has MU-MIMO? I can' t find any details about this in the chip info itself.

I've found this one here in Argentina available: Comfast CF-926AC (1200AC) which has MT1612u

Thanks!

morrownr, 2021-04-29:
Hola Guztavo,

You show up all over the place. The adventure continues.

For a good answer about MU-MIMO, you will have to wait for me to refresh my memory... but off the top of my head,,,

I think MU-MIMO was part of the Wave 2 spec as was 160 MHz channel width and a few other things. I'm pretty sure the mt7612u is a Wave 1 chipset. It has MIMO capability but not MU-MIMO.

Okay, I couldn't wait...I had to go digging:

mt7612u > MIMO config: 2x2:2 (867Mbps) - Misc. 802.11ac 2T2R, STBC, TxBF, WoWLAN, USB 3.0

My experience with it shows that the chipset has no problem kicking out 400+ Mb/s in AP mode. It does 80 MHz channel width and supports WPA-3 SAE very well.

I just thought of what started the conversation over in the 8814au repo: I have two adapters based on 7612u chipsets. Would you like for me to test to see if the 7612u driver will drive two adapters at the same time?

Here is the Linux Wireless page that shows the status of the Mediatek drivers:

https://wireless.wiki.kernel.org/en/users/drivers/mediatek

I can't predict when or what the next usb adapter chipset will be. I see the mt7663 has a usb version. Will the mt7915 chipset have a usb version? I don't know yet.

What you might do is look for a PCIe version with the mt7915 chipset. That chipset is 4x4 and 11ax and the driver is already in the kernel. Cables to get the antennas where they in good position and maybe that works for 5g. This is speculation because I have no experience with PCIe wifi adapters.

If you do get that COMFAST adapter, I'd appreciate a good report for the USB-WIFI repo. Do be careful and make sure you can return it because I think I have read that COMFAST makes the 926-AC with two different chipsets - 7612u and 8812bu. I hate it when companies use the same numbers for products that have different chipsets.

Nick

gusarg81, 2021-04-29:
> Hola Guztavo,

Hola! 

> You show up all over the place. The adventure continues.

Hahaha, I am sorry, I am your worst nightmare!
 
> For a good answer about MU-MIMO, you will have to wait for me to refresh my memory... but off the top of my head,,,
>
> I think MU-MIMO was part of the Wave 2 spec as was 160 MHz channel width and a [...]

Oh! That would be nice to test, yes please. No hurry doh!

> Here is the Linux Wireless page that shows the status of the Mediatek drivers:
> 
> https://wireless.wiki.kernel.org/en/users/drivers/mediatek

Cool, lets have a read.

> I can't predict when or what the next usb adapter chipset will be. I see the mt7663 has a usb version. Will the mt7915 chipset have a usb version? I don't know yet.
>
> What you might do is look for a [...]

Well, at least the seller told me that has the 7612u, so If don't, is a goog excuse to return it back otherwise.

So here is what I was planning: now temporary solved the AP using the recovered Archer c7 router. The motherboard of this server has 2 pci x4, buuuuuut one of those can' t be used because the videoboard take that place (I hate this I hate this I hate this...) and the other available slot, used for the secondary ethernet adapter (I can remove the videoboard and buy a Ryzen with the vega video, but I saw there is no video driver from AMD for this one yet, and I don' t trust much the open source one).

Just in case you ask, why I need a video board?: this server acts as an NVR too (for my IPCams and the Smart doobell which supports Rtsp/Onvif as wel), and If I don' t use HW accel, CPU will be used a lot.

So, when the Minipci -> PCIe adapter arrives in June/July (If arrives...) I plan to buy a USB ethernet adapter so I can use that PCIE X4 for the Intel AX wlan.
As for the secondary radio, to buy one of this USB adapters with the MT chipset.

I've asked about MU-MIMO because when those rtl8814 were working good, has a better performance/speed than the Archer C7 router which does not have MU-MIMO.

Again thanks for all the extra data!

gusarg81, 2021-04-30:
Nice!! Thats a good news! Another chip to have in mind once I can come back to this subject (because now I have to wait one month to have my money back...).

Also, and I know you hate TP-Link, but I've found one model with the Intel ax200ngw. The Archer Tx50e (which basically is an MiniPCI to PCIe adapter with the Intel ax200ngw card, which is the same card I have waiting for the adapter to arrive):

https://images-na.ssl-images-amazon.com/images/I/71J-htTi0vL._SL1600_.jpg
https://images-na.ssl-images-amazon.com/images/I/71xZ3T0Jw1L._SL1600_.jpg

¡Que tengas un buen fin de semana también Nick!

closed 2021-12-05 by morrownr