morrownr/88x2bu #48: not working on manjaro

chacles, 2021-03-11:
**uname -a**
Linux 4790K 5.9.16-1-MANJARO #1 SMP PREEMPT Mon Dec 21 22:00:46 UTC 2020 x86_64 GNU/Linux

**execute**
sudo ip link set wlp0s20u13 down && sudo ip link set wlp0s20u13 name wlan2 && sudo ip link set wlp0s20u13 up

**dmesg**
[  115.531593] WARNING: CPU: 2 PID: 270 at /var/lib/dkms/rtl88x2bu/5.8.7.4/build/os_dep/linux/rtw_proc.c:5311 rtw_adapter_proc_deinit+0x77/0x7b [88x2bu]
[  115.531593] Modules linked in: 88x2bu(OE) squashfs loop intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp iwldvm kvm_intel nouveau mac80211 kvm snd_hda_codec_realtek libarc4 snd_hda_codec_generic irqbypass i915 crct10dif_pclmul at24 mei_hdcp ledtrig_audio snd_hda_codec_hdmi snd_soc_rt5640 crc32_pclmul iwlwifi ghash_clmulni_intel ttm snd_soc_rl6231 iTCO_wdt intel_pmc_bxt iTCO_vendor_support snd_hda_intel aesni_intel snd_soc_core snd_intel_dspcfg snd_hda_codec crypto_simd cryptd i2c_algo_bit glue_helper snd_hda_core snd_compress mxm_wmi drm_kms_helper ac97_bus rapl snd_hwdep snd_pcm_dmaengine intel_cstate cec intel_gtt rc_core syscopyarea cfg80211 intel_uncore i2c_i801 snd_pcm sysfillrect sysimgblt fb_sys_fops e1000e snd_timer mei_me rfkill mei lpc_ich i2c_smbus snd pcspkr soundcore evdev wmi mac_hid drm uinput crypto_user fuse agpgart ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 xhci_pci crc32c_intel xhci_hcd ehci_pci ehci_hcd
[  115.531616] CPU: 2 PID: 270 Comm: kworker/2:2 Tainted: G        W  OE     5.9.16-1-MANJARO #1
[  115.531616] Hardware name: Gigabyte Technology Co., Ltd. Z97X-UD3H/Z97X-UD3H-CF, BIOS F9 09/18/2015
[  115.531618] Workqueue: usb_hub_wq hub_event
[  115.531641] RIP: 0010:rtw_adapter_proc_deinit+0x77/0x7b [88x2bu]
[  115.531642] Code: ec 4c 89 f7 e8 58 fe ff ff 4c 89 fe 4c 89 ef e8 a5 5b 16 ec 49 c7 86 88 50 00 00 00 00 00 00 5b 5d 41 5c 41 5d 41 5e 41 5f c3 <0f> 0b eb f1 0f 1f 44 00 00 41 57 41 56 41 55 41 54 55 53 4c 8b b7
[  115.531643] RSP: 0018:ffff98a701183ae8 EFLAGS: 00010246
[  115.531643] RAX: ffffffffc11f9f93 RBX: ffff8ed74cea0000 RCX: 000000000098e802
[  115.531644] RDX: 000000000098e602 RSI: e6093d3c1e6d9bc1 RDI: ffff8ed74cea0000
[  115.531644] RBP: 0000000000000000 R08: 0000000000000000 R09: ffff8ed7400de248
[  115.531645] R10: 0000000000000000 R11: 0000000000000000 R12: ffff98a701183b90
[  115.531645] R13: dead000000000122 R14: ffff98a700641000 R15: ffff8ed73c922140
[  115.531646] FS:  0000000000000000(0000) GS:ffff8ed767a80000(0000) knlGS:0000000000000000
[  115.531646] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  115.531647] CR2: 00007f1afeb47eb0 CR3: 000000001a40e001 CR4: 00000000001706e0
[  115.531647] Call Trace:
[  115.531670]  rtw_ndev_uninit+0xa/0xb [88x2bu]
[  115.531673]  rollback_registered_many+0x368/0x5a0
[  115.531675]  unregister_netdevice_queue+0x8d/0x110
[  115.531676]  unregister_netdev+0x18/0x20
[  115.531696]  rtw_os_ndev_unregister+0x56/0xb1 [88x2bu]
[  115.531716]  rtw_os_ndevs_unregister+0x23/0x2d [88x2bu]
[  115.531736]  rtw_dev_remove+0x24/0xa7 [88x2bu]
[  115.531737]  usb_unbind_interface+0x8a/0x270
[  115.531739]  ? kernfs_find_ns+0x35/0xd0
[  115.531741]  __device_release_driver+0x17a/0x230
[  115.531742]  device_release_driver+0x24/0x30
[  115.531743]  bus_remove_device+0xdb/0x140
[  115.531745]  device_del+0x16f/0x410
[  115.531746]  usb_disable_device+0xc6/0x1f0
[  115.531749]  usb_disconnect.cold+0x7e/0x250
[  115.531750]  hub_event+0xc67/0x1890
[  115.531752]  ? _raw_spin_unlock_irq+0x1d/0x30
[  115.531755]  process_one_work+0x1da/0x3d0
[  115.531756]  worker_thread+0x4d/0x3d0
[  115.531757]  ? rescuer_thread+0x410/0x410
[  115.531758]  kthread+0x142/0x160
[  115.531759]  ? __kthread_bind_mask+0x60/0x60
[  115.531761]  ret_from_fork+0x22/0x30
[  115.531763] ---[ end trace b359bfc27ca8a303 ]---
[  115.894930] usb 2-13: new high-speed USB device number 57 using xhci_hcd
[  116.035351] usb 2-13: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[  116.035352] usb 2-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  116.035353] usb 2-13: Product: USB3.0 802.11ac 1200M Adapter
[  116.035354] usb 2-13: Manufacturer: Realtek
[  116.035354] usb 2-13: SerialNumber: 123456

Thanks for help

morrownr, 2021-03-11:
I'll try to set aside some time to try to duplicate this problem later today.

morrownr, 2021-03-13:
I was able to test on my system that has the Manjaro partition earlier this evening. It is running a 5.9.16 kernel. I tested with a Cudy WU1400 which has a 8812bu chipset. I could not duplicate the problem. The interface was renamed to wlan2 with no errors.

I then tried another adapter with a 8811au chipset and same thing. Worked as expected. Hmmm... what driver options are you using? Is the adapter in a USB 3 port or USB 2?

closed 2021-05-17 by morrownr