morrownr/88x2bu #49: unable to get working on raspberry pi with dnsmasq, vpn

GeneDot3, 2021-03-11:
I have another pi with an older driver working fine as a network bridge, when I install this driver and use the same bridge and iptable configs it does not pass traffic through the vpn.
Raspberry Pi mod 4, Raspian OS, 
iptable config:
-A FORWARD -i eth0 -o wlan1 -m state   --state RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i wlan1 -o eth0 -j ACCEPT
-A FORWARD -i eth0 -o tun0 -m state --state   RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i tun0 -o eth0 -j ACCEPT
-A FORWARD -i eth0 -o wlan0 -m state --state   RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i wlan0 -o eth0 -j ACCEPT
-t nat: 
-A POSTROUTING -o tun0 -j MASQUERADE.
I would like to use your driver as when installed, and vpn is unloaded, it provides much better speed than the old driver.
Not sure what other info you might need, just ask and i'll provide.

Thanks in advance,
Gene.

morrownr, 2021-03-11:
Gene, it is good to meet you.

It may take me a couple of days to have time to investigate this. It will help me if I fully understand what setup you have:

What version of the RasPi are we talking about?
What version of the RasPi OS?
What adapter are you using?
Do you use a powered hub? If so, what is it?
Do you have a checklist you follow to set this up? If so, can I see it? Things work better when I can duplicate things.
Are you using hostapd?

Some info: This 88x2bu driver has made me mad at times. Yes, it is fast... but it has a few fleas. In fact, of the drivers I am maintaining and the info sites I maintain, there are only 2 drivers that I have found that are fully stable at high speeds (80 mhz channel width) and of those 2, only one is capable across board and by that, I mean things like WPA3-SAE, etc. Here goes:

Alfa AWUS036ACM - mt7612u - driver is in-kernel - wonderful AP mode support - does WPA3 with hostapd. Overall, best AC1200 class usb adapter for Linux. Period.

Alfa - AWUS036ACH - rtl8812au - driver repo - https://github.com/morrownr/8812au - really solid driver - best out-of-kernel driver from Realtek - lacks some features that all Realtek drivers lack like WPA3.

For more info on usb adapters and Linux - https://github.com/morrownr/USB-WiFi

GeneDot3, 2021-03-11:
Nice to meet you as well.Let's see if this translates well from my spreadsheet...
 
| What version of the RasPi are we talking about?      Raspberry Pi Mod4b with 4GB memory |
| What version of the RasPi OS?                                 Raspian OS Buster |
| What adapter are you using? It seems to be a no-name -  https://www.amazon.com/Wireless-USB-WiFi-Adapter-PC/dp/B07V4R3QHW/ref=sr_1_3?dchild=1&keywords=RTL881cu+USB+Wifi+adapter&qid=1607538337&s=electronics&sr=1-3 |
| Do you use a powered hub?    No. |
| Are you using hostapd?          No |
| Do you have a checklist you follow to set this up? Yes. If so, can I see it? Sure, it's a spreadsheet. |
| Things work better when I can duplicate things. - I agree
 |
| Bridge Configuration (iptables set for use with eth0, wlan0, wlan1 & tun0) |
| sudo apt update && sudo apt upgrade -y |
| sudo apt install dnsmasq -y |
| sudo nano /etc/dhcpcd.conf |
|   | interface eth0 |
|   | static ip_address=192.168.220.1/24 |
|   | static routers=192.168.220.0 |
| sudo service dhcpcd restart |
| sudo mv /etc/dnsmasq.conf /etc/dnsmasq.conf.orig |
| sudo nano /etc/dnsmasq.conf |
|   | interface=eth0                         # Use interface eth0 |
|   | listen-address=192.168.220.1   # Specify the address to listen on   |
|   | bind-interfaces                         # Bind to the interface |
|   | server=8.8.8.8                         # Use Google DNS   |
|   | domain-needed                        # Don't forward short names   |
|   | bogus-priv                               # Drop the non-routed address spaces.   |
|   | dhcp-range=192.168.220.50,192.168.220.150,720h    # IP range and lease time   |
| sudo nano /etc/sysctl.conf |
| find - | #net.ipv4.ip_forward=1 |
| remove #  - | net.ipv4.ip_forward=1 |
| sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward" |
| sudo iptables -A FORWARD -i eth0 -o wlan1 -m state --state RELATED,ESTABLISHED -j ACCEPT |
| sudo iptables -A FORWARD -i wlan1 -o eth0 -j ACCEPT |
| sudo iptables -A FORWARD -i eth0 -o tun0 -m state --state RELATED,ESTABLISHED -j ACCEPT |
| sudo iptables -A FORWARD -i tun0 -o eth0 -j ACCEPT |
| sudo iptables -A FORWARD -i eth0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT |
| sudo iptables -A FORWARD -i wlan0 -o eth0 -j ACCEPT |
| sudo iptables -t nat -A POSTROUTING -o tun0 -j MASQUERADE |
| sudo sh -c "iptables-save > /etc/iptables.ipv4.nat" |
| sudo nano /etc/rc.local |
| find - | exit 0 |
| add above it - | iptables-restore < /etc/iptables.ipv4.nat |
| sudo service dnsmasq start |
| sudo reboot
 |
| 8822bu Driver for USB WiFi antenna |
| https://github.com/morrownr/88x2bu |
| expressvpn disconnect |
| sudo apt update && sudo apt upgrade -y |
| sudo apt-get install raspberrypi-kernel-headers bc build-essential dkms git -y |
| mkdir src |
| cd src |
| sudo git clone https://github.com/morrownr/88x2bu.git |
| cd |
| cd src/88x2bu |
| sudo ./raspi32.sh |
| sudo ./install-driver.sh |
| sudo iwconfig wlan0 txpower off && sudo reboot |
| sudo reboot |

morrownr, 2021-03-12:
So, let me make sure everything is clear in my mind: You are using a RasPi 4b with a 8812bu based usb adapter to run a bridged access point with ethernet cable providing the internet. Correct me if I am wrong. This and the other issues I am working makes for a lot of information to process.

Questions:

You say you are running  Raspian OS Buster. I'd like to get clarification without me having to take time to research this. My RasPi 4b is running the Raspberry Pi OS 32 bit fully updated to the present. The kernel is v5.10. The name of the official RasPi OS was changed from Raspian to Raspberry Pi OS at some point last year. Are you telling me that you are running an old version from when it was still called Raspian? What version is the kernel?

I have no experience with VPN's. Can you point me to information that can help me spin up to speed on the topic?

Regards/

GeneDot3, 2021-03-12:
Thanks for considering my problem.

"So, let me make sure everything is clear in my mind:"
I am sorry if I caused confusion.  
"You are using a RasPi 4b with a 8812bu based usb adapter to run a bridged access point with ethernet cable providing the internet. Correct me if I am wrong."
I am using the very latest Raspberry Pi OS(Buster), I always do a "sudo apt update && sudo apt upgrade -y" to make sure it is current.
The Pi is connected via WiFi to an AP for both inbound and outbound traffic, so no ethernet on that end.   
A wireless router is connected via ethernet cable to provide wireless and wired internet to multiple devices.
So, a few things need to happen: WiFi traffic has to get routed to and from the ethernet port,and, all the while, I use the Pi for other activities via wireless usb keyboard input (kodi - hdmi to TV, pandora - phono jack out to stereo).(these mod4's are so amazing)
My idea stemmed from: https://pimylifeup.com/raspberry-pi-wifi-bridge/Some reference material I used: https://thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html and https://ipset.netfilter.org/iptables.man.html
I know these things get crazy to comprehend sometimes and it is quite the challenge. I, myself, do not possess the understanding to great depth, just enough 
to make me dangerous, lol. Like I said, I have it running with another driver, also 88x2bu, but I like the fact that you provide an easy install as well as an updated driver.
Kudos to you for your work!
I tried to answer your questions, if more info is needed I will provide what I can.

GeneDot3, 2021-03-13:
I left out one important piece of info, I run expressvpn which is part of my Pi image that is also working on my current config - this is actually the part that won't work. AND I ACTUALLY FORGOT TO INCLUDE IT EARLIER!!!!  I am losing it!
Very sorry!

GeneDot3, 2021-03-13:
SUCCESS!!!!!
I had my iptables fouled up.
New iptables structure:

 -t nat -A POSTROUTING -o tun0 -j   MASQUERADE
 -A FORWARD -i tun0 -o eth0 -m state --state   RELATED,ESTABLISHED -j ACCEPT
 -A FORWARD -i eth0 -o tun0 -j ACCEPT
 -A FORWARD -i tun0 -o wlan1 -m state --state   RELATED,ESTABLISHED -j ACCEPT
 -A FORWARD -i wlan1 -o tun0 -j ACCEPT

You may close this issue, and I thank you for the attention you gave!
Best Regards,
Gene

closed 2021-03-13 by GeneDot3