morrownr/88x2bu #52: please add LICENSE?

colemickens, 2021-03-16:
Can you list an explicit license on this repo? I'm working on packaging this for nixpkgs and would like to ensure that I can mark the right license for the module.

Thanks!

morrownr, 2021-03-17:
Done. See document LICENSE.

morrownr, 2021-03-17:
@colemickens 

A few things for you to consider packaging:

New or prospective Linux users need to understand that this Realtek driver and other Realtek out-of-kernels drivers are not Linux Wireless standards compliant. This can and does affect operation of the drivers and interaction with system utilities. 

They also need to understand that Realtek provides no support for these drivers. Have you tried to submit a bug report to Realtek? I have. No contact whatsoever.

New or prospective Linux users need to understand that there are alternatives to Realtek based USB WiFi adapters. Adapters that are Linux Wireless standards compliant, the drivers are maintained in-kernel making the adapters plug and play and where there is a clear path to report bugs with an expectation that the issue will be addressed. See the following site for more information:

https://github.com/morrownr/USB-WiFi

My opinion is that almost all Linux USB WiFi adapter users would be much more satisfied by using adapters that are supported with in-kernel drivers. If you are packaging something, how about packaging some of this information with the driver. This information should serve Linux users well.

Regards,

Nick

colemickens, 2021-03-17:
Thank you for the fast turn around and the detailed response Nick. In fact, it's motivating enough to cause me to return these devices, and then refocus my efforts on better things.

Thank you for your time and efforts.

colemickens, 2021-03-17:
(Please let me know if there's a donate/paypal link or crypto addr to say thank you for maintaining the USB-Wifi repo. That's a lot of good info and I really think this sort of advocacy is important. I'm so much more excited to get a better product now.)

morrownr, 2021-03-18:
I appreciate the kind offer but I do not have a need for donations. This work is just me paying forward based on the kindness that Linux users have shown me over the years. What I could use is a helping hand or two that would help me maintain and improve the USB-WiFi repo. USB WiFi adapters can be problematic on Linux if users do not have good access to the information that they need. The question that USB-WiFi is trying to answer is: What USB WiFi adapter can I buy that will just work? 

Cheers

closed 2021-04-04 by morrownr