morrownr/88x2bu #54: Realtek-rtl8812bu not active on Fedora 33 Server; no GUI (Kernel 5.11.11-200)

ThiloteE, 2021-04-05:
Dear morrownr,

First of all, thank you very much! Your repository has made life a little bit easier, at least for me, since i've stumbled upon it. I've been using your driver for the Realtek-rtl8812bu chipset on a machine with both Linux Mint (Ubuntu based) and LMDE 4 (Debian based) very successfully 👍 
(here the wifi adapter: https://www.amazon.de/EasyULT-1200Mbps-802-11ac-Dualband-Empf%C3%A4nger/dp/B07YDHTHH3/ref=cm_cr_arp_d_product_top?ie=UTF8).
Both USB 2.0 and USB 3.1 (gen1) were working fine. 👍 

**Here is the the problem:**

When i use Fedora 33 on this same machine without GUI it doesn't work for me (i am trying to turn this machine into a server, but i am rather new to Linux). I am at a loss on how to activate the wifi adapter.

What i did to install the driver:

sudo dnf update
sudo dnf install -y dkms git
mkdir src
cd ~/src
git clone https://github.com/morrownr/88x2bu.git
cd ~/src/88x2bu
sudo ./install-driver.sh
sudo reboot

after the reboot i tried to check for wifi signals with:
(following this guide: https://ostechnix.com/connect-and-disconnect-wifi-from-commandline-in-linux/)

$ nmcli device wifi list 

It doesn't give me any output though. No wifi networks are listed that i could connect to. See here: 

[image]

Also the wifi-stick doesn't blink blue, so i think it still is turned off.

**How to turn it on? Is the driver not compatible with Fedora or am i missing something?**

morrownr, 2021-04-05:
Hello ThiloteE,

To see if the driver is loading, run

$ lsmod

Do you see the following line in the output?

88x2bu               3059712  0

If 88x2bu is there, the driver is loading.

Something that I need to do is try to duplicate what you are seeing. I'll make a new partition on a test system to run Fedora. Can you provide me with a URL to the exact ISO of Fedora you are using?

ThiloteE, 2021-04-05:
Thank you for your fast response!!

https://download.fedoraproject.org/pub/fedora/linux/releases/33/Server/x86_64/iso/Fedora-Server-dvd-x86_64-33-1.2.iso

This one should be it, but i used an older version (still called fedora 33) from January and then used the "update" function in cockpit (https://cockpit-project.org/running), which installed a security update. Cockpit is preinstalled on Fedora Server.

During the installation of Fedora, i chose to go with a minimalistic approach and only installed the "server edition" + one or two  additional softwarepackages that had something to do with user accounts management and DNS settings or something similar (unfortunately, i can't remember the names; haven't used them conciously). Later i added firefox and syncthing, as well as git.
All in all not really a complex system environment.

1. i can't see 88x2bu 3059712 0  when using $ lsmod
2. i tried to run the installation again. It definitely is installed:
[image]

i don't see it under "$ dnf history userinstalled" either.

morrownr, 2021-04-05:
Quote: "i tried to run the installation again. It definitely is installed:"

Well, the script is really just telling us that the files that were copied during a previous installation are still in place so a "remove" is needed to clean things up before trying again or upgrading an older version. It is not telling us if the driver compiled correctly.

Quote: "i can't see 88x2bu 3059712 0 when using $ lsmod"

Something went wrong. The driver is not installed. There is likely a problem with the compilation. I have absolutely zero experience with Fedora or Red Hat but am willing to give it a shot in order to figure this out.

Do this:

Go back into the driver directory:

$ cd ~/src/88x2bu

Run the following to clean things up:

$ sudo ./remove-driver.sh

$ cd ..

$ rm -r 88x2bu

Now let's start clean:

$ git clone https://github.com/morrownr/88x2bu.git

$ cd 88x2bu

$ make

Be prepared to copy the last 15-20 lines of the result and post it here.

ThiloteE, 2021-04-06:
[image]

could it be that fedora is not a usermerged system? would the driver have to be placed in /home/usr/lokal/src/ ?

these are the src folders i located under home directory:
[image]

Let me quote Clem (https://blog.linuxmint.com/?p=4037):

_This is quite technical. When we moved from Linux Mint 19.3 to Linux Mint 20, we moved from an Ubuntu 18.04 to an Ubuntu 20.04 package base. One important difference between these two package bases is the fact that Ubuntu 20.04 is a merged system (i.e. a system where some paths such as /bin and /usr/bin are merged via symlinks and are effectively the same, this is described here: https://wiki.debian.org/UsrMerge)._

_On a merged system if a package places a binary in /bin and another package calls it with the wrong path (/usr/bin for instance), it doesn’t matter, it works anyway. This is an improvement for users but it also hides potential issues from a maintainer’s point of view. If you package something with the wrong path, your package continues to work on merged systems but not on non-merged ones._

If nothing has changed since then, the lib files should be under /usr since 2012, though (https://fedoraproject.org/wiki/Features/UsrMove)

morrownr, 2021-04-11:
Okay. Finally getting time to work this issue.

I can see now that I am going to have to install Fedora to get to the bottom of this. Can you please provide a link to the version of Fedora you are using?

Thanks.

ThiloteE, 2021-04-11:
https://download.fedoraproject.org/pub/fedora/linux/releases/33/Server/x86_64/iso/Fedora-Server-dvd-x86_64-33-1.2.iso

here we go. I posted it once, but yes, there is a lot of text by now, so you might not have seen it. :D

morrownr, 2021-04-12:
Thanks for posting the link again. As much as I try to keep up with everything that goes on here, I can't. I have several repos that take a lot of work... https://github/morrownr

If you have not visited the following site please do...

https://github.com/morrownr/USB-WiFi

It provides a LOT of information about USB WiFi adapters on Linux.

I have a busy week coming up and since this is not my day job, I will install Fedora and see what is going as I am able.

Ashex, 2021-04-12:
I'm experiencing the same issue after using this driver without issue for some months until I decided to go in and do some driver cleanup, some quick details:

OS: Fedora 32 Server
Kernel: 5.11.11-100.fc32.x86_64

dmesg shows detection of the USB Device but the module does not show up in lsmod. Things I have done:

* Remove all drivers and versions from dkms
* Uninstall morrownr driver with repo script
* Installed latest copy of morrownr driver with repo script 

dkms shows the driver is installed to the kernel. When I connect the usb wifi device I do see the following message in dmesg:

```
88x2bu: version magic '5.11.7-100.fc32.x86_64 SMP mod_unload ` should be `5.11.11-100.fc32.x86_64 SMP mod_unload `
```

Ashex, 2021-04-19:
@ThiloteE What happens when you load the module directly with `insmod`?

For example:
```
insmod /lib/modules/5.11.11-200.fc32.x86_64/extra/88x2bu.ko.xz
```

For my case I had to load it manually to get wifi working then ran a full upgrade and now it loads reliably, it seems that dkms had a dirty build environment for me which caused the issue.

morrownr, 2021-04-25:
You can check the dkms status with...

$ dkms status

Running the remove script should clean any dkms problems up...

$ sudo ./remove-driver.sh

ThiloteE, 2021-05-25:
Sorry for the late reply. I decided to use a cable connection for now. It took me some effort to make space for the setup in the room with my router, but for now that might be the best.

**Regarding the problem: It still does not work on my Fedora 33 Server.**
Unfortunately i am not on the 5.11.11-200 kernel anymore, but rather on the 5.12.6-200 fc33x86. Maybe i should not have upgraded 😅

i checked for dkms status though:
[image]
noticed, that it does not show anything and the driver seemed to not have been entered in the tree.

I checked if dkms is installed properly, which it was, but i found another command you posted specifically for Fedora server, which i used:

    # sudo dnf -y install git dkms kernel-devel kernel-debug-devel

It seemed to help a lot, because after a restart i got this info:

[image]

insmod didn't seem to work for me. The driver in itself got compiled without any errors by dkms. On first glance everything seems to be in order, but ultimately my device is still acting dead.

After that. i tried to get some more info if and how my device gets recognized by the system:

[image]

It doesn't show my device. I am out of ideas.

morrownr, 2021-06-21:
Sorry for the slow reply. I missed this. The only I can recommend is for you to delete the driver directory, get the driver again per instructions and then run $ sudo ./remove-driver.sh as the new version of ./remove-driver.sh really cleans things up. At that point, run the full installation instructions and see what happens.

safforddr, 2021-07-23:
The insmod error "Key was rejected by service" means that secure boot is turned on, and the new driver is not signed.
I have 8822bu and had the same error, which went away when I signed the driver with a local registered key.
You can see if this is the problem by temporarily turning off secure boot in BIOS.

I'm now getting "Invalid parameters" when I try to insmod a signed 88x2bu.ko.xz on fedora with the current 5.13.4 kernel, but I'm still trying to figure that one out.

closed 2021-08-16 by morrownr