morrownr/88x2bu #55: txpower strange behavior

nikallian, 2021-04-06:
Hi,
 it might be strange but I have noticed with the latest version of the driver a strange behavior. I check the txpower power from time to time and I notice that it changes from 20 up to 23 dBm. I swear I don't touch the txpower neither me nor a script. I have tested other similar drivers but its only with this one that I noticed it. My dongle is a T4U v.3

nikallian, 2021-04-06:
Just to give more info. I use T4U with hostapd setup as 2.4G AP. Everything runs on a RPI3B+ It's not the best WiFi dongle for using it as AP but it's acceptable.

morrownr, 2021-04-06:
Hi,

Regarding the power, can I get you to write down the steps you use to check the power? If I can duplicate the issue, I stand a better chance of seeing if this is a bug or normal ops.

Regarding the hostapd setup, I I run hostapd on a Pi also. I do a lof testing test. I might be able to help tweak your hostapd.conf if you post it here.

nikallian, 2021-04-07:
Hi Nick (I'm Nikolas too :)

well, I just issue the command "iw dev". I just did it and I saw that while yesterday the txpower was 21 dBm (is this the default ?) today is....23 I don't know how and why but I don't think that hostapd can change the txpower.

This is the hostapd.conf file. 

`# /etc/hostapd/hostapd.conf
# https://w1.fi/hostapd/
interface=wlan0
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
ssid=HOME
ieee80211d=0
ieee80211h=0
hw_mode=g
channel=11
beacon_int=100
dtim_period=1
max_num_sta=32
macaddr_acl=0
ignore_broadcast_ssid=0
rts_threshold=2347
fragm_threshold=2346
send_probe_response=1
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_pairwise=CCMP
wpa_passphrase=xxxxxxxxx
wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256
wpa_group_rekey=1800
rsn_pairwise=CCMP
ieee80211n=1
wmm_enabled=1
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
force_ht40=1'

I'm not sure but I think that I copied from you (I have compiled and tested 6-8 different versions of 88x2bu driver from various gits repos and I can't remember if this is from you or someone else #34 

Thanks in advance for your help and time!

Nikolas

nikallian, 2021-04-07:
I forgot to post the options of the driver module

`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=2`

morrownr, 2021-04-08:
The following is not necessary unless you need to offer security of less than WPA-2 AES

wpa_pairwise=CCMP

Other than that, I see no issues. My main "wifi router" is a RasPi4b dual band. For 5g I use an Alfa AWUS036ACM (mt7612u chipset) and for 2g I use the onboard wifi. This is a combination that works very well. Here is a link to the setup

https://github.com/morrownr/7612u/blob/main/Bridged_Wireless_Access_Point-2.md

You can see at the end of the file where I posted the 5g iperf3 results. It averages around 430 Mb/s in my setup.

You probably know this but the 88x2bu driver here in this repo is not particularly good in master (AP) mode. It does better in 2g than 5g. I have tested and worked on it until i am blue in the face.

nikallian, 2021-04-09:
Hi Nick
Thanks for your reply.

I corrected the hostapd conf as per your suggestion. Yes I know that 88x2bu isn't the best choice for AP but it's cheap. Your driver version and a couple more only support AP. ALFA was my first priority (I already own AWUS036NH) but in my country the AWUS036ACM isn't available. I have located some European eshops that ship the aforementioned ALFA adapter and I may buy one. 

Any clues on what is the max txpower for 88x2bu? I'm not a programmer and it's quite difficult for me to dig in the code and find it my self. I assume that txpower is part of the FW part, right? I may finally choose your solution for a complete 2.4/5 AP although RasPI WiFi antennas are a bit crappy.

morrownr, 2021-04-09:
Hi Nikolas,

Quote: "Any clues on what is the max txpower for 88x2bu?"

This is a harder question to answer than we might think. The answer depends on what band you are using and what country you are in. The driver and firmware has to operate in accordance with the laws of the country where the adapter is located. As long as your computer has the correct country code set, the easiest way to see what the max txpower should be is to issue this command:

$ iw list

You have to dig through the output to look at the sections called "Frequencies". There will be one of these sections under Band 1 and another under Band 2. Note the differences between the bands so if you connect with a 2.4g ap, you may have a txpower of X and if you then change to 5g, you may see txpower of Y and it can change depending on exactly which channel you select.

Regarding the "crappy RasPi wifi antennas": In general I agree with your assessment but the RasPis were not designed to be flame throwing wifi routers. They are low cost general purpose computing devices. The performance I get from the internal 2.4g of the RasPi is not bad at all. There is not much to be said for the 5g performance as an AP but that is why I use the Alfa adapter for 5g. I support systems with a wide variety of wifi adapters and cards so I get to do a lot of testing. Performance depends on the adapter/card, the driver, the configuration and the location. I have tested numerous usb adapters with RasPis and the runaway winner is the Alfa ACM. It is almost like Alfa designed that adapter to work with RasPis.

Adapters based on the rtl8812bu chipset have exceptional performance when operating in managed (client) mode using this driver. The real world performance will push adapters based on the rtl8814au which is an AC1900 chipset. So what I do with adapters with the rtl8812bu chipsets is I use them for duty as managed (client) adapters. They are just not that good for master (ap) or monitor modes. This is entirely due to the abilities of the current driver. There is the possibility that there may be an in-kernel driver for the 8812bu chipset at some point so the situation could change.

Regards,

Nick

nikallian, 2021-04-09:
Hi Nick
thanks for your prompt reply.

So the driver follows strictly the regulatory domain and there is no way to derive even a bit from it (I suppose). I still though wondering why some of these drivers (all based on RTL sources) sometimes are configured with fixed 12dBm power and some others with 19 or 21 and in these cases you also be able to chance the txpower (although it seems that this is just a "cosmetic" feature). Do you have any guessing?

Anyway, I will try to get on my hands the ACM version of ALFA 802.11ac or maybe some other based on 8814au. To be honest I wanted to buy something like "swiss army knife" meaning to have AP/monitor/high TXpower/injection support all in one but it seems I have to settle down with multiple adapters solution. 

Thank you once more for your help!

nikallian, 2021-04-09:
Oh! by the way, what's your opinion on AX200/210 adapters from Intel? I have a laptop and I am thinking of replacing the installed with one of these two.

morrownr, 2021-04-09:
Quote: "or maybe some other based on 8814au"

My recommendation for Linux users is to avoid adapters with the 8814au chipset like it is the plague. Realtek's driver for that chipset is old and not good. Please don't buy an adapter with that chipset. You will probably not be happy.

Quote: "Oh! by the way, what's your opinion on AX200/210 cards from Intel?"

Thumbs up. Intel does a GREAT job of providing and supporting "in-kernel" drivers for their wifi chipsets. If only Intel would support a wifi chipset to the USB wifi market.

Regarding your question about txpower: Many adapters lock power settings in the onboard firmware so that folks can't break the law. This is a topic where there are a lot of misconceptions. We really need an EE to explain some radio theory to fully understand the topic. For me, I am going to support these drivers to operate in accordance with applicable laws. I have no desire for the FCC to pay me a visit and there are plenty of other things to have fun with. Besides that, many an adapter, in the old days, was burned out from turning up the txpower. That txpower is not easy to change these days has probably saved the lives of countless wifi adapters.

Nick

nikallian, 2021-04-09:
Thanks for your reply and info Nick.

Just red your great post about USB-WiFi adapters. Great info indeed! I will select more carefully my next usb adapter after that. 

Thanks once more!

morrownr, 2021-04-13:
You are welcome.

closed 2021-12-05 by morrownr