morrownr/88x2bu #56: DietPI 64bit linker problem

ClaudioCimarelli, 2021-04-13:
Hello everyone, I am posting this as a continuation of #16 as I am experiencing the same issue reported here https://github.com/morrownr/88x2bu/issues/16#issuecomment-734972237
``` 
ERROR: "__const_udelay" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                           
ERROR: "__arch_copy_from_user" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                    
ERROR: "_raw_spin_unlock" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                         
ERROR: "__rcu_read_lock" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                          
ERROR: "__rcu_read_unlock" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                        
ERROR: "cpu_hwcap_keys" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                           
ERROR: "csum_ipv6_magic" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                          
ERROR: "__warn_printk" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                            
ERROR: "_mcount" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                                  
ERROR: "arm64_const_caps_ready" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                   
ERROR: "__udelay" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                                 
ERROR: "preempt_schedule_notrace" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined!                                                 
ERROR: "__arch_copy_to_user" [/var/lib/dkms/rtl88x2bu/5.8.7.4/build/88x2bu.ko] undefined! 
```

My configuration is as follows:
model Raspi 3B
Os DietPios v6.31.2 aarch64
kernel 5.4.51-v8+

Hope someone can help.

morrownr, 2021-04-13:
Did you run...

$ sudo ./raspi64.sh

...per the installation instructions?

ClaudioCimarelli, 2021-04-13:
Yes, I did follow the instructions.

morrownr, 2021-04-13:
I am backed up with some other issues right now so it may take a while to look at this. You did look at #16 . There have been issues with 64 bit ARM support and it has thus far been problems in the distros, not with this driver. I will make the change to 64 bit RasPiOS when the 64 bit version comes out of beta. If you can get in touch with PieGuy, he is probably your best bet in the short term... or you could go with the good ole RasPiOS 32 bit and wait for things on the 64 bit world to mature.

morrownr, 2021-04-13:
#34  seems  to show the same issue.

ClaudioCimarelli, 2021-04-13:
Thanks a lot for your effort. I did not know the 64-bit os was not so mature. I will try to flash the 32 bit to try.

morrownr, 2021-04-13:
The 64 bit ARM world, as far as RasPi hardware is concerned, is still problematic. Why is it taking so long for the RasPi Foundation to get the 64 bit version out of the door? I don't know but the 5 driver and 2 info sites I maintain here keep me busy enough that I will worry about the 64 bit versions of RasPiOS and DietPi when they are no longer test versions. There were multiple conversations in the 5 driver repos here about this and I think it was something messed up in a header file but it has been a while. The conclusion was to press on with OSs that are not test versions.  The basic 32 bit RasPiOS is pretty solid.

ClaudioCimarelli, 2021-04-14:
Still, I am curious to hear from @PieGuy314 ( if he ever can read this ) as he managed to compile this successfully on DietPiOS 64-bit.

PieGuy314, 2021-04-14:
> Still, I am curious to hear from @PieGuy314 ( if he ever can read this ) as he managed to compile this successfully on DietPiOS 64-bit.

It used to compile. But then there was a change to the kernel code and the Pi kernel devs stopped applying the patches for the time being (resource best spent doing more important stuff). Worth highlighting that this isn't a driver issue, it's entirely due to the 64-bit Pi kernel. 

As @morrownr has suggested, stick with the 32-bit for now.

rjbrown99, 2021-04-23:
Same issue here on Raspberry Pi OS 64bit, with the latest prerelease kernel 5.10.31-v8+.

  MODPOST /root/src/88x2bu/Module.symvers
ERROR: modpost: "__const_udelay" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_from_user" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "_raw_spin_unlock" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_lock" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_unlock" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "cpu_hwcap_keys" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "csum_ipv6_magic" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__warn_printk" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "arm64_const_caps_ready" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__udelay" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "preempt_schedule_notrace" [/root/src/88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_to_user" [/root/src/88x2bu/88x2bu.ko] undefined!

Have not worked through it yet.

morrownr, 2021-04-23:
Sir, you are in good company. We have already spent time on this. It was a kernel header bug that was introduced at some point and was fixed in everything but the 64 bit RasPiOS and the distros derived from it. There really isn't anything we can do here so my position is that I will worry about the 64 bit RasPiOS when it is released. It is going to be released, right? The RasPiOS devs working on the 64 bit version are making Debian stable releases look like a race car. Well, maybe not but you get my point.

Your best bet is going back to the 32 bit version for now.

rjbrown99, 2021-04-23:
Appreciate the reply. I'm going to stick with 64 bit either way, so I'm about to dive further into the rabbit hole to determine what broke in the kernel or where. If you have any leads that would certainly be helpful, otherwise I'll see what I can turn up.

PieGuy314, 2021-04-23:
> Appreciate the reply. I'm going to stick with 64 bit either way, so I'm about to dive further into the rabbit hole this weekend to determine what broke in the kernel or where. If you have any leads that would certainly be helpful, otherwise I'll see what I can turn up.

Best advice I can give is to learn how to use GDB. On kernel modules. And the kernel. Good luck.

closed 2021-12-05 by morrownr