morrownr/88x2bu #59: Build Err: ERROR (dkms apport): binary package for rtl88x2bu: 5.8.7.4 not found

w407022008, 2021-04-24:
Hi, there is an error when i run `sudo ./install_driver.sh`.

> 'make' -j12 KVER=5.4.0-72-generic KSRC=/lib/modules/5.4.0-72-generic/build...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.8.7.4 not found
Error! Bad return status for module build on kernel: 5.4.0-72-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.4/build/make.log for more information.
An error occurred. dkms build error = 10

and the make.log shows below:
[make.log](https://drive.google.com/file/d/1fRmjDoxFAJtTTebpaeZBdfKLToL0Bfn8/view?usp=sharing)
Anyone help me pleeeeee..

morrownr, 2021-04-24:
We need more information. What Linux distro and version are you using? What hardware are you using?

w407022008, 2021-04-25:
Hi, @morrownr .
Sorry for that, which should be necessary. 
Well, the **Linux distro and version**: **Ubuntu 18.04.1  5.4.0-72-generic  x86_64 GNU/Linux**
**the hardware**: the PC is **Dell Optiplex 7060**, well i don't know what kind of motherboard it is. the wifi device is [this](https://www.amazon.fr/gp/product/B08CZCJJZ9/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1) which has the chip of **rtl8812bu**.

In fact, I have been using the driver you provided normally for the past six months until the day before yesterday when I was prompted for a software update. Maybe, I'm just guessing, it conflicted with dkms after the update or something. When I turned on my computer yesterday and found that I couldn't search for wifi, I initially planned to remove and reinstall it, but I found that I couldn't remove it completely, and when I installed it again, he reminded me that the driver was already installed, so I manually removed the rtl package in the var/lib/dkms path and the corresponding files (/usr/src/${DRV_NAME}-${DRV_ VERSION}, /etc/modprobe.d), and when I bash install_driver.sh again, I found that dkms could not be successfully built.

Thanks a lot.

morrownr, 2021-04-25:
Okay. Well. I recently did some work on the installation and remove scripts to try to prevent a situation where it is sort of half installed and remove would not clean it up. This may be where you are or were.

If you have an alternate means of internet access for the computer in question, how about trying the following:

Destroy your current driver along with its folder: ~/src/88x2bu

Go through the installation process until it is time to run install-driver.sh but instead run remove-driver.sh first and the install-driver.sh.

Let me know what happens.

w407022008, 2021-04-26:
Hi, @morrownr 
I tried to do what you said. I destroyed the current driver and I made sure that there are no files under these three paths(/usr/src/; /etc/modprobe.d/; /var/lib/dkms/) that we want to add. I removed the folder (~/src/88x2bu), and then re-cloned it according to the [git guide](https://github.com/morrownr/88x2bu#readme). Before installing it, I ran the file (remove-driver.sh) as you said, and then the terminal showed:

> Start removal.
Error! There are no instances of module: rtl88x2bu
5.8.7.4 located in the DKMS tree.
Deleting 88x2bu.conf from: /etc/modprobe.d
Deleting source files from: /usr/src/rtl88x2bu-5.8.7.4
The driver was removed successfully.
Are you ready to reboot now? [y/n]

morrownr, 2021-04-26:
I guess we need this to be hard because if it was easy, it wouldn't be fun.

Let's try to do some stuff manually. Open a terminal and do the following:

$ cd ~/src/88x2bu

$ clear

$ make clean

$ make

Now copy everything that appears onscreen after $ make and paste it here.

w407022008, 2021-04-26:
[log.txt](https://github.com/morrownr/88x2bu/files/6380508/log.txt)

Well, It is too long, but all the errors are related to an expected identifier or '('.

morrownr, 2021-04-26:
Something is causing the compilation to fail. I just tried to duplicate your results using the same kernel...

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-72-generic/build M=/home/morrow/src/88x2bu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-72-generic'
  CC [M]  /home/morrow/src/88x2bu/core/rtw_cmd.o
  CC [M]  /home/morrow/src/88x2bu/core/rtw_security.o
  CC [M]  /home/morrow/src/88x2bu/core/rtw_debug.o
  CC [M]  /home/morrow/src/88x2bu/core/rtw_io.o
  CC [M]  /home/morrow/src/88x2bu/core/rtw_ioctl_query.o
  CC [M]  /home/morrow/src/88x2bu/core/rtw_ioctl_set.o
  CC [M]  /home/morrow/src/88x2bu/core/rtw_ieee80211.o
.
.
.
  CC [M]  /home/morrow/src/88x2bu/hal/phydm/txbf/haltxbf8822b.o
  CC [M]  /home/morrow/src/88x2bu/hal/btc/halbtc8822bwifionly.o
  CC [M]  /home/morrow/src/88x2bu/hal/btc/halbtc8822b1ant.o
  CC [M]  /home/morrow/src/88x2bu/hal/btc/halbtc8822b2ant.o
  CC [M]  /home/morrow/src/88x2bu/platform/platform_ops.o
  CC [M]  /home/morrow/src/88x2bu/core/rtw_mp.o
  LD [M]  /home/morrow/src/88x2bu/88x2bu.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /home/morrow/src/88x2bu/88x2bu.mod.o
  LD [M]  /home/morrow/src/88x2bu/88x2bu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-72-generic'

That is a totally clean compile.

We need to figure out what is causing the failure.

Do the following:

$ sudo apt update
$ sudo apt full-upgrade
$ sudo apt autoremove
$ sudo rm -rf ~/src/88x2bu
$ sudo reboot

Once the system is back up, very carefully go through all of the installation steps again.

Some things to think about:

"sudo" is a weapon. Use it only if you know exactly what the result is going to be. I cringed when you said you were cleaning files out of bla bla bla.

The manual complilation we did tells us that this problem likely has nothing to do with dkms as we took it out of the picture.

w407022008, 2021-04-27:
well, It looks like this compilation error has nothing to do with the kernel.

I tried to do the following:
$ sudo apt update
$ sudo apt full-upgrade
`nothing should be upgraded`
$ sudo apt autoremove
`nothing should be autoremoved`
$ sudo rm -rf ~/src/88x2bu
$ sudo reboot

$ sudo apt-get update
$ sudo apt-get install -y dkms git
`dkms Already the latest version (2.3-3ubuntu9.7)`
`git Already the latest version (1:2.17.1-1ubuntu0.8)`
$ cd ~/src
$ git clone https://github.com/morrownr/88x2bu.git
$ cd ~/src/88x2bu
$ sudo ./install-driver.sh
> Running: install-driver.sh version 20210421
> Start installation.
> Copying source files to: /usr/src/rtl88x2bu-5.8.7.4
> Copying 88x2bu.conf to: /etc/modprobe.d
> 
> Creating symlink /var/lib/dkms/rtl88x2bu/5.8.7.4/source ->
>                  /usr/src/rtl88x2bu-5.8.7.4
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> 'make' -j12 KVER=5.4.0-72-generic KSRC=/lib/modules/5.4.0-72-generic/build...(bad exit status: 2)
> ERROR (dkms apport): binary package for rtl88x2bu: 5.8.7.4 not found
> Error! Bad return status for module build on kernel: 5.4.0-72-generic (x86_64)
> Consult /var/lib/dkms/rtl88x2bu/5.8.7.4/build/make.log for more information.
> An error occurred. dkms build error = 10
> Please report this error.

In addition, I have also tried to make it with what you showed.
$ make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-72-generic/build M=/home/morrow/src/88x2bu
> make: Entering directory '/usr/src/linux-headers-5.4.0-72-generic'
>   Building modules, stage 2.
>   MODPOST 0 modules
> make:  Leaving directory '/usr/src/linux-headers-5.4.0-72-generic'

morrownr, 2021-04-27:
Sigh... I suspect there is an issue with the development environment in your system but I don't have and can't find a clear cut answer. We need to make a decision. We can either continue down this rabbit hole or we can go another direction. Let me throw out some options:

- Do a clean upgrade to a newer version of Ubuntu. 20.04 LTS is available and so is 21.04. You would need to copy any important files to an external storage device or store them online at a place like Google Docs. When you install the new version, select format during the installation so as to have a totally clean installation. Then go through the installation process for the adapter.

- Read the following site: https://github.com/morrownr/USB-WiFi

That site has a lot of information about USB WiFi adapters on Linux. While I enjoy working on this driver, most the time, I realize many Linux users would be happier with an adapter that is plug and play and those are available. In fact, the adapters that use the in-kernel drivers, such as the mt7612u and mt7610u, have far more capabilities than these Realtek drivers and they are fully Linux Wireless standards compliant. Examples:

WPA3 support - WPA3 support works with the mt7612u driver as I was testing it with Ubuntu 21.04 yesterday. Rock solid. It probably works with the mt7610u but I haven't tested that yet... but I will when I have time.

Support for interface combinations - The Realtek drivers don't do that. The Mediatek driver do.

Solid AP mode support - The only Realtek driver that has solid AP mode support is the 8812au. The mt7612u and mt7610u as well as all other in-kernel drivers have solid support for AP mode.

Solid monitor mode support - The only Realtek driver that has good monitor mode support is the 8812au. The mt7612u and mt7610u as well as the other in-kernel drivers (I have tested several) have solid support for monitor mode.

I could go on but the simple truth is that these Realtek drivers are not very good. I work on them simply to help my fellow Linux users that have existing hardware.

Let me know what direction you decide to go. We can continue down the rabbit hole but my to-do list has several other items I need to be working on so this rabbit hole option could go slow.

Nick

w407022008, 2021-04-29:
Hi, Nick. Thank you very much for your detailed guidance and enthusiastic help. I still haven't found the root cause of this issue yet, so he is indeed a rabbit hole, but I have led a cable from the router, so I am currently using Ethernet instead of WiFi. I will not consider this hassle again unless I have to someday. Upgrading to Ubuntu20 might be an option, but some of the packages used currently rely on Ubuntu18, so I can't do that yet. Thanks again Nick, thanks for this repo which I have been using safely for a long time.

closed 2021-04-29 by w407022008