morrownr/88x2bu #60: support for sdio interface

ChalesYu, 2021-05-02:
just find out a way to make sdio interface worked ,pushed to [my repo](https://github.com/ChalesYu/rtl8822bs-aml/tree/test-5.8.7.4-200216) , If I prepare a PR for you can you merge it? 

And the plan is set `CONFIG_USB_HCI = n` and  `CONFIG_SDIO_HCI = y` in Makefile to support this interface.

BTW, https://github.com/ChalesYu/rtl8822bs-aml/commit/27fef585225a52913b2936f91abf36859ec20a43  should can fix [this issue](https://forum.armbian.com/topic/15759-please-help-with-rtl8822bs/)

morrownr, 2021-05-03:
Hi CharlesYu,

If you submit a pull request, I will certainly give it a look.

First, I need to make sure I understand what you are doing.

It appears you have a rtl8822bu based adapter and you are trying to get the bluetooth working?

I'm all for fixing things and adding capability but we must be careful. I have no capability to test a 8822bu based adapter because I don't have one so I would depend on you to test if this works. I could test to see if we are breaking other things.

Regards,

Nick

ChalesYu, 2021-05-03:
Nope , AFAIK, The 8822B have 3 type interface: PCI-E , USB , SDIO .  pci-e type has benn supported by rtw88.
But I only can test on rtl8822bs , I don't have rtl8822bu adapter.

Add sdio support file commit (some copy-paste work):
https://github.com/ChalesYu/rtl8822bs-aml/commit/2a3ecb1d016c057964259d40ec807e11768b783b
https://github.com/ChalesYu/rtl8822bs-aml/commit/6ec143b4b2570042910a80562740405e956461ff
https://github.com/ChalesYu/rtl8822bs-aml/commit/0669bc6b6d2736d17cdb2c71c35fa3614b34ea3f
after that , we can just need to set `CONFIG_USB_HCI = n` and `CONFIG_SDIO_HCI = y` in Makefile to support rtl8822bs, default Makefile support rtl8822bu. 

this commit fix a error on compile progress
https://github.com/ChalesYu/rtl8822bs-aml/commit/f8cdb1677070646c46e9a5dde45eac4c3820fa9a

this commit can force load generic `rtw8822b_fw.bin` in linux-firmware  :
https://github.com/ChalesYu/rtl8822bs-aml/commit/5c5b7ea9e306c85a8cc2259208c14966aa84aea2

you can test if these commit broken other things (like USB support).

The bluetooh on rtl8822bs is connect via UART, and use PCM for bluetooth audio.  I have already get that bluetooth worked, but bluetooth audio still stuck.

morrownr, 2021-05-04:
Okay. Now I understand what you are doing. Would you mind sharing a link to the hardware you are using so I can visualize what you are using?

I am going to create a separate local repo for this testing  as this is a very heavy usage repo and the last thing I need to do is make a mess. I am not feeling well today so this may take me a few days.

Nick

ChalesYu, 2021-05-05:
Hi Nick ,

Thanks for your help.
> Would you mind sharing a link to the hardware you are using so I can visualize what you are using?

I'm using a generic aml gx board with a generic rtl8822bs model to test.

AFAIK, these set-top box have variant using this model : X96 max ,Tanix TX5 max ,Tanix TX6 , Z28 pro

and this chip variant should also work is :
 + RL-SM02F-8822BS
 + 6222B-SRB
 + BL-M8822BS1
 + Cdtech 47822BS-10
 + and More

Regards

ChalesYu

ChalesYu, 2021-06-07:
If you really want me send a link, [here it is](https://forum.freaktab.com/forum/tv-player-support/amlogic-based-tv-players/813731-amlogic-s905l2-tv-boxes), but I think you will not like this hardware.

I wrote a package commit for openwrt, use a dirty way to compile this driver. (Although cmd line working, but luci not )
https://github.com/ChalesYu/openwrt/commit/db21f025d57139d6a2042c0ad53f0e946a9ff193

also, here is a patch to fix compiler can't find some header,you can see if it need to merge , I don't want to open another PR again...
```
diff --git a/Makefile b/Makefile
index 60f290d..4ce976c 100644
--- a/Makefile
+++ b/Makefile
@@ -23,7 +23,7 @@ ifeq ($(GCC_VER_49),1)
 EXTRA_CFLAGS += -Wno-date-time	# Fix compile error && warning on gcc 4.9 and later
 endif
 
-EXTRA_CFLAGS += -I$(src)/include
+EXTRA_CFLAGS += -I$(src)/include -I$(src)/hal -I$(src)/hal/phydm
 
 EXTRA_LDFLAGS += --strip-debug
 
--
```

morrownr, 2021-06-07:
@ChalesYu 

I am so sorry. Somehow this issue was left off of my to-do list at some point. My workload is such that I can't work on this today but will soon. It doesn't bother me if you want to send a message to get my attention if it looks like I have forgot. Regards.

closed 2021-12-05 by morrownr