morrownr/88x2bu #62: VHT in AP mode 802.11AC

gteca, 2021-05-04:
Hi,
I am running the driver in AP mode using VHT configuration, I want to achieve 867 Mbps for research purpose.

Here is my `hostapd.conf`
`### hostapd configuration file
interface=wlxd03745a92803
bridge=br0
driver=nl80211
### IEEE 802.11
ssid=covert-channel
hw_mode=a
channel=136
### IEEE 802.11AC
ieee80211ac=1
auth_algs=1
#require_vht=1
vht_capab=[HTC-VHT][MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE]
#require_vht=1
#vht_oper_chwidth=0
#vht_oper_centr_freq_seg0_idx=42
#vht_oper_centr_freq_seg1_idx=159
### WMM
wmm_enabled=1
### DFS
country_code=PL
ieee80211d=1
ieee80211h=1
### IEEE 802.11i
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=3
wpa_passphrase=secretkey
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
`
88x2bu.conf
`options 88x2bu rtw_drv_log_level=4 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_switch_usb_mode=0
`
Hardware: TP-LInk TU4 Archer AC1300 
OS: Ubuntu 20.04 LTS

iwconfig output
`wlxd03745a92803  IEEE 802.11bg  ESSID:"covert-channel"  Nickname:"WIFI@RTL88X2BU"
          Mode:Master  Frequency:5.68 GHz  Access Point: D0:37:45:A9:28:03   
          Bit Rate:54 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=-100 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
`
piece of iw reg get
`Band 2:
		Capabilities: 0x1863
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		VHT Capabilities (0x03d071b2):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			SU Beamformee
			MU Beamformee
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 867 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 867 Mbps
`
How can I achieve the VHT TX/RX highest support 867 Mbps?

morrownr, 2021-05-04:
Hi gteco,

I'll be glad to help.

Before getting into the details, I need to point something out to you. If you look here...

https://github.com/morrownr?tab=repositories

...you will see that I maintain several repositories related to USB WiFi adapters and Linux. This is gives me the ability to see the big picture as far as USB WiFi support on Linux is concerned. 

My testing has shown that there are currently only 2 chipsets that can reliably sustain 867 Mb/s service in AP mode. The mt7612u and the rtl8812au. Don't use the rtl8812au driver in the Ubuntu repos as it is old and does not work well, Use the rtl8812au driver here. I have a document in the rtl8812au repo that outlines setting up an AP with hostapd on the Raspberry Pi. The parts you need help with are mostly in the hostapd.conf and that is the same whether using a RasPi or Ubuntu on amd64...

The document is called - Bridged Wireless Access Point

It can be found here...

https://github.com/morrownr/8812au

I set the document up such that it should work with any of the Realtek drivers that I have up but, like I said, the only Realtek driver/ chipset combination that I have found to work reliably is the 8812au. With that said, the overall best chipset for AP mode on Linux is the mt7612u. You can find more information here...

https://github.com/morrownr/USB-WiFi

I even have a special repo for the mt7612u chipset here...

https://github.com/morrownr/7612u

...and it has documents showing how to set up both a single band and dual band access point.

Now, back to the subject. I see problems in the files you posted. You need some changes and additions:

wpa=3 > wpa=2 (read the hostapd doc file)

vht_capab=[HTC-VHT][MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE] >

vht_capab=[HTC-VHT][MAX-MPDU-11454][SHORT-GI-80]

[TX-STBC-2BY1] causes problems. Get rid of it.

# Required for 80 MHz width channel operation
vht_oper_chwidth=1
#
# Use the next line with channel 36 (the number below needs to be 6 higher than your channel)
vht_oper_centr_freq_seg0_idx=42

The below switches your adapter into USB3 mode. I assume you want USB3 because you can get nowhere near the performance you are seeking without USB3 support...

88x2bu.conf

rtw_switch_usb_mode=0 > rtw_switch_usb_mode=1

Let me know how it goes and I am going to ask you a favor in return. Could you take the Bridged Wireless Access Point.md document and convert it for use on Ubuntu AMD64 systems. I would certainly appreciate it and you can take credit in the document.

Regards,

Nick

gteca, 2021-05-05:
Hi @morrownr,
Thank you for the hints, I did everything exactly as you wrote, and for me seems like the driver nl80211 supports only Mode IEEE 802.11a/b/g, as the shown logs below
```
nl80211: Added 802.11b mode based on 802.11g information
nl80211: Mode IEEE 802.11g: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472 2484[DISABLED]
nl80211: Mode IEEE 802.11a: 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720[DISABLED] 5745[DISABLED] 5765[DISABLED] 5785[DISABLED] 5805[DISABLED] 5825[DISABLED]
nl80211: Mode IEEE 802.11b: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472 2484[DISABLED]

```

This suspicious is even stronger when after the configuration in the hostapd for 802.11ac:
```
hw_mode=a
channel=36
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

```
I am activating 802.11ac and configure to use 80MHz, and I get the following logs 
```
DFS support is enabled
Completing interface initialization
Mode: IEEE 802.11a  Channel: 36  Frequency: 5180 MHz
DFS 0 channels required radar detection
Could not set channel for kernel driver
Interface initialization failed
wlxd03745a92803: interface state COUNTRY_UPDATE->DISABLED
wlxd03745a92803: AP-DISABLED 
wlxd03745a92803: Unable to setup interface.
 
```
I am able to run the AP if I change the channel width to 20 MHz, it is a prove that channel aggregation one of the features of 802.11ac is not supported by kernel driver. When I run `iw list` is clear that the drive supports 802.11AC and I tested in STA mode, nevertheless in AP mode I read a lot about this problem, for me is strange that nl80211 loads only a/b/g mode, but I don't know where is the source code of nl80211 to see what happens when it receives a request from hostapd. For now it is ok for my research, I will keep diging into the problem and if I find any solution I will reopen this issue. 
As to return a favor, I will below provide the full setup to how to configure AP mode running and providing internet access for Ubuntu using this driver.

gteca, 2021-05-05:
**Brigded Wireless Access Point**
OS: Ubuntu 20.04 LTS
Hardware: T4U AC1300 TP Link

**Install hostapd and dnsmasq**
`sudo apt-get install hostapd dnsmasq`

- Open and save the configuration file for hostapd (Software to interact with Kernel to create 802.11 AP)
`sudo nano /etc/hostapd/hostapd.conf`

Copy and paste the configuration

```
# File Name: hostapd.config
# Description: 802.11a/g/n/ac/ax AP Configuration
# Official hostapd documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf

# Interface description
# @interface - Your WLAN interface (run iwconfig command to check the name) 
# @bridge - Interface name to bridge traffic from adapter interface to the interface connected to the internet 
# @driver - The driver interface for cfg80211 in the Linux Kernel
interface=wlxd03745a92803
bridge=br0
driver=nl80211

# IEEE 802.11
# @ssid - Service Set Identifier, simply the name of your WIFI network
# @hw_mode - The IEEE 802.11 Standard (a = IEEE802.11a [5GHz], b/g = IEEE 802.11b [2.4GHz], n = IEEE802.11n(HT) [2.4/5 GHz])
# For IEEE802.11 ac(VHT)/ax(HE) set hw_mode = a
# @channel - 802.11 Channel (Recommend for 2.4 GHz use: 1, 6, 11, 14 to avoid overlap and for 5GHz use: 36)
ssid=TPLINK_WIFI
hw_mode=a
channel=36

# WMM - Must be Enabled for 802.11n/ac/ax AP to work
# These parameters are used by the access point when transmitting frames
wmm_enabled=1

# IEEE 802.11n
# @ieee80211n - 0 = disable (default), 1 = enable, Remember that 802.11 is dual band, set hw_mode = g(2.4 GHz)/a(5 GHz)
# @disable_11n - 0 = disable, 1 = enable
# @ht_capab: HT capabilities (Guard Interval, Fragmentation, Bandwidth ... See the official documentation)
# @require_ht - Require stations to support HT PHY (reject association if they do not)
# Uncomment the lines below if you wish 802.11n, IN my case I would like to run 802.11ac
#ieee80211n=1
#ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
#require_ht= 0

# IEEE 802.11ac
# @ieee80211ac - Enable 802.11AC
# @require_vht - Require STA to support 802.11AC 
# @vht_capab - 802.11AC VHT capabilities (Beaforming, Channel Width ... See the official documentation)
# @vht_oper_chwidth - Channel Width, 0 = 20[MHz] or 40[MHz], 1 = 80[MHz], 2 = 160[MHz], 3 = 80+80[MHz]
# @vht_oper_centr_freq_seg0_idx - Channel agregation , vht_oper_centr_freq_seg0_idx=42 (80 MHz channel) 
ieee80211ac=1
#require_vht=0
vht_capab=[+HTC-VHT]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

# DFS (Dynamic Frequency Selection) - For cases you want to use channels reserved for radar detection, where the WIFI traffic is low
# Enable to check the legal regulations in your country/region or set it definitive in /etc/default/crda
country_code=PL
ieee80211d=1
ieee80211h=1
# IEEE 802.11i - Authentication/Security
# @macaddr_acl - Access Control List based on MAC Address, macaddr_acl = 0 accepts all mac adress unless in deny list (To create your own ACL See the official documentation)
# @ignore_broadcast_ssid - Ignore probe request with Broadcast SSID, and handle only direct probe request, 0 = disable(default), 1 = enable
# @ auth_algs - 0 = Open Ssystem Authentication, 1 = Shared Key Authentication
# @wpa - 802.11 Authentication algorithm (Use 3)
# @wpa_passphrase - WIFI Password
# @wpa_key_mgmt - Key exchange algorithm 
# @wpa_pairwise - Group key
# @rsn_pairwise - Robust Secure Network
macaddr_acl=0
ignore_broadcast_ssid=0
auth_algs=1
wpa=3
wpa_passphrase=randompwd
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

# All the parameter above are the basic that is needed to 802.11 AP, if you need more please, consult the oficial documentation and add as your need
```

Then we need to tell hostapd where to find the AP configuration
`sudo nano /etc/default/hostapd`
find the line that starts with #DAEMON_CONF and remove #
_DAEMON_CONF="/etc/hostapd/hostapd.conf"_

- Configure dnsmasq (The DHCP Server)
`sudo nano /etc/dhcpcd.conf`
And indicate the configuration of DHCP server which will be the AP
Copy and paste (replace the interface name and fell free to assign any private address network)
```
interface wlxd03745a92803
static ip_address=10.10.0.1/24
nohook wpa_supplicant
```

We need to specify to dnsmasq the IP range and the lease time
`sudo nano /etc/dnsmasq.conf `
Remeber to replace the interface name and is up to you the range of IP to lease
```
interface=wlxd03745a92803
dhcp-range=10.10.0.2,10.10.0.100,255.255.255.0,24h
```
Then we need to tell the dnsmasq what is the interface acting as DHCP server
```
sudo tee /etc/default/dhcpd.conf <<EOF
INTERFACES="wlxd03745a92803"

```
Run DHCP Server
`sudo service isc-dhcp-server start`
Run hostapd
`sudo hostapd /etc/hostapd/hostapd.conf`
For more details you can run hostapd with
`sudo hostapd /etc/hostapd/hostapd.conf&
`
or
`sudo hostapd -dd /etc/hostapd/hostapd.conf
`

- Connect the AP to the internet

The WIFI name will be available and if you connect, the device shall receive an IP Address from the DHCP pool, but the AP will not have internet connection, if you take a look at `hostapd.conf` we need to set the bridge which is the interface that connects the AP interface and the interface connected to the internet in you computer.

Stop network manager
`sudo service network-manager stop`
Assign static IP and turn on the interface of AP (The configure disapear after reboot, there is a way to do it permanently 
, that was my workaround)
```
ip addr add 10.10.0.1/24 dev wlxd03745a92803
ip link set dev wlxd03745a92803 up
```
Add the interface connected to the internet to the bridge and assgin IP to the bridge interface via DHCP
```
sudo brctl addif br0 eth0
sudo dhclient br0

```

The steps worked for me, please if is possible try to reproduce and in case of any problems let me know and then you can add to the repo to help the community.

morrownr, 2021-05-05:
The following are good...

hw_mode=a
channel=36
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

You also need...

ieee80211d=1
ieee80211h=1

And you need...

options 88x2bu rtw_vht_enable=2 rtw_switch_usb_mode=1

in /etc/modprobe.d/88x2bu.conf

The following link gives the entire proven setup as tested on a RasPi:

https://github.com/morrownr/8812au/blob/5.9.3.2/Bridged_Wireless_Access_Point.md

All it takes is one little setting being off and things don't work right. You are welcome to post the entire contends of hostapd.conf.

With all that said, let me point out that I have tested this driver and an 8812bu based adapter many times and 11ac/80 MHz channel width does work. The problem is that it is not stable. It will stop working so this is a case where you can win the battle but lose the war.

If you need a stable AP, my previous recommendations stand.

Nick

gteca, 2021-05-05:
I have the exact same configuration, but I will in the future try the drives that you recommended, thank you very much for the hints.

morrownr, 2021-05-05:
wpa=3

That is a bad setting. It needs to be wpa=2

morrownr, 2021-05-06:
gteca,

I copied the document you sent above. I will be working on it and testing it. I will post it for your testing as I am able.

Some things to point out: You titled the document Bridged Wireless Access Point (BWAP) but what you are describing is actually a Routed Wireless Access Point (RWAP).  A RWAC is basically what we think of as a WiFi Router. A BWAP (or Dumb AP) is simply an extension of an existing network. A RWAC is far and away more complicated to setup and maintain. There is a place for both and we can produce documents for both but which is it that you actually need at this point?

Nick

gteca, 2021-05-06:
Hi Nick,
Currently, for my research, I need an AP that behaves as RWAP which is able to create a BSS, exchange 802.11 frames and provide an internet connection to STA clients associated with it, that is why I provided the configuration in such way, besides I am modifying the drive code to implement few changes, I use the drive for research purpose.

morrownr, 2021-05-06:
Quote: " I am modifying the drive code to implement few changes, I use the drive for research purpose."

Reply: "I do not understand what this means. When you say 'drive code", what are you speaking about?"

morrownr, 2021-05-06:
Okay. I understand. Driver code.

I've been looking things over. Using Ubuntu is going to cause me to have to do some research because it has some moving parts in the networking stack that I am not familiar with so I think the best plan of attack on this end of things is for me to setup and document a BWAP. Once it is solid, then I convert it over to a RWAP.

Going directly into a RWAP setup using some things I am not yet familiar with is just too many moving parts for now.

Question: I have an adapter based on the rtl8812bu chipset. In fact, I have two so I can try with an 8812bu adapter but I would prefer using an adapter that uses a 8812au or 7612u chipset since I know the drivers for them work well in AP mode. The problem with using this driver is how buggy it is in AP mode. Will we know if the problem is our setup or the driver?

morrownr, 2021-05-06:
A short note from my work so far: 

It might be a good idea for you to start using "iw" instead of "iwconfig"

iwconfig is depreciated.

https://wireless.wiki.kernel.org/en/users/documentation/iw

Something else to keep in mind while working on this project, The Realtek out-of-kernel drivers are not Linux Wireless standards compliant in many ways. Various commands may give you false information. Just something for you to keep in mind.

morrownr, 2021-05-06:
Okay. I have a desktop system configured for this project. It has a CAT 6 ethernet connection with internet service and it has a 8812bu adapter connected via USB3 port. There is a clean Ubuntu 21.04 installation. I know, that is not 20.04 but 21.04 is what I had on a stick ready to install. Close enough for now.

I am going to try your instructions and see what happens.

I am setting up to put the master document online in a repo here so we can both work from it.

FYI: I am also working some other projects so I am available when I am available.

Nick

gteca, 2021-05-06:
Hi Nick, 
No problem, you can try out the instructions and let me know about the progress when you will be available :)

morrownr, 2021-05-07:
Hi gteca,

I was able to finish stepping through your instructions this morning. I did not end up with an AP. Let me ponder this situation.

gteca, 2021-05-07:
Hi Nick,
Could you provide the output after you run the `hostapd` command and the interface state?

morrownr, 2021-05-07:
Run hostapd
$ sudo hostapd /etc/hostapd/hostapd.conf
Configuration file: /etc/hostapd/hostapd.conf

Line 64: invalid WPA passphrase length 6 (expected 8..63)
WPA-PSK enabled, but PSK or passphrase is not configured.
2 errors found in configuration file '/etc/hostapd/hostapd.conf'
Failed to set up interface with /etc/hostapd/hostapd.conf
Failed to initialize interface

$ sudo brctl addif br0 eth0

sudo: brctl: command not found

-----
Let me offer some suggestions:

It might benefit your effort if you break your work system harddrive into multiple partitions. That would give you one or more extra partitions that can be used for clean/default installations of your test operating system. You need to start from a given situation or you really don't know if your instruction document is good. Why do I say this? A lot of experience and the above result "sudo: brctl: command not found" which tells me that you have installed something that you did not document. I install fresh OS's sometimes dozens of times per week in my little lab.

Provide a break between each separate step in the document...
-----
Maybe something like this...
-----
Do you see how this can help?
-----

Also, to help clarify what is a written instruction for the human and  what is a command line instruction, maybe put a command prompt in front of the command line instruction like this...

$ sudo nano /etc/hostapd/hostapd.conf

------
Another thing: You must be very very careful. I had to correct a lot of typos. Typos in .conf files and on the cli will only lead to errors and questions about what went wrong. The attention to detail has to be perfect.
-----
Let's back up for a minute to make sure I understand what you are doing:

You want a solid routed access point setup on Ubuntu running on an amd64 based system with ethernet supplied internet and a rtl8812bu based usb wifi adapter to provide the wifi. Once that setup is stable and working, you plan to work on the code in this driver to do some testing.

Is the above an accurate statement?

Nick

gteca, 2021-05-07:
I Nick,
I wrote down the remarks, indeed I forgot to mention the installation of  `brctl` command and string length `wpa_passphrase` has to be more within 8 - 63 characters. After fixing these errors, did you managed to have AP mode running on the interface?

By the way, the answer to your question is yes, I want a solid routed access point setup on Ubuntu running on an amd64 based system with ethernet supplied internet and a rtl88x2bu based USB wifi adapter to provide the wifi, which I managed to do in mode 802.11a, but I want it to work in 802.11ac.

morrownr, 2021-05-08:
Hi gteca,

Quote: " After fixing these errors, did you managed to have AP mode running on the interface?"

Reply: I was able to get an AP mode interface going. I was not able to finish the work on the bridging. I had to stop where I was yesterday. I have to be on the road for a couple of days so I plan to start working on this project again as able sometime during the following week.

I did put a copy of a document that is in-progress online:

https://github.com/morrownr/7612u/blob/main/Bridged_Wireless_Access_Point-Ubuntu-1.md

If isn't a copy of your document but rather one that is more closely formatted to the style I use online for my repos. As we progress we can compare each other's documents.

I should be able to reengage on this project again sometime this coming week.

Nick

gteca, 2021-05-09:
Hi Nick,
Great! I am also working on the issue whenever I am available, and let you know if something changes.

morrownr, 2021-05-11:
Geovani,

Below is a link to the new Ubuntu guide. It works here but it is really an alpha version. It is not well documented so please make notes for me. 

https://github.com/morrownr/7612u/blob/main/Bridged_Wireless_Access_Point-Ubuntu-1.md

I know that you are looking for a routed access point setup, but if you hang in there with me while I solidify a bridged setup, that would be great. I realize the new guide is poorly documented at this point but it is working here on Ubuntu 21.04.

closed 2021-08-16 by morrownr