morrownr/88x2bu #64: ad hoc problem

Dunkem, 2021-05-08:
I have two EDIMAX wireless devices with RTL8822BU chipset, and I want to build ad hoc network. But I follow https://help.ubuntu.com/community/WifiDocs/Adhoc official docs, it didn't work for me, so I want to know if this driver supports ad hoc?

morrownr, 2021-05-08:
Hi @Dunkem 

The short answer is that I do not know because I have never had cause to test it. I try to keep the README accurate as best I can but Realtek does not provide any docs. If you run the following command, you can see what the driver says it can do:

$ iw list

One big issue that comes up a lot with these Realtek drivers, and I support 5 Realtek drivers, is that they are not Linux Wireless standards compliant. That means that they often don't work well with system utilities or various networking apps or setup instructions that are written in accordance with Linux Wireless standards. This issue has come up so much that I started and maintain a site with info on what and where users can obtain USB WiFi adapters that are Linux Wireless standards compatible:

https://github.com/morrownr/USB-WiFi

That site contains a LOT of information.

If you want to pursue this issue, I will try to help but it will probably be at least one week before I can look at setting up to do testing as I have other projects in progress. Let me know.

Nick

Dunkem, 2021-05-09:
thank you for your reply. And the fellowings are what I get after I run iw list 
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
I see it can support IBSS mode which is ad-hoc mode, but it seems not work. I really happy if you can test this. And I will continue test or find other solution in my project.

closed 2021-12-05 by morrownr