morrownr/88x2bu #65: It's also working on TP-Link Archer T4U Plus AC1300 (dual)

PirvuCatalin, 2021-05-13:
Just wanted to mention that it's also wroking for TP-Link Archer T4U Plus AC1300 (dual). And to thank you!!

morrownr, 2021-05-13:
You are welcome. Glad to hear things are working for you. This driver along with an adapter with a 8812bu chipset provide very fast performance in managed (client) mode. It should serve you well. If you decide at some point to try AP mode or monitor mode, I must warn you that there are much better options available. I've worked on this driver and just have not been able to fix some things. Take care.

PirvuCatalin, 2021-05-13:
Surprisingly, I'm probably going to use it in AP mode, but haven't got time to test it. I'll be using it on a Raspberry Pi 4 and hoped to be able to server as an AP for a couple of clients at ~50m  distance, outdoor. I'll see how it goes.

morrownr, 2021-05-13:
I'm a fan of the RasPis also. I use two Dumb APs here and one of them is a RasPi4b. Outstanding AP. Mine is dual band.

Some things that might help you along from a guy that has been there:

In the follow repo is a file called Bridged_Wireless_Access_Point which will give you a big push on getting the Pi4b set up as an access point. The repo is the for the 8812au driver but the settings will work on this driver. The reason the file is there is because that driver and chipset work well as an AP, unlike this driver and chipset: (but I know you will have to prove it to yourself)

https://github.com/morrownr/8812au

There is also this repo that contains some more documents about setting up the Pi4b to be an access point:

https://github.com/morrownr/7612u

Be careful, one of the documents is in testing but it will tell you that when you open it up. This site is using adapters based on the mt7612u.

My experience is that there are only two chipset driver combinations that are fully stable and capable of full 80 channel width 5g performance... rtl8812au and mt7612u. The mt7612u does hold the advantage in that the driver is in the Linux kernel (and in the RasPiOS) waiting on you. The mt76 driver provides several advantages including WPA3 support. More info here:

https://github.com/morrownr/USB-WiFi

For what it is worth: I have worked on this driver and have not been able to improve AP mode support. It is frustrating. Good luck.

PirvuCatalin, 2021-05-13:
You are awesome! Thanks for the info, it's really nice to see all the details of the AP setup and not only "defaults".
Just to give you an insight, the project I'm working on is for my BSc and there are lots of other alternatives for outdoor communication in ranges of 50m - 150m (like LoraWan), but I'm looking for something a little "friendlier" so to say, based on WiFi. 

In the next 2-3 weeks I'll test a couple of different USB WiFi devices (I'll use your drivers, if compatible) and maybe I can share some results here, like the distance and data transfer.

morrownr, 2021-05-13:
This sounds interesting. I have several usb wifi adapters so if you want to set the parameters for me, I can do some testing here.

rtl8812bu
rtl8812au
rtl8814au
rtl8811cu
rtl8811au
mt7612u
mt7610u
ar9271
and more

I'm sure you realize this but there can be a big difference in communications capability between adapters. Even adapters with the same chipset. The quality of the amp and the quality of the antennas, etc. Some things to watch out for:

The Raspberry Pi 3b, 3b+ and 4b only allocate 1200mA of power to the usb subsystem. Many 8812au based adapters can use up to around 800 mA at heavy load. Adapters based on the 8814au chipset can and do go over 950 mA at heavy load. Your 8812bu chipset based adapter can go up over 500 and maybe close to 600 mA. It depends on the adapter. With that being said, combine the power needs of many adapters with the fact that usb3 powered hubs basically do not work well with RasPis due to backfeeding and you have to be very careful not to cause a brown out...even with a good power supply.

Now you will see what I am a fan of the mt7612u adapters for use with the RasPis. My Alfa AWUS036ACM, when working as an AP on my Pi4b, which it normally does, can be pushed as hard as you can push it with iperf3 and other connections and it maxes out at 380 mA. No kidding. It it like that specific adapter was made to work with RasPis. In fact, it leaves enough juice left over for me to hook up an external 500gb ssd for super speed io.

Anyway, lots of fun to be had and lots of ways to save more on things like expensive wifi routers and the like.

lucky930, 2021-06-11:
DKMS make.log for rtl88x2bu-git for kernel 5.10.0-kali8-amd64 (x86_64)
Friday 11 June 2021 12:00:03 PM IST
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-kali8-amd64/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-kali8-amd64'
Makefile:1: /usr/src/linux-headers-5.10.0-kali8-common/Makefile: No such file or directory
make[1]: *** No rule to make target '/usr/src/linux-headers-5.10.0-kali8-common/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-kali8-amd64'
make: *** [Makefile:2378: modules] Error 2

Please help on this..been trying for 3 days and now no option left other than posting on hub for the resolution.!!!!

morrownr, 2021-06-11:
You really should have started a new issue.

Try this and then run sudo ./install-driver.sh again:

$ sudo apt-get install -y linux-headers-$(uname -r) build-essential dkms git libelf-dev bc

Not a lot of folks running Kali use a 8812bu adapter. Can I get you to give me a report on monitor mode?

lucky930, 2021-06-11:
The issue is I've TP link archer T4U V3 device and I'm not able to install it. 
It finally installed in kali inside virtualbox but not in the original installed image. 

# hostnamectl             
   Static hostname: localhost
         Icon name: computer-laptop
           Chassis: laptop
        Machine ID: 86bc68a3634e4cdba861966ecf01320f
           Boot ID: 91b535b0f22d46cbbc199f17ca4bfb87
  Operating System: Kali GNU/Linux Rolling
            Kernel: Linux 5.10.0-kali8-amd64
      Architecture: x86-64

On the virtualmachine, this below link helped me to get in working:
https://github.com/RinCat/RTL88x2BU-Linux-Driver
but not in the installed kali. Both installed and virtualbox images are latest(as shown in hostnamectl output)

I think the kernel not allowing me to install the driver but how come it worked in vm image????

PS: Thank you for your fast response :)

lucky930, 2021-06-11:
OK, @morrownr  you are the magician and below your magic command helped me to resolved it.

$ sudo apt-get install -y linux-headers-$(uname -r) build-essential dkms git libelf-dev bc

at least the device is working(I'm able to see the adapter now),

Many thanks to you :+1: 

PS: can u let me know what that command did "build-essential dkms git libelf-dev bc" ? so that I can know technically.

morrownr, 2021-06-14:
I'm not a magician but you are welcome. 

Now I know you did not follow the installation instructions...or you skipped a step.

That line is installing things like a C compiler and the Linux kernel headers that are needed to compile the driver. You need git to download the driver from here. dkms is used with the installation scripts that I have made. It is a utility made by Dell that does a lot of cool things like it keeps you from manually having to reinstall this driver every time a kernel upgrade flows into your system.

closed 2021-12-05 by morrownr

imtester122, 2024-06-10:
hello is it actually clear  this version supports monitor mode or not? if it does support it, please take  the steps  to activate monitor mode. thank you >:)

morrownr, 2024-06-10:
Monitor mode is turned on with this driver. That does not mean that Realtek's monitor mode is good. It is not.

For monitor mode users, I have recommendations:

The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

I recommend reading menu item 1 and taking a look at menu items 2 and 3.

The Plug and Play List has some adapters that are really good at monitor mode. The most popular are probably the Alfa ACM and ACHM (range) and many folks use the AXM and AXML. There are other adapters in the list that would work well also. You can use Issues at the Main Menu page to ask questions.

imtester122, 2024-06-10:
this means whether i should return the product or not because i ordered it previously :/

morrownr, 2024-06-10:
I assume since you are in this repo that your adapter has a 8812bu or 8822bu chipset. If that is the case and you are looking to do or learn monitor mode work such as wireless security testing and the like, I would send the adapter back to get my money back. Then follow my instructions above. There are good adapters and chips for what you are looking to do, none of them currently have Realtek chips.

I am aware that there is a lot pure bs on the internet as far as what adapters to run with Kali or other platforms to do pen testing and security analysis. There is probably no better place on the internet to actually get good information about what to use than right here at the Main Menu. Exactly what adapter is best can depend on what you are doing and a lot of factors like if you will be traveling or if you also need 6 GHz band support so asking questions is a good way to go.

Let me know if you have additional questions.