morrownr/88x2bu #68: Unable to set the "4addr at on"

sartoriuz, 2021-05-17:
I have Mint 20.1 with the 5.4.0-72 kernel and in the current version 5.8.7.4 it is not possible to set the 4addr on: RTNETLINK answers: command failed: Operation not supported (-95).

morrownr, 2021-05-18:
I need to be able to duplicate the problem. Please provide a list of steps that will allow me to duplicate the problem. Also, give a general idea what you are trying to accomplish.

I have a system with Mint 20.1 and an adapter that runs this driver.

Regards.

sartoriuz, 2021-05-18:
I use a simple commands:
iw dev
phy#4
	Interface wlan0
		ifindex 14
		wdev 0x400000001
		addr 1c:bf:ce:35:a1:fc
		type managed
		txpower 42949572.96 dBm
ip link set  wlan0 up
iw wlan0 set 4addr on
command failed: Operation not supported (-95)
or
ip link set  wlan0 down
iw wlan0 set 4addr on
command failed: Operation not supported (-95)

I need add wlan0 to the bridge

morrownr, 2021-05-18:
sartorius,

I am very busy today but I will point you to an issue that was resolved in another of my repos:

https://github.com/morrownr/8812au/issues/9

While not exactly the same issue, the basic idea may be the same. Just because it isn't working one way doesn't mean another way won't work. Something that I am only too aware of is that these Realtek drivers are not Linux Wireless standards compliant so they don't always work the official way and sometimes they simply lack features. 

I use usb wifi adapters that use Realtek drivers and also usb wifi adapters that use in-kernel drivers. I prefer the adapters with in-kernel drivers as they work the way I expect. It can be frustrating working with these Realtek drivers.

morrownr, 2021-09-02:
Is this issue still active?

closed 2021-12-05 by morrownr