morrownr/88x2bu #69: install error

ln90s, 2021-05-20:
Sorry, I got an error when using install.sh, I want to use 8812bu as AP, but this does not work, the following is my information

wireless adapter : `Realtek 8812bu`

system info : `Linux version 3.10.0-1160.25.1.el7.x86_64 (mockbuild@kbuilder.bsys.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-44) (GCC) ) #1 SMP Wed Apr 28 21:49:45 UTC 2021`

kernel: ``Linux localhost.localdomain 3.10.0-1160.25.1.el7.x86_64 #1 SMP Wed Apr 28 21:49:45 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux``

error print: 
```
`Start` installation.
Copying source files to: /usr/src/rtl88x2bu-5.8.7.4
Copying 88x2bu.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl88x2bu/5.8.7.4/source ->
                 /usr/src/rtl88x2bu-5.8.7.4

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
...(bad exit status: 2)1160.25.1.el7.x86_64 KSRC=/lib/modules/3.10.0-1160.25.1.el7.x86_64/build
Error! Bad return status for module build on kernel: 3.10.0-1160.25.1.el7.x86_64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.4/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error

```

error log:
```
DKMS make.log for rtl88x2bu-5.8.7.4 for kernel 3.10.0-1160.25.1.el7.x86_64 (x86_64)
Thu May 20 10:40:14 CST 2021
 M=/var/lib/dkms/rtl88x2bu/5.8.7.4/build  modules.10.0-1160.25.1.el7.x86_64/build
: No such file or directory.  Stop.25.1.el7.x86_64/build
make: *** [modules] Error 2
```

Please help me, Thank you!

ln90s, 2021-05-20:
System Info: `CentOS Linux release 7.9.2009 (Core)`

morrownr, 2021-05-20:
Good morning In90s,

I am hesitant to work on issues with distros that use kernels that are no longer supported. There are many reasons. For me to fix the issue, I would have to download and install the distro you are using. My to-do list is fairly long so a major project like this would not happen very soon. However, even if I were to do this, you might not like the results. You would end up using a version of software, such as hostapd, that is old and probably does not support this more modern driver, not to mention most guides and help you can get might not apply to what you are doing with that very old distro. Setting up an AP with this adapter is something that needs to be done with a modern, supported kernel.

If you will give me some details and about your computer hardware and your goals for the AP, I can suggest a Linux distro that should work well. I even have some guides that may help you.

Regards,

Nick

morrownr, 2021-05-26:
I am closing this issue due to the lack of a response.

closed 2021-05-26 by morrownr