morrownr/88x2bu #71: Bridged Wireless Access Point - Testers Needed

morrownr, 2021-05-31:
With the recent patches that appear to have fixed AP mode in this driver, I decided to upload a file called [Bridged Wireless Access Point](https://github.com/morrownr/88x2bu/blob/5.8.7.4/Bridged_Wireless_Access_Point.md) here in the repo. It gives guidance for using AP mode to make an access point. Please feel free to test this document and provide feedback. I'm sure there is room for improvement. I use a dual band setup like is shown in this guide. It is extremely reliable. While I have tested Realtek based adapters in my setup, the USB WiFi adapters that are a permanent part of the setup include a mt7612u based adapter for 5 GHz and an ar9271 based adapter for 2 GHz. These adapters allow WPA3-SAE to work as well as other features that are not available with Realtek based adapters.

Also, if anyone wants to modify the document to support a Routed Wireless Access Point, please do so and I'll be glad to post it so others can test it.

Regards

christian1211, 2021-09-30:
It works perfectly, thank you a lot!
But one question left... I want to open an AP with 2,4GHz and 5GHz simultaneos with only one WiFi adapter, how is this possible if I have two hostaps configs and each of them needs an adapter?

morrownr, 2021-09-30:
Hello @christian1211 

You are welcome. Glad to hear you are having success.

>But one question left... I want to open an AP with 2,4GHz and 5GHz simultaneos with only one WiFi adapter, how is this possible if I have two hostaps configs and each of them needs an adapter?

To have a dual band setup, you will need 2 usb wifi adapters or in the case of a RasPi4b, you could use the internal wifi as one of the wifi devices though I don't really recommend that.

These usb adapters have one radio inside. To have simultaneous operation on both bands, you need two radios. Your typical wifi router has two radios inside... actually some have three radios in this day in time.

morrownr, 2021-09-30:
@christian1211 

I forgot to mention. You can find a lot of information about usb wifi adapters at this repo:

https://github.com/morrownr/USB-WiFi

If you have questions about adapters. you can post an issue in that repo.

Regards

morrownr, 2021-09-30:
> Thank you for information. Anyway only 5 GHz should be enough as an AP I guess since most of the devices these days support 5 GHz WiFi.

It all depends on what you need and what makes you happy. Generally you will get better range with 2.4 GHz but then 5GHz is a much bigger pipe. RasPi access point has both bands as I need 2.4 GHz for 2 devices that need the better range and 2 devices that aren't capable of 5 GHz. Everything else goes on 5 GHz. 

> Just for information, I got around 260 Mbps at 5GHz and around 50 Mbps at 2,4 GHz. Very good in my opinion.

That is moving data at a pretty good clip so your configuration must be good.

> As a second try I used only some points of your guide to set up a Raspberry Pi 4 B (1 GB) with RaspAP and wlan0 as AP and wlan1 as client, each of them have a WiFi stick with RTL8812BU chip and seems working fine.

Interesting. I may have to test that again here when I get some time.

christian1211, 2021-09-30:
I think it's about your hostapd config, it seems to be very effective.

Just for completion, I used this WiFi adapter (one for AP and one for client):
https://www.amazon.de/CSL-Adapter-1200MBit-Wireless-Netzwerk/dp/B089YTSC6P/

Yes. Please tell me if you have any issues with this, I was working for many days on it until it worked.

closed 2021-12-05 by morrownr