morrownr/88x2bu #72: Confused on which adapter to choose from

vvavepacket, 2021-06-13:
Hello All,

I got a new system
Mobo: X570 MSI Gaming Plus
Linux: Ubuntu 21.10 Kernel 5.12.9

It has no wifi on it, and Im shopping for wifi adapters. Most wifi I can find have chipset of RTL8812BU.

There are many USB adapter that use the same chipset, is there any difference? Like I can go with the [cheapest $7 one](https://www.amazon.com/Adapter-1200Mbps-Whew-Wireless-10-4-10-12/dp/B0829HWNMV/ref=cm_cr_arp_d_product_top?ie=UTF8), mid one [$10 on Amazon](https://www.amazon.com/Adapter-1200Mbps-Whew-Wireless-10-4-10-12/dp/B0829HWNMV/ref=cm_cr_arp_d_product_top?ie=UTF8), or go on the expensive one $17 (Cudy WU1300S)

Also is my mobo and kernel supported by all of these 3 hardware I stated? Why should one expect the expensive one instead of cheapest one? All of them have the same chipset right?

morrownr, 2021-06-13:
@vvavepacket 

Hello, it is nice to meet you. You are wise to seek advice prior to purchase. I'll be glad to give you the information that I know.

You are seeing a lot of adapters based on the rtl8812bu on the market but that is not necessarily what is going to serve you best based on your operating system. I recommend you read the entire document at the following link:

 https://github.com/morrownr/USB-WiFi

That link is also a github repo so you can post questions in Issues there as well.

Given that you are looking for a Class AC1200/1300 adapter and it appears this is a desktop computer, my first suggestion would be to look at a Alfa AWUS036ACM. I realize that the $35 to $40 price is a little higher than what you were looking at but I'll recommend you consider everything that is involved. This Alfa adapter is Plug and Play with your operating system. That means you will never have to chase down and compile a driver because the driver is in the kernel and is Linux Standards compliant and supports WPA3. The Alfa adapter is a premium quality adapter that should last for a long time.

If you would like additional suggestions, let me know two or more of your favorite online retailers and I will send some links with additional suggestions.

Regards,

Nick

vvavepacket, 2021-06-13:
Hey Nick thanks a lot for your advise. I guess its a little expensive, but it works out of the box and save me some headache. I invested a lot on the machine im building so I guess its logical to not cheap out and try the experimental drivers.

Much love 💕

vvavepacket, 2021-06-13:
Can you send suggestions? My favorite retailer is Amazon and Ebay

morrownr, 2021-06-13:
No problem. I was trying to be nice with my words but to use your words, don't buy a nice computer and then cheap out on your usb wifi adapter. My opinion is that there is more junk on the market than there are good usb wifi adapters. I started the USB-WiFi site to help Linux users find the good quality usb wifi adapters that will work well on Linux. I do not recommend Realtek based adapters to users spending new money. Yes, I maintain this and 4 others Realtek driver repos here but that is for Linux users with existing adapters.

Given that you have  desktop system, you likely have the option of using a PCIe wifi card as well. If you go that route, I suggest you check in at the Ubuntu Networking and Wireless forum for suggestions. Cards using the Intel chipsets are probably what would serve your needs. While PCIe and mPCIe cards can and do work well, they don't provide the flexibility of a USB adapter as you can use a USB adapter on desktops, laptops, Raspberry Pi's and anything with a USB port... so I lean toward USB adapters because I can repurpose the usb adapters as time passes and my needs change.

I just checked availability of the Alfa AWUS036ACM with the links I have up and it appears several retailers are currently out of stock. This is understandable as Linux users around the world have been increasing the demand for this adapter. I see that DBOTH on Amazon still has some in stock for $40. I'm sure retailers are working to restock. Why is demand from Linux users high for this Alfa adapter? Because it is a good quality adapter and it "just works."

morrownr, 2021-06-14:
Amazon:

[TEROW ROW02FD](https://www.amazon.com/gp/product/B086L3D3NB)

Rokland:

[ALFA AWUS036ACM](https://store.rokland.com/collections/wi-fi-usb-adapters/products/alfa-awus036acm-802-11ac-dual-band-2-4-5-ghz-wifi-usb-adapter)

Walmart:

[COMFAST CF-WU785AC](https://www.walmart.com/ip/Zewfffr-COMFAST-CF-WU785AC-1300M-WiFi-Adapter-USB-3-0-4-Antennas-Wireless-Receiver/789726734)

[COMFAST CF-WU782AC](https://www.walmart.com/ip/COMFAST-CF-WU782AC-Wireless-Network-Card-Dual-Antenna-High-Power-WIFI-Signal-Extender-1300Mbps-Wireless-Routing-AP/728697397)

On ebay search for:

"ALFA AWUS036ACM"
"COMFAST CF-WU782AC"
"CF-WU785AC"
"CF-7500AC-V2.0"

All of the above adapters are based on the mt7612u or mt7612un chipset which is supported in kernel. If something gets your attention, post the link back here and I will do a double check on.

vvavepacket, 2021-06-14:
I ended up getting this
https://www.amazon.com/dp/B086L3D3NB/ref=cm_sw_r_cp_apa_glt_fabc_1MA7QNH15M0R4R4757GE?_encoding=UTF8&psc=1

Thanks

morrownr, 2021-06-14:
I have one of those. If you need a driver:

https://github.com/torvalds/linux/tree/master/drivers/net/wireless/mediatek/mt76

This is a joke. Look closely at the link. When you see "github.com/torvalds/linux" that means it is in the Linux kernel.

What is not a joke is that the link is for the driver for your adapter.

I'd appreciate you coming back to make a report on the adapter after you get it.

Nick

vvavepacket, 2021-06-14:
Well the wifi adapter arrived today and guess what. It worked out of the
box!!!

[image: image.png]

Saved me some headache..

Thanks a lot Nick... This repo is very helpful

Ticket can be closed

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
06/14/21,
07:33:50 PM

morrownr, 2021-06-15:
How about we leave the ticket open for a while. Others might have the same issue.

Glad it worked for you out of the box. One other thing that you may not need yet, but it is there, is that it is WPA3 ready. The Realtek drivers can't do that yet. Let us know how the performance is.

vvavepacket, 2021-06-15:
Sure let it open then..

Ping times are okay, I recorded any drop outs, and there were 0 packet losses.. Which is a good thing

[image]

in terms of speed, its able to max out my 1Gbps connection / fiber at home which is amazing as well

morrownr, 2021-06-15:
You are seeing pretty much what I have seen out of this adapter. It is not fancy and it doesn't cost much but it is a solid adapter. Mine mostly pulls duty in AP mode these days and it shines at that as it sustains 400+ Mb/s output when connected to AC1200+ clients in iperf3 tests. I keep a repo just for adapters that use the mt7612u chipset:

https://github.com/morrownr/7612u

There is not much there other than some info and a guide or two but it is there.

You might consider doing a review at Amazon to let fellow Linux users know and it you know any other Linux users, spread the word about this repo:

https://github.com/morrownr/USB-WiFi

Yes, I know it is a long read but it has a ton of info for Linux users that run or want to run a usb wifi adapter on their system. Note that I only have one review below the adapter you bought so if you want to write one up and let me post it, go for it.

Thanks

closed 2021-12-05 by morrownr