morrownr/88x2bu #74: Update vendor list

swinkapepa, 2021-06-25:
Hi

Could you add this device to vendor list ,thanks.

https://www.amazon.com/1200Mbps-Adapter-802-11Ac-Rtl8812Bu-Rodalind/dp/B08BLC7GHB/ref=sr_1_5?dchild=1&keywords=rtl8812bu&qid=1624633376&sr=8-5

lsusb

Bus 003 Device 004: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

spcharc, 2021-06-25:
https://github.com/morrownr/88x2bu/blob/cc94acc06fb7fb85ec27f4983953bef1fe35a474/os_dep/linux/usb_intf.c#L270
It is already supported by the driver.

morrownr, 2021-06-26:
You are wanting me to add it to the Compatible Devices section of the README?

If so, I'll be glad to do so. I went to the Amazon listing. I see "Rodalind" as the brand but I don't see a model number or anything to let folks know what it is. What do you think we should do?