morrownr/88x2bu #75: can't enable 802.11r

ZeroChaos-, 2021-07-16:
okay, so I know that 802.11r isn't even in the makefile, but it is in the code.  if I manually enable it, the errors don't look too crazy.  any chance you care to try to an fix this?

```
  x86_64-pc-linux-gnu-gcc -Wp,-MMD,/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/.rtw_mlme_ext.o.d -nostdinc -isystem /usr/lib/gcc/x86_64-pc-linux-gnu/10.3.0/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=1024 -fstack-protector-strong -Wimplicit-fallthrough -Wno-unused-but-set-variable -Wno-unused-const-variable -fomit-frame-pointer -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -fplugin=./scripts/gcc-plugins/structleak_plugin.so -fplugin=./scripts/gcc-plugins/randomize_layout_plugin.so -fplugin=./scripts/gcc-plugins/stackleak_plugin.so -fplugin-arg-structleak_plugin-byref-all -DSTRUCTLEAK_PLUGIN -DRANDSTRUCT_PLUGIN -DSTACKLEAK_PLUGIN -fplugin-arg-stackleak_plugin-track-min-size=100 -fplugin-arg-stackleak_plugin-arch=x86 -O1 -Wno-error -Wno-unused-variable -Wno-unused-label -Wno-unused-function -Wno-date-time -I/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/include -I/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/platform -I/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/hal/btc -DCONFIG_MP_INCLUDED -DCONFIG_POWER_SAVING -DCONFIG_BT_COEXIST -DCONFIG_EFUSE_CONFIG_FILE -DEFUSE_MAP_PATH=\"/system/etc/wifi/wifi_efuse_88x2bu.map\" -DWIFIMAC_PATH=\"/data/wifimac.txt\" -DCONFIG_LOAD_PHY_PARA_FROM_FILE -DREALTEK_CONFIG_PATH=\"/lib/firmware/\" -DCONFIG_TXPWR_BY_RATE=1 -DCONFIG_TXPWR_BY_RATE_EN=1 -DCONFIG_TXPWR_LIMIT=1 -DCONFIG_TXPWR_LIMIT_EN=0 -DCONFIG_RTW_ADAPTIVITY_EN=0 -DCONFIG_RTW_ADAPTIVITY_MODE=0 -DCONFIG_IEEE80211W -DCONFIG_RTW_80211R -DHIGH_ACTIVE_HST2DEV=0 -DCONFIG_BR_EXT '-DCONFIG_BR_EXT_BRNAME="'br0'"' -DCONFIG_WIFI_MONITOR -DCONFIG_RTW_NAPI -DCONFIG_RTW_GRO -DCONFIG_RTW_NETIF_SG -DCONFIG_LED_CONTROL -DCONFIG_SW_LED -DCONFIG_RTW_SW_LED -DCONFIG_LED_ENABLE -DCONFIG_RTW_DEBUG -DRTW_LOG_LEVEL=0 -DCONFIG_RTW_UP_MAPPING_RULE=0 -DCONFIG_DISABLE_PHYDM_DEBUG_FUNCTION -DDM_ODM_SUPPORT_TYPE=0x04 -DRTW_VHT_2G4=1 -DCONFIG_GTK_OL -DCONFIG_LITTLE_ENDIAN -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT -I/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/hal/phydm -DCONFIG_RTL8822B  -DMODULE  -DKBUILD_BASENAME='"rtw_mlme_ext"' -DKBUILD_MODNAME='"88x2bu"' -D__KBUILD_MODNAME=kmod_88x2bu -c -o /var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.o /var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.c
In file included from /var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/include/drv_types.h:31,
                 from /var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.c:17:
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.c: In function ‘OnAction_ft’:
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/include/ieee80211.h:1459:38: error: subscripted value is neither array nor pointer nor vector
 1459 | #define is_zero_mac_addr(Addr) ((Addr[0] == 0x00) && (Addr[1] == 0x00) && (Addr[2] == 0x00) &&   \
      |                                      ^
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.c:6965:7: note: in expansion of macro ‘is_zero_mac_addr’
 6965 |   if (is_zero_mac_addr(&pframe_body[8]) || is_broadcast_mac_addr(&pframe_body[8])) {
      |       ^~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/include/ieee80211.h:1459:59: error: subscripted value is neither array nor pointer nor vector
 1459 | #define is_zero_mac_addr(Addr) ((Addr[0] == 0x00) && (Addr[1] == 0x00) && (Addr[2] == 0x00) &&   \
      |                                                           ^
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.c:6965:7: note: in expansion of macro ‘is_zero_mac_addr’
 6965 |   if (is_zero_mac_addr(&pframe_body[8]) || is_broadcast_mac_addr(&pframe_body[8])) {
      |       ^~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/include/ieee80211.h:1459:80: error: subscripted value is neither array nor pointer nor vector
 1459 | #define is_zero_mac_addr(Addr) ((Addr[0] == 0x00) && (Addr[1] == 0x00) && (Addr[2] == 0x00) &&   \
      |                                                                                ^
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.c:6965:7: note: in expansion of macro ‘is_zero_mac_addr’
 6965 |   if (is_zero_mac_addr(&pframe_body[8]) || is_broadcast_mac_addr(&pframe_body[8])) {
      |       ^~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/include/ieee80211.h:1460:22: error: subscripted value is neither array nor pointer nor vector
 1460 |                 (Addr[3] == 0x00) && (Addr[4] == 0x00) && (Addr[5] == 0x00))
      |                      ^
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.c:6965:7: note: in expansion of macro ‘is_zero_mac_addr’
 6965 |   if (is_zero_mac_addr(&pframe_body[8]) || is_broadcast_mac_addr(&pframe_body[8])) {
      |       ^~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/include/ieee80211.h:1460:43: error: subscripted value is neither array nor pointer nor vector
 1460 |                 (Addr[3] == 0x00) && (Addr[4] == 0x00) && (Addr[5] == 0x00))
      |                                           ^
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.c:6965:7: note: in expansion of macro ‘is_zero_mac_addr’
 6965 |   if (is_zero_mac_addr(&pframe_body[8]) || is_broadcast_mac_addr(&pframe_body[8])) {
      |       ^~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/include/ieee80211.h:1460:64: error: subscripted value is neither array nor pointer nor vector
 1460 |                 (Addr[3] == 0x00) && (Addr[4] == 0x00) && (Addr[5] == 0x00))
      |                                                                ^
[... cut ...]
      |                                            ^~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:271: /var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4/core/rtw_mlme_ext.o] Error 1
make[1]: *** [Makefile:1859: /var/tmp/portage/net-wireless/rtl8822bu-5.8.7.4_p20210626/work/88x2bu-d57710b441e71c5dbdb0bc3daae05904a03b21e4] Error 2
make[1]: Leaving directory '/usr/src/linux-5.12.13-pentoo'
make: *** [Makefile:2433: modules] Error 2
```

morrownr, 2021-07-17:
Hi. Interesting. Might I ask what situation you have at hand that causes you to be looking at 11r?

This is something I might like to dig into but I am in the process of moving right now and won't be able to take on something like this for a few more weeks.

ZeroChaos-, 2021-07-19:
802.11r is part of the standard for a decade.  I use it on my network, it's pretty great for a lot of reasons.  It works in this fork https://github.com/EntropicEffect/rtl8822bu which unfortunately doesn't build against my currently 5.12 kernel :-(

Your version of the driver seemed well maintained, but when I tested, 802.11r support was missing.  As this is (I'm assuming) mostly a vendor dump that you are basically doing community service hours for, it's up to you if you want to even try to support this.  :-)

morrownr, 2021-07-19:
Good assessment. This driver is a vendor dump and I am doing community service.

There are a couple ways to work this. We could try to patch the other driver you mentioned for kernels 5.11 and 5.12 or we could use meld to see the difference in the two drivers. The preferred option, imho, would be to work on this driver as it is based on much more modern source and has some key fixes. Do you know how to use meld?

henkv1, 2021-07-20:
A quick look at the code gives me this patch:
```
--- include/ieee80211.h	2021-07-20 10:59:56.475986578 +0200
+++ /usr/src/rtl88x2bu-5.8.7.4/include/ieee80211.h	2021-07-20 11:06:59.724255108 +0200
@@ -1452,13 +1452,31 @@
 #define PORT_FMT "%u"
 #define PORT_ARG(x) ntohs(*((u16 *)(x)))
 
+#ifdef PLATFORM_FREEBSD /* Baron change func to macro */
 #define is_multicast_mac_addr(Addr) ((((Addr[0]) & 0x01) == 0x01) && ((Addr[0]) != 0xff))
 #define is_broadcast_mac_addr(Addr) ((((Addr[0]) & 0xff) == 0xff) && (((Addr[1]) & 0xff) == 0xff) && \
 	(((Addr[2]) & 0xff) == 0xff) && (((Addr[3]) & 0xff) == 0xff) && (((Addr[4]) & 0xff) == 0xff) && \
 				     (((Addr[5]) & 0xff) == 0xff))
 #define is_zero_mac_addr(Addr)	((Addr[0] == 0x00) && (Addr[1] == 0x00) && (Addr[2] == 0x00) &&   \
                 (Addr[3] == 0x00) && (Addr[4] == 0x00) && (Addr[5] == 0x00))
+#else
+extern __inline int is_multicast_mac_addr(const u8 *addr)
+{
+        return (addr[0] != 0xff) && (0x01 & addr[0]);
+}
+ 
+extern __inline int is_broadcast_mac_addr(const u8 *addr)
+{
+        return ((addr[0] == 0xff) && (addr[1] == 0xff) && (addr[2] == 0xff) &&   \
+                (addr[3] == 0xff) && (addr[4] == 0xff) && (addr[5] == 0xff));
+}
 
+extern __inline int is_zero_mac_addr(const u8 *addr)
+{
+        return ((addr[0] == 0x00) && (addr[1] == 0x00) && (addr[2] == 0x00) &&   \
+                (addr[3] == 0x00) && (addr[4] == 0x00) && (addr[5] == 0x00));
+}
+#endif /* PLATFORM_FREEBSD */
 
 #define CFG_IEEE80211_RESERVE_FCS (1<<0)
 #define CFG_IEEE80211_COMPUTE_FCS (1<<1)
```
I'm not sure if it breaks anything, but it works on my system.

ZeroChaos-, 2021-07-20:
I might have broken it with copy and paste, but this didn't cleanly apply for me

henkv1, 2021-07-20:
This works for me:
- copy the code in /tmp/patch
- cd 88x2bu/
- cat /tmp/patch | patch -p1

ZeroChaos-, 2021-07-22:
yeah, it was my fault, I corrupted the patch while copy/paste.  Tested and confirmed it compiles, I'll try to find my hardware to confirm 802.11r actually works with this fix

morrownr, 2021-07-24:
@henkv1 @ZeroChaos- 

If you report that this patch works and doesn't appear to break anything, let me know and I will merge it.

henkv1, 2021-07-25:
I do not have any issues with the patched driver. But I don't know if 802.11r works, because I don't use it.

henkv1, 2021-08-10:
@ZeroChaos- Can you confirm that 802.11r now works?

closed 2021-12-05 by morrownr

misha4gps, 2023-05-14:
@henkv1  - current version successfully compiles with CONFIG_RTW_MBO = y
but 802.11r does not work