morrownr/88x2bu #79: Unable to control 2 or more wifi adapters at the same time single usb hub.

Quackoman, 2021-08-15:
Hey Everyone,

i've had an issue for these 8812BU adapters. If i plug in two or more to the same usb hub, wpa_supplicant is only able to control the last one plugged in. 
I reproduced the same using ubuntu 20.04 arm64-5,4.0-1041raspi and centos 7 arm64-5.10.xx.

To reproduce:
1. Install prerequisites and follow instructions to build driver: ./raspi64.sh ./install-driver.sh
               - or run raspi64.sh and then 
               - make
               - cp 88xbu.ko /lib/modules/uname -r/kernel/drivers/net/wireless/realtek/rtl88xbu/
               - depmod -a
               - modprobe 88xbu
2. Plug in two RTL8812BU usb drives to raspberry pi directly on usb 2.0 and one on usb 3.0 
3. using nmtui connect both of them to wireless, this will work.
4. unplug both from being directly connected to the pi and attach a hub to usb 3.0 or 2.0 port (powered or unpowered)
5. using nmtui connect both to internet, one will work and the other will fail.
6. unplug the unit that fails and plug back into usb-hub
7. using nmtui connect the newly plugged in adapter to wifi(successful)
8. try to change ssid connected on the previously connected adapter(fails)

if i unbind and bind the usb adapter port that is not working, i can return functionality as it will now be the last one plugged in, and so on for the previous one that no longer works if i need to change ssid i can bind and unbind it. 

It seems to me that this can be fixed driver side to read that multiple adapters are on a single hub using the same root usb address: i.e if adapters are plugged in such that their usb assignment is: 1-1.1.1. 1-1.1.2 then issue arises and only last plugged in adapter is able to be controlled. If adapters are plugged in such that they get 1-1.1.1, 1-1.3 then it works. 

Another case tested:
on the same usb hub, if you plug in two wifi adapters that load different wifi-drivers: plug in a RTL8812AU and a RTL8812BU adapter, there is no issue to control each of them via wpa_supplicant/nmtui only happens when same driver module 88xbu is loaded with two identical adapters on the hub.

morrownr, 2021-08-15:
This issue is in the FAQ at the end of README.

Quote: "It seems to me that this can be fixed driver side to read that multiple adapters..."

No doubt about it. Trying plugging in two identical adapters with MT7612u chipsets and you will get two interfaces ready to go.

https://github.com/morrownr/USB-WiFi

Back on topic: I haven't tested this but I think there is a work around:

- download repo to 88x2bu
- copy 88x2bu to 88x2bu-2
- edit dkms.conf in 88x2bu-2 as such:

PACKAGE_NAME="rtl88x2bu-2"
BUILT_MODULE_NAME[0]="88x2bu-2"

- run $ sudo ./install-driver.sh in both folders and see what happens. Let me know.

My thoughts on this issue: Why on earth can Realtek not support us with good in-kernel drivers that don't need basic functionality fixed or added?

Most of my time is taken up keeping this and the other drivers I support up with kernel dev as well as fixing some basic functionality. It could be a while before I have time to look at a real fix but you and others are welcome work on it, document it and submit it.

Lastly, if you need really dependable identical AC1200 usb adapters, get a couple of adapters based on the MT7612u chipset. You won't even have to find and compile a driver.

Quackoman, 2021-08-15:
Thanks for the quick feedback and sorry I missed that ending of the readme. 

I agree, it is crazy that they sell so many of these chipsets and have no care for bettering or simply stabilizing it's functionality.

I'll check out the MT7612u.

Way back when i first ran into this issue with the multiple 8812bu adapters, i tried that workaround with compiling the driver under the -2/-3 name but when loaded it would still name itself 88xbu if I ran an lsmod. and the two adapters will still load the same driver. 

it's one that i gave up on, but when i saw this repo somewhat active and making progress on making it a more stable, I thought maybe let me ask if anyone ever did figure it out. 

thanks again!

closed 2021-08-15 by Quackoman

morrownr, 2021-08-15:
My memory is bad. Here is copy and paste from a guy in the repo that says this works:

There are these things to change:

/include/autoconf.h: DRV_NAME (I'm not sure if changing it is necessary. It's not the kernel module name for the kernel, but the usb driver name for usbcore)
Makefile: MODULE_NAME
dkms.conf: BUILD_MODULE_NAME, PACKAGE_NAME (I'm not sure if changing PACKAGE_NAME is necessary)
/os_dep/linux/usb_intf.c: Change supported vendors, so only the driver you want to load get loaded when a usb device is detected.
Now I don't have to return them to amazon, nice.
Thanks for your repo! (200 stars congrats)
(Feel free to close this issue. I'm leaving it open for now)

currently my lsmod output is showing both drivers:

$ lsmod | grep cfg80211
cfg80211              704512  5 iwlmvm,88x2bu1,88x2bu2,iwlwifi,mac80211

-----
I haven't tested this but in another later msg he reported stable operation. I realize his documentation is not totally complete and I have not tested this myself.

spcharc, 2021-08-21:
Hi @Quackoman , I am the "a guy in the repo that says this works" mentioned by @morrownr 
Currently I am running a very stable hotspot using two 88x2bu adapters. I can tell you that it definitely works, and works well.
@morrownr has already posted how I managed to make it work, so if you have any questions you can ask me. You can also check #73

My setup is:
```
-------------------    ----------------------------------------------------------    ---------------------------------
|(wireless router)|>>>>|88x2bu adpater 1---(wifi extender pc)---88x2bu adapter 2|>>>>|builtin wifi card---(my laptop)|
-------------------    ----------------------------------------------------------    ---------------------------------
```
As you can see, the internet traffic has to go through 3 different wireless dongles to reach the actual router in my home.

I am using 300Mbps AT&T fiber cable. Since 88x2bu supports 867mbps, I can reach the maximum speed with this setup.

--------------
Here is a screenshot of downloading World of Warcraft using Battle.net app. Constantly 30+MB/s
[image]

--------------
Here is a screenshot of "speed test" on Google:
[image]

--------------
This is `lshw` listing NIC devices on my wifi extender pc.
[image]

--------------
This is another screenshot of `lshw` listing USB devices showing both of the drivers.
[image]

--------------
This is `lsmod | grep cfg80211`
[image]

spcharc, 2021-08-21:
The virtual machine method I mentioned under #73 may work for you, but unfortunately letting the virtual machine to control USB devices consumes a lot of CPU resources. If you do not mind you can have a try

spcharc, 2021-08-21:
> If adapters are plugged in such that they get 1-1.1.1, 1-1.3 then it works.

I cannot believe it works this way! I do not even realized USB hubs has something to do with this problem. If one driver can handle multiple devices in this setup, then just use a hub and do not mess with the driver.